zwave-js/zwave-js #1842: [bug] Network heal fails

maivorbim, 2021-02-21:
Network heal command fails with error: _refreshing neighbor list failed: Timeout while waiting for a callback from the controller_

Healing starts at `13:33:09.939 CNTRLR   starting network heal...`

Running Zwavejs 6.4.0. Please let me know if any further details are required to debug this. Thank you.

[zwavejs_4017.log](https://github.com/zwave-js/node-zwave-js/files/6017631/zwavejs_4017.log)

AlCalzone, 2021-02-21:
Which stick are you using? I've had reports that healing on the new 700 series sticks only worked after a firmware update.

maivorbim, 2021-02-21:
Aeotec Zstick Gen5+. I remember healing working on 6.1.0 or 6.2.0, so I don't think it is related to the stick.

AlCalzone, 2021-02-21:
> Timeout while waiting for a callback from the controller

Often means that the controller is stuck somehow and could use a power cycle.
In case of a network heal it is possible that the timeout is chosen too short, but I don't guess your network grew substantially?

maivorbim, 2021-02-21:
it did actually...I think i have doubled network size from 6.1.0/6.2.0 to 6.4.0. I will try doing a power cycle and let you know if that solves it.

AlCalzone, 2021-02-21:
Hm how many nodes? Of those, how many are mains powered and how many FLIRS?

maivorbim, 2021-02-21:
56 total nodes, out of which 11 mains powered nodes (including 2 Aeotec Range Extender 7), 36 battery flirs nodes and 9 battery non-flirs nodes.

AlCalzone, 2021-02-21:
I'll need to calculate, but this could actually mean we're not waiting long enough.

maivorbim, 2021-02-21:
You might be right about the timeout. I shutdown the machine, removed power cable, powered it back on, started HA, let it finish node interview and still getting the same errors as before.

I also saw this in the log:  _17:19:55.575 DRIVER     no handlers registered!_

```
17:19:29.649 DRIVER « [REQ] [RequestNodeNeighborUpdate]
                        callback id:   174
                        update status: UpdateFailed
17:19:29.650 CNTRLR « [Node 092] refreshing neighbor list failed...
17:19:29.650 CNTRLR   [Node 092] failed to update the neighbor list after 5 attempts, healing failed
17:19:55.569 SERIAL « 0x01050048ae223e                                                     (7 bytes)
17:19:55.570 SERIAL » [ACK]                                                                   (0x06)
17:19:55.571 DRIVER « [REQ] [RequestNodeNeighborUpdate]
                        callback id:   174
                        update status: UpdateDone
17:19:55.575 DRIVER   handling request RequestNodeNeighborUpdate (72)
17:19:55.575 DRIVER     no handlers registered!
17:21:28.375 SERIAL « 0x010b0004005a05310505011788                 
```

I have also tested healing on a different smaller network (19 nodes), also on 6.4.0, and there the healing worked, so I don't think it is a bug, more likely a timeout issue as you suspected.

AlCalzone, 2021-02-21:
Okay, based on your network size it seems like we should be waiting around 3 minutes for the callback, rather than the fixed 65 seconds we currently wait.

scyto, 2021-02-24:
should the timeout dynamically scale with the number of devices?

AlCalzone, 2021-02-24:
It should, but doesn't right now.

AlCalzone, 2021-02-24:
@maivorbim can you test the linked PR please?

PR #1895 (2021-02-26): fix: make heal node callback timeout depend on the network size
fixes: #1842
closed 2021-02-26 by AlCalzone