zwave-js/zwave-js #1844: Thermofloor Z-TRM3 thermostat climate entity unavailable in version 1.2.1 (working in 1.1.1)

robinsmidsrod, 2021-02-22:
### Version

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.2.1
zwavejs version: 6.4.0
Home Assistant Core (in Docker) 2021.2.3 
Thermofloor Z-TRM3 thermostat firmware version 4.0.33 (config/devices/0x019b/z-trm3.json)

### Describe the bug

My Z-TRM3 thermostat (climate entity) is no longer visible in Home Assistant. It was visible and working with version 1.1.1. The device that should have the climate entity is available, but all of the temperature sensors are shown as unavailable.

### To Reproduce

Steps to reproduce the behavior:

1. Go to 'Thermostat device in Home Assistant'
2. Click on 'Climate' entity (it was loaded from cache)
3. See error message about the device being unavailable.

### Expected behavior

The climate entity should be detected and it should have the correct current setpoint and current temperature of the specified temperature sensor (based on the [10-112-0-2] Sensor mode).

This last thing about the correct temperature sensor being used has never worked, even in previous versions. It would always use the sensor [10-49-2-Air temperature] (which is the first in the list).

### Additional context

When I start up zwavejs2mqtt 1.2.1 I get this message twice (I have two Z-TRM3 thermostats):

`Unable to discover climate device, there is no valid temperature valueId`

This seems to be related to https://github.com/zwave-js/node-zwave-js/issues/1184 which added compat values for CC 0x31 to the definition of z-trm3.json.

If I removed all of those compat additions to z-trm3.json from that issue, then all of the temperature sensors show up in Home Assistant, and the error message mentioned above is no longer shown, but the climate entity is still not detected.

I even went as far as fully re-interviewing the node, removing the device and all the entities (manually) in Home Assistant to ensure it was fully detected again. I also stopped HA and zwavejs2mqtt, let zwavejs2mqtt start up fully before I restarted HA, but still no dice.

My only solution to get the climate entity back to working was to downgrade zwavejs2mqtt to version 1.1.1.

I didn't see anything in the release notes between 1.1.1 and 1.2.1 that indicates that work was done on anything thermostat-related, so this might be related to something else causing action at a distance.

One last note of relevance; if I keep the compat changes (0x31 CC) made to z-trm3.json mentioned above, but only downgrade to 1.1.1, the climate entity shows up in HA, but the current temperature value is unavailable. This seems to indicate that here might be two (or potentially three, if we count the wrong temperature sensor being used) inter-related issues at play here.

ismarslomic, 2021-02-23:
I'm experiencing exactly same issue. I got 3 Heatit Z-TRM3 thermostats and after upgrading `zwavejs2mqtt` from `v1.1.0` to `v1.2.3` yesterday (Home Assistant was at same version `2021.2.3` before and after upgrade) the _thermostat_ entity for all 3 devices is in `Unavailable` status in Home Assistant and I cant control thermostats anymore. All three thermostats worked as expected in `zwavejs2mqtt` `v1.1.0`. I tried to remove and add one of the thermostats, but that did not help.

AlCalzone, 2021-02-23:
@ismarslomic node-zwave-js 6.3+ is incompatible with the current HA release. Downgrade zwavejs2mqtt or wait for the next beta (probably tomorrow).

ismarslomic, 2021-02-23:
Tnx for quick reply! I will wait for tomorrow then, not a big issue! Just wanted to help out with confirming the issue. Please let me know if you need support on digging more into details and logs from my side!

ismarslomic, 2021-02-23:
Side question, ZWave To Mqtt UI display the Product code for my HeatIt Z-TRM3 devices as `TF016`. Is this correct?

[image]

AlCalzone, 2021-02-23:
I think that is a different device - at least the configuration parameters are completely different. See https://devices.zwave-js.io/
If you're sure, please open a separate issue, so we can track that.

ismarslomic, 2021-02-24:
I need to demount my devices from the wall to verify. It shows that they are labeled with "TF 021". I sent an email to HeatIt and they informed that "TF 021" and "TF 016" are the same product, just differently labeled. So it means that this issue is also relevant for TF016 and not only Z-TRM3.

AlCalzone, 2021-02-24:
And `TF 021` is the same as `Z-TRM3`?

ismarslomic, 2021-02-24:
I don´t think they are identical. Both are thermostats, but different versions/models. HeatIt sent me Installers Manual for my product and it is not labeled `Z-TRM3` but `Firmware 1.8`. I'm quite confused myself, because it is not possible to find any products on their web site labeled with `TF 021` nor `TF 016`. However, I think that mapping to [TF 016](https://devices.zwave-js.io/?jumpTo=0x019b:0x0001:0x0001:1.8) in zwavejs should work fine.

AlCalzone, 2021-02-24:
Oh wow, that is confusing. Just let us know if your device isn't detected correctly and/or has wrong config parameters, so we can update the files to match.

AlCalzone, 2021-02-25:
@robinsmidsrod I've had contact with the HeatIt support. They can't reproduce with the PC controller. Unfortunately I don't have the device. Maybe you could try to reproduce with that and generate a log?
Requesting the heating setpoint would be a start.

robinsmidsrod, 2021-02-25:
@AlCalzone Not sure I know which software you're referring to when you say _PC controller_. Can you provide a link?

robinsmidsrod, 2021-02-25:
@AlCalzone If you're referring to this one, https://aeotec.freshdesk.com/support/solutions/articles/6000226205-z-wave-command-class-configuration-tool-download- , should I download version 5 or is it enough with the older v4 at the bottom of the page?

AlCalzone, 2021-02-25:
@robinsmidsrod It is part of Simplicity Studio, which you'll find under https://www.silabs.com/developers/z-wave -> downloads.
I've also tried to contact you on the HA discord, are you still active there?

robinsmidsrod, 2021-02-25:
@AlCalzone Not so active on Discord, but I found your communication there. Will follow up there and report back here when something of significance shows up.

ismarslomic, 2021-02-25:
I can confirm that this issue has been resolved in `zwavejs2mqtt:1.4.0` and ha `2021.3.0b1`

robinsmidsrod, 2021-02-26:
@robertsLando The issue with the first temperature sensor always being using for the climate entity in HA, instead of the one specified in [10-112-0-2] Sensor mode, is that a Home Assistant issue, or is there something zwavejs or zwavejs2mqtt can do about that? (See the expected behavior paragraph in the bug description for details)

robertsLando, 2021-02-26:
Hass issue if you are using zwavejs integration. BTW you could use MQTT to discover that climate and manually fix it

robinsmidsrod, 2021-02-26:
@robertsLando I'm using the zwavejs HA integration, so I'll create an issue on the HA side referencing this one. Thanks for the clarification.

AlCalzone, 2021-03-01:
Reopening so we have something to track the investigation.

robinsmidsrod, 2021-03-08:
@AlCalzone I saw this text in the 6.6.1 release notes: "Fixed the length validation in sequenced Security S0 Message Encapsulation commands" Should it have any impact on this issue?

AlCalzone, 2021-03-08:
No - that was purely about the payload length, not invalid nonces.

jmgiaever, 2021-03-16:
Hi, 

Any news on this? I've tried to solve this matter with ThermoFloor for the 2020 version of the Heat-IT ZDim devices. Their flooding the network with Nonce's (I'll call them nonesenses :stuck_out_tongue_winking_eye:) and Meter CC reports. The Meter CC reports is sent often, even though there's no changes in the values - and particularly when they're turned off.  I also have one 2019 version that is silent as a grave whenever it's off or no changes in values. 

Today they wrote me that they had in co-operation with the lead developer of Z-Wave JS @AlCalzone found out that it's not possible to reproduce the problem regards to the «nonesenses» and that both the devices and the driver works correctly. :thinking:

AlCalzone, 2021-03-16:
They were in fact not able to reproduce this issue with the PC controller - I've seen the logs. But one of their support staff agreed to send me a Z-TRM3 so I can investigate with zwave-js. Will let you know when I know more. Good to know that the ZDim could also be affected.

jmgiaever, 2021-03-16:
That's great that they will. I guess they use the same logic across their devices. Could it be an idea to actually ask Silabs how they deal with it in PC controller? If they actually invalidate nonces as the specs say they should.

jmgiaever, 2021-03-16:
> I don´t think they are identical. Both are thermostats, but different versions/models. HeatIt sent me Installers Manual for my product and it is not labeled `Z-TRM3` but `Firmware 1.8`. I'm quite [...]

Hi, I have one unmounted Z-TRM3. It's written `TF 058` on the back of it.

AlCalzone, 2021-03-16:
> Could it be an idea to actually ask Silabs how they deal with it in PC controller?

My information is that the PC controller itself is a dev tool. It does some things that would not be okay in a certified controller. But I've seen the Zniffer traces - there were no duplicate nonces in the lab environment they tested.

robinsmidsrod, 2021-05-25:
In release 7.5.2 this is mentioned in the bugfixes section: _A failed transmission of a NonceReport is no longer retried and no longer has an influence on the device status_

Does this in any way have anything to do with this issue?

AlCalzone, 2021-05-25:
This fix is for the opposite direction. If the device requested a nonce from us and sending that failed, we won't try again.

Do I understand correctly that the original problem is still happening for you guys? If so, could you try re-interviewing and healing the network in the latest version? Now that the SUC return route is being assigned and the healing strategy was improved, maybe things got better.

robinsmidsrod, 2021-05-25:
@AlCalzone I do have to admit that the issue has been seen less lately. I've seen it 4 times in today's log, and only for a single device (not both of my Z-TRM3s). I also have to confirm that the original bug reported now matches the expected behavior (no issues), also because of additional fixes on the Home Assistant side of things.

Do you want to keep this open for further investigation, or should we close it? Have you heard anything more from Heatit/Thermofloor on this issue?

AlCalzone, 2021-05-25:
> Have you heard anything more from Heatit/Thermofloor on this issue?

No, but I don't think it is terribly helpful to try and reproduce this on my own. The logs I've seen from Heatit seemed clean, which makes me think I won't be able to reproduce in a test network. If you got your hands on a Zniffer though, that could give us something to investigate further.