zwave-js/zwave-js #1845: Fibaro FGR222 reporting value of 254 in field that should be between 0-99

foxy82, 2021-02-22:
In my Fibaro FGR222 I'm getting a value of 254 in a field that ZWaveJS says can only be between 0-99:

[image]

I set this to 2 and then 0 but it gets set back to 254 but this doesn't look to be due to anything received from the device:

```
2021-02-22 09:04:13.209 INFO ZWAVE: Calling api writeValue with args: [
  {
    nodeId: 39,
    commandClass: 145,
    endpoint: 0,
    property: 'fibaro',
    propertyKey: 'venetianBlindsPosition'
  },
  2,
  [length]: 2
]
2021-02-22 09:04:13.210 INFO ZWAVE: Writing 2 to 39-145-0-fibaro-venetianBlindsPosition
2021-02-22 09:04:13.246 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-02-22 09:04:14.368 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:14.371 INFO ZWAVE: Node 39: value updated: 49-0-Power 0 => 25.1
2021-02-22 09:04:17.214 INFO ZWAVE: Node 27: metadata updated: 49-1-Power
2021-02-22 09:04:17.217 INFO ZWAVE: Node 27: value updated: 49-1-Power 0 => 0
2021-02-22 09:04:23.495 INFO ZWAVE: Node 39: value updated: 38-0-currentValue unknown => 99
2021-02-22 09:04:23.512 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:23.513 INFO ZWAVE: Node 39: value updated: 49-0-Power 25.1 => 0
2021-02-22 09:04:32.287 DEBUG SOCKET: Event ZWAVE_API emitted to VrcMDxkJFuHVm-3-AAAT
2021-02-22 09:04:32.288 INFO ZWAVE: Calling api writeValue with args: [
  {
    nodeId: 39,
    commandClass: 145,
    endpoint: 0,
    property: 'fibaro',
    propertyKey: 'venetianBlindsPosition'
  },
  0,
  [length]: 2
]
2021-02-22 09:04:32.289 INFO ZWAVE: Writing 0 to 39-145-0-fibaro-venetianBlindsPosition
2021-02-22 09:04:32.325 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-02-22 09:04:33.453 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:33.454 INFO ZWAVE: Node 39: value updated: 49-0-Power 0 => 13.9
2021-02-22 09:04:35.277 INFO ZWAVE: Node 25: value updated: 50-1-value-65537 4 => 4
2021-02-22 09:04:35.280 INFO ZWAVE: Node 25: value updated: 50-1-deltaTime-65537 0 => 0
2021-02-22 09:04:36.034 INFO APP: GET /health/zwave 200 1.073 ms - 1875
2021-02-22 09:04:37.513 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:37.514 INFO ZWAVE: Node 39: value updated: 49-0-Power 13.9 => 12.2
2021-02-22 09:04:45.638 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:45.639 INFO ZWAVE: Node 39: value updated: 49-0-Power 12.2 => 0
2021-02-22 09:04:47.029 INFO ZWAVE: Node 39: value updated: 38-0-currentValue 99 => 0
2021-02-22 09:04:47.048 INFO ZWAVE: Node 39: metadata updated: 49-0-Power
2021-02-22 09:04:47.049 INFO ZWAVE: Node 39: value updated: 49-0-Power 0 => 0
```

```
09:04:00.094 CNTRLR   [Node 039] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
09:04:00.096 CNTRLR   [Node 039] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 0]
09:04:00.098 SERIAL » [ACK]                                                                   (0x06)
09:04:00.100 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[MultilevelSensorCCReport]
                          type:  Power
                          scale: Watt
                          value: 0
09:04:13.216 SERIAL » 0x010f0013270891010f260102020025e5b4                                (17 bytes)
09:04:13.216 DRIVER » [Node 039] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      229
                      └─[FibaroVenetianBlindCCSet]
                          position: 2
[... 80 lines cut ...]
09:04:45.635 SERIAL « 0x010c0004002706310504220000c4                                      (14 bytes)
09:04:45.636 CNTRLR   [Node 039] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
09:04:45.639 CNTRLR   [Node 039] [~] [Multilevel Sensor] Power: 12.2 => 0               [Endpoint 0]
09:04:45.640 SERIAL » [ACK]                                                                   (0x06)
09:04:45.641 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[MultilevelSensorCCReport]
                          type:  Power
                          scale: Watt
                          value: 0
09:04:47.028 SERIAL « 0x01090004002703260300f3                                            (11 bytes)
09:04:47.028 CNTRLR   [Node 039] [~] [Multilevel Switch] currentValue: 99 => 0          [Endpoint 0]
09:04:47.030 SERIAL » [ACK]                                                                   (0x06)
09:04:47.030 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 0
09:04:47.047 SERIAL « 0x010c0004002706310504220000c4                                      (14 bytes)
09:04:47.048 CNTRLR   [Node 039] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
09:04:47.049 CNTRLR   [Node 039] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 0]
09:04:47.049 SERIAL » [ACK]                                                                   (0x06)
09:04:47.050 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[MultilevelSensorCCReport]
                          type:  Power
                          scale: Watt
                          value: 0
```

AlCalzone, 2021-02-22:
@foxy82 please **attach** a complete logfile so I can investigate.

foxy82, 2021-02-22:
```
09:03:05.510 SERIAL « 0x0114000400180e600d0101320221440000052b0000ef                      (22 bytes)
09:03:05.512 CNTRLR   [Node 024] [~] [Meter] value[65537]: 13.23 => 13.23               [Endpoint 1]
09:03:05.515 CNTRLR   [Node 024] [~] [Meter] deltaTime[65537]: 0 => 0                   [Endpoint 1]
09:03:05.517 SERIAL » [ACK]                                                                   (0x06)
09:03:05.518 DRIVER « [Node 024] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      1
                        │ destination: 1
                        └─[MeterCCReport]
                            type:       Electric
                            scale:      kWh
                            rate type:  Consumed
                            value:      13.23
                            time delta: 0 seconds
09:03:07.091 SERIAL « 0x01140004000f0e3202213400000029000100000030d3                      (22 bytes)
09:03:07.092 CNTRLR   [Node 015] [~] [Meter] value[66049]: 4.8 => 4.1                   [Endpoint 0]
09:03:07.095 CNTRLR   [Node 015] [~] [Meter] previousValue[66049]: 4.1 => 4.8           [Endpoint 0]
09:03:07.097 CNTRLR   [Node 015] [~] [Meter] deltaTime[66049]: 1 => 1                   [Endpoint 0]
09:03:07.099 SERIAL » [ACK]                                                                   (0x06)
09:03:07.101 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:        Electric
                          scale:       W
                          rate type:   Consumed
                          value:       4.1
                          time delta:  1 seconds
                          prev. value: 4.8
09:03:09.683 SERIAL « 0x0110000400200a600d0101310504220000be                              (18 bytes)
09:03:09.685 CNTRLR   [Node 032] [Multilevel Sensor] Power: metadata updated            [Endpoint 1]
09:03:09.687 CNTRLR   [Node 032] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 1]
09:03:09.689 SERIAL » [ACK]                                                                   (0x06)
09:03:09.690 DRIVER « [Node 032] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      1
                        │ destination: 1
                        └─[MultilevelSensorCCReport]
                            type:  Power
                            scale: Watt
                            value: 0
09:03:10.091 SERIAL « 0x01140004000f0e320221340000003000010000002ad0                      (22 bytes)
09:03:10.092 CNTRLR   [Node 015] [~] [Meter] value[66049]: 4.1 => 4.8                   [Endpoint 0]
09:03:10.094 CNTRLR   [Node 015] [~] [Meter] previousValue[66049]: 4.8 => 4.2           [Endpoint 0]
09:03:10.097 CNTRLR   [Node 015] [~] [Meter] deltaTime[66049]: 1 => 1                   [Endpoint 0]
09:03:10.098 SERIAL » [ACK]                                                                   (0x06)
09:03:10.099 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[MeterCCReport]
                          type:        Electric
                          scale:       W
                          rate type:   Consumed
                          value:       4.8
                          time delta:  1 seconds
                          prev. value: 4.2
09:03:26.546 SERIAL « 0x0110000400210a600d0101310504220000bf                              (18 bytes)
09:03:26.547 CNTRLR   [Node 033] [Multilevel Sensor] Power: metadata updated            [Endpoint 1]
09:03:26.551 CNTRLR   [Node 033] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 1]
09:03:26.553 SERIAL » [ACK]                                                                   (0x06)
09:03:26.556 DRIVER « [Node 033] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      1
                        │ destination: 1
                        └─[MultilevelSensorCCReport]
                            type:  Power
                            scale: Watt
                            value: 0
09:03:41.260 SERIAL « 0x010d0004002507600d01012603009c                                    (15 bytes)
09:03:41.261 CNTRLR   [Node 037] [~] [Multilevel Switch] currentValue: 99 => 0          [Endpoint 1]
09:03:41.264 SERIAL » [ACK]                                                                   (0x06)
09:03:41.266 DRIVER « [Node 037] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
[... 163 lines cut ...]
                          current value: 0
09:04:47.047 SERIAL « 0x010c0004002706310504220000c4                                      (14 bytes)
09:04:47.048 CNTRLR   [Node 039] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
09:04:47.049 CNTRLR   [Node 039] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 0]
09:04:47.049 SERIAL » [ACK]                                                                   (0x06)
09:04:47.050 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[MultilevelSensorCCReport]
                          type:  Power
                          scale: Watt
                          value: 0
09:12:18.361 SERIAL » 0x010e00132607600d000125010025e749                                  (16 bytes)
09:12:18.361 DRIVER » [Node 038] [REQ] [SendData]
[... cut ...]
                        │ destination: 1
                        └─[BinarySwitchCCSet]
                            target value: false
09:12:18.364 SERIAL « [ACK]                                                                   (0x06)
09:12:18.371 SERIAL « 0x0104011301e8                                                       (6 bytes)
09:12:18.371 SERIAL » [ACK]                                                                   (0x06)
09:12:18.371 DRIVER « [RES] [SendData]
                        was sent: true
```

foxy82, 2021-02-22:
I just found this in my logs from yesterday:
```
16:45:32.400 DRIVER « [Node 039] [REQ] [ApplicationCommand]
                      └─[FibaroVenetianBlindCCReport]
                          position: 254
16:45:32.404 SERIAL » 0x010a00130b0370050f25d061                                          (12 bytes)
16:45:32.404 DRIVER » [Node 011] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      208
                      └─[ConfigurationCCGet]
                          parameter #: 15
16:45:32.405 CNTRLR « [Node 039] received venetian blind state:
                      position: 254
                      tilt:     undefined
```

AlCalzone, 2021-02-22:
Ah, so that comes from the initial query.

PR #1847 (2021-02-22): feat: support configuring the treatment of "unknown", fix Fibaro CC
With this PR, values that could previously have `"unknown"` as the value now expose `undefined` in this case instead, since exposing a string where numbers/booleans are used can be very unexpected. To configure this and gain access to the "unknown" state, the driver option `preserveUnknownValues` (default: `false`) was added.

This PR also fixes the parsing of the Fibaro CC values, which can also report unknown.

fixes: #1845
closed 2021-02-22 by AlCalzone