zwave-js/zwave-js #1856: CT101 Thermostat setpoint reverts to old value

jjlawren, 2021-02-22:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

After manual control of my CT101 thermostat at the device, the updated temperature setpoint will accurately reflect in zwave-js. However, forcing a value refresh (via zwjs2m control panel) after this point will cause the reported setpoint to revert back to the value from _before_ the manual adjustment. Further adjustments to the thermostat from zwave-js work and are accurately reported.

[fb4f583f.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6025936/fb4f583f.values.jsonl.txt)
[fb4f583f.json.txt](https://github.com/zwave-js/node-zwave-js/files/6025937/fb4f583f.json.txt)

**Device information**

Which device(s) is/are affected (make/model)?  Radio Thermostat CT101
What are the node IDs? 10

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [x] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Described above.

**Additional context**

Behavior was recently improved for this device in #1749. Manual control was one of the triggers for that issue, so perhaps may be relevant.

**Logfile:**
[zwavejs_1.setpoint_revert.log](https://github.com/zwave-js/node-zwave-js/files/6025906/zwavejs_1.setpoint_revert.log)
Events:
1. Change setpoint from 68->69 with manual adjustment
2. Refresh values, which reverts zwave-js setpoint to 68 while the actual value remains at 69 on the thermostat.

AlCalzone, 2021-02-23:
Tbh, the log looks perfectly normal:
The unsolicited report comes in with 69 °F
```
17:00:43.271 DRIVER « [Node 010] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         69 °F
```

The response to the query contains 68 °F:
```
17:21:45.911 DRIVER » [Node 010] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      73
                      └─[ThermostatSetpointCCGet]
                          setpoint type: Heating
...
17:21:46.112 DRIVER « [Node 010] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Heating
                          value:         68 °F
```

This should go away with #1787 because the value query won't be repeated, but the underlying issue is that the device reports a wrong value.

jjlawren, 2021-02-23:
Ah, too bad. I was hoping there would be some indication that we were reusing some cached value. These devices have some odd behaviors.

closed 2021-02-23 by jjlawren

melyux, 2022-02-18:
I noticed this problem with the CT101 a while back too. Manual setpoints are not reported when later queried from the thermostats. I wrote it off as "never going to be fixed" and wrote a Home Assistant automation to repeat all setpoint changes  (which ARE accurately reported at the time they happen) to the thermostats, so even manual changes would be permanently stored in the thermostat for the next time the thermostat is queried.

So #1787 fixed this issue for you @jjlawren? Looks like I won't need the repeat automation anymore if so.

jjlawren, 2022-02-18:
Yes, my CT101 has been behaving well for a very long time now. The only "issue" I had was infrequent humidity reports, but that seems to be a device firmware behavior. That can be addressed by manually polling the device if needed.

jjlawren, 2022-02-18:
No, I do not need to poll anything. In fact, I just checked and the humidity sensor is updating regularly now.

I'd make sure you're on a recent release and to reinterview the device.

jjlawren, 2022-02-19:
Not sure what to tell you. It was common knowledge about a year ago that they didn't push updates but it's been reliable for me without any additional workarounds for quite a while now. Maybe different firmware versions behave differently? I've also configured the temp/humidity reporting level thresholds to the lowest available on the device.

[image]

This particular issue was resolved long ago. I'd suggest to open a new one if the root cause is different.

melyux, 2022-02-19:
It's not an issue, just a discussion about something related. Is there a discussion forum we could use to investigate this? It'd be a godsend.