zwave-js/zwave-js #1860: SES302 Temperature Sensors Issue [bug]

jasonwragg, 2021-02-23:
[node_11.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6013535/node_11.txt)
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version

Build/Run method

- [ X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.2.2
zwavejs version: 6.4.0

On Secure Meters (UK) Temperature sensor the external tank sensors all report the same value as the main sensor

A clear and concise description of what the bug is.

Added the device

### Expected behavior

Each end point should receive its own temperature

### Additional context

Add any other context about the problem here.

AlCalzone, 2021-02-23:
Okay, the fact that the values are mapped to all endpoints is a bug. I've already fixed it in master.

Please re-interview the device, wake it up and post a log of that. The current log is pretty empty because it only contains the update after restart from cache.

jasonwragg, 2021-02-23:
> Okay, the fact that the values are mapped to all endpoints is a bug. I've already fixed it in master.
> 
> To figure out why only one report for the root endpoint comes in, please re-interview the device, wake it up and post a log of that. The current log is pretty empty because it only contains the update after restart from cache.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6027857/zwavejs_1.log)

AlCalzone, 2021-02-23:
You didn't have to exclude it, just repeat the interview :)

AlCalzone, 2021-02-23:
Okay what I can see from the log:
The device supports 5 endpoints, but does not support Multi Channel Association. In this case, zwave-js sets up a node association from the root device to the controller and calls it a day.

It seems however that this device expects each single endpoint to be associated with a target. At least I think that, because the manual is not exactly very specific:
> Product supports one group for each end point that have a maximum of 2 nodes.
> Only one association group is supported on each end point and its name is “Lifeline”

This is currently not possible, as we need to add support to configure associations an an endpoint.

commit 10e53aff (2021-07-13): feat: rework how endpoints and lifeline associations are handled (#2997)

fixes: #2286
fixes: #2257
fixes: #1989
fixes: #2714
fixes: #1860
fixes: #2910
closed 2021-07-17 by AlCalzone