zwave-js/zwave-js #1861: Add support for configuring associations on endpoints

AlCalzone, 2021-02-23:
This is required for https://github.com/zwave-js/node-zwave-js/issues/1860

jasonwragg, 2021-03-18:
If i was to try and look in to this where would I start in the code?

AlCalzone, 2021-03-19:
Let's see:
`packages\zwave-js\src\lib\controller\Controller.ts` -> The methods to manage associations (`getAssociationGroups`, `getAssociations`, etc.) need to be changed so they optionally accept an endpoint index in addition to the node ID (and then operate on the endpoint).
The used API methods are able to handle the endpoint already.

`packages\zwave-js\src\lib\commandclass\AssociationCC.ts` -> `skipEndpointInterview` might need to be edited so it checks whether the root endpoint supports the CC before returning `true`. I'm not 100% sure here, this would need to be tested. Same for the MultiChannelAssociation CC.

`docs\api\controller.md` needs to be updated to match the changed signatures. It should probably be mentioned that this way should only be used when only the endpoints and not the root device support Associations.

And I'm sure I forgot something...

jasonwragg, 2021-03-23:
Ok thanks, looking at it i think this maybe be beyond my technical ability.

closed 2021-04-23 by AlCalzone