zwave-js/zwave-js #1864: Missing device config for Everspring TST01-1 Flood Detector

mnaz, 2021-02-23:
This sensor is not recognized with node-zwave-js

Everspring TST01-1 Flood Detector
https://products.z-wavealliance.org/products/736

Closest Device Configuration I could find
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0060/st812.json

This doc seems to show the format
https://zwave-js.github.io/node-zwave-js/#/development/config-files

Should I just try and add my sensor product type/id values to the existing file? I am not sure how best to move forward to test with the sensors I have.

thanks for any pointers..
-mark

_Originally posted by @mnaz in https://github.com/zwave-js/node-zwave-js/discussions/1858_

mnaz, 2021-02-23:
I will take on this issue and try to create the appropriate config

mnaz, 2021-02-23:
Project doc on adding a config for reference.
https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=contributing-configuration-files

mnaz, 2021-02-23:
Oddly this existing config from OpenZWave does not follow the pattern
https://github.com/OpenZWave/open-zwave/blob/master/config/everspring/st812.xml

the file name does not seem to match the file content which matches the sensor I am attempting to add.

Not sure if it is better to import this into the proper named file or create a new file from scratch.

AlCalzone, 2021-02-23:
What do you mean? OZW config files look completely different than ours.

mnaz, 2021-02-23:
I was looking at this doc
https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=importing-config-files-from-other-sources
that mentioned importing, maybe I am missing something. I get that xml vs json and all just being extra cautious to step through this correctly :)

I will construct a new file as I am already working on but I have a few question on what should be for associations and paramInformation. Right now I have included what was form the the st812.json file although the content for paramInformation seems wrong. How do I know what this content should be?

AlCalzone, 2021-02-23:
> How do I know what this content should be?

It should be mentioned in the device manual.

mnaz, 2021-02-23:
Ok now I am getting some where. It seems my sensor is actually a rebranded st812. 
https://www.everspring.com/portfolio-item/st812-flood-detector/

From this site it seems they are one in the same: https://community.hubitat.com/t/utilitech-water-leak-sensor/3696
 *      Everspring Flood Sensor       ST812-2
 *      Utilitech Water Leak Detector TST01-1 (Lowe's)

Now how do you suggest I proceed?

AlCalzone, 2021-02-23:
@blhoward2 can you help out? What do you suggest?

blhoward2, 2021-02-25:
I'll check tomorrow.

blhoward2, 2021-02-25:
@mnaz You didn't fill out the issue template. What is the device's manufacuterID, productType, and productId? Sometimes rebranded devices are indistinguishable from the fingerprint. Sometimes they are.

mnaz, 2021-02-25:
I did not see that there was a template. I updated the issue with what I think you are looking for based on another issue

AlCalzone, 2021-02-25:
Might be because you used "reference in new issue". No idea why Github doesn't make you choose a template that way.

mnaz, 2021-03-05:
@blhoward2 any help would be great. thanks

blhoward2, 2021-03-08:
This device information seems suspect to me. As you note, we support the st812. You say the manufacturer should be 0x27a, but that is Zooz's manufacturerId. Now, sometimes manufacturers make devices for other brands, but I'm not aware of Zooz ever before doing so. Are you sure that's the correct fingerprint? Can you upload a screenshot of the UI showing that string for the node?

mnaz, 2021-03-08:
I added node to HA and dumped the network information. This is the part related to the device I am working with it looks like I was wrong in the setting you pointed out.
            "manufacturerId": 96,
            "productId": 1,
            "productType": 11,

          {
            "nodeId": 68,
            "index": 0,
            "status": 3,
            "ready": false,
            "deviceClass": {
              "basic": {
                "key": 4,
                "label": "Routing Slave"
              },
              "generic": {
                "key": 161,
                "label": "Alarm Sensor"
              },
              "specific": {
                "key": 2,
                "label": "Routing Alarm Sensor"
              },
              "mandatorySupportedCCs": [
                32,
                156,
                133,
                128,
                114,
                134
              ],
              "mandatoryControlledCCs": [
                156
              ]
            },
            "isListening": false,
            "isFrequentListening": false,
            "isRouting": true,
            "maxBaudRate": 40000,
            "isSecure": "unknown",
            "version": 3,
            "isBeaming": true,
            "manufacturerId": 96,
            "productId": 1,
            "productType": 11,
            "firmwareVersion": "1.0",
            "neighbors": [],
            "interviewAttempts": 1,
            "interviewStage": 1,
            "commandClasses": [
              {
                "id": 32,
                "name": "Basic",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 156,
                "name": "Alarm Sensor",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 133,
                "name": "Association",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 128,
                "name": "Battery",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 114,
                "name": "Manufacturer Specific",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 134,
                "name": "Version",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 132,
                "name": "Wake Up",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 112,
                "name": "Configuration",
                "version": 1,
                "isSecure": false
              },
              {
                "id": 113,
                "name": "Notification",
                "version": 1,
                "isSecure": false
              }
            ],
            "endpoints": [
              {
                "nodeId": 68,
                "index": 0
              }
            ],
            "values": []
          }

mnaz, 2021-03-08:
I started attempting to add device again with HA and noticed that it did get added and be default had a number of entities disables. After attempting to enable then a few times they seemed to show and then a lot more elements became available.

this is what I see from the UI and note it shows the sensor as Everspring even though it is labeled as "UTILITECH Wireless Leak Detector" on the battery door. I wish I could say why this works now ( i did update HA a few times through this), I will see how it goes but it seems this is working. If you think no changes are needed then I can close this tracker item.

thanks for the help and patients.

[image]

AlCalzone, 2021-03-16:
@mnaz anything left to do here?

mnaz, 2021-03-17:
@AlCalzone 
Lets call it a day... The sensors have been working and I do not know specifically why they did not before. I guess I do not understand the config fully but it is an odd case with these devices.
I have installed. a few updates since the time I started this and since it is working I do not see any point in keeping this open.
If you agree I can close this issue as no longer an issue.
thanks for the support!

closed 2021-03-17 by AlCalzone