zwave-js/zwave-js #1865: WD200+ sometimes doesn't respond with updated values after turning on.

danieljy, 2021-02-23:
**Describe the bug**

ZWaveJS doesn't seem to consistently send a "value updated" event after turning on a Homeseer WD200+.
The WD100+ doesn't seem to suffer from this problem, sending the "value updated" event after 5 seconds (which I understand is due to the following code:

zwave-js/node-zwave-js@6cd0dfa#diff-9397c8948a06d7061964655d80272f34166e53676c4642a76c3a7799b61e4360R322

which only repolls a device if it doesn't support supervision.

The WD200+ is known not to properly support supervision, so zwavejs should treat it as such.
Despite #1531, which should tell zwave-js to do exactly that, the problem still persists.

**Device information**

Homeseer WD200+

**Did you change anything?**
- [ X] **No**

**Did this use to work before?**
- [ X] **Yes**, in: *(specify application with versions)
  It seems like it sometimes works, but it seems independent of versions.

**How are you using `node-zwave-js`**
- [X ] `zwavejs2mqtt` (latest) docker image
- [X ] `HomeAssistant` version core-2021.2.3

**To Reproduce**

Steps to reproduce the behavior:
You can't reproduce without the device itself but:

Use home assistant `light.turn_on` service on the light associated with the node.
The light turns on (to last dim level), but the state in home assistant flips back to off and never returns to on.

**Additional context**

Add any other context about the problem here.

**Logfile:**

This starts from setting the debug level to `DEBUG` and then after all nodes are interviewed, performing a reinterview on Node 8 which is one of the ones that exhibits the problem.

Please note that after doing all of this and checking the node again, *it's not exhibiting the issue* right now. I will ping back if I can get it back in the state that it's problematic.

[zwavejs_357 4.log](https://github.com/zwave-js/node-zwave-js/files/6029340/zwavejs_357.4.log)

AlCalzone, 2021-02-23:
There's no supervision CC used in the log you posted, but you also didn't try to control the device.

danieljy, 2021-02-23:
I thought you'd asked for just a reinterview. I'll actually try and control the device with logging next. The thing is, now it seems to be working again, so I'm going to wait until i can see it failing. Now I understand what you're trying to do. I guess I'll have to leave ultra debug logging on for a while, since each time I change the logging level, it seems to reset internally, if I understand what I'm seeing.

danieljy, 2021-02-23:
Okay. So I think I figured out how to reproduce. Once you interact with the device itself at least once (manually turn it on and then off), any future requests to turn the device on and off via HA/zwavejs start using supervision it seems.

I believe I turn the device on and off starting at 10:32 manually (hence the scene commands get sent). Then at some point I tried flipping the switch via HA and you can see supervision there!

```
10:32:56.834 SERIAL « 0x011c00498404160411015e86855955725a732627705b2c2b9f6c7aef5b32      (30 bytes)
10:32:56.838 SERIAL » [ACK]                                                                   (0x06)
10:32:56.841 DRIVER « [Node 004] [REQ] [ApplicationUpdateRequest]
                        payload: 0x8404160411015e86855955725a732627705b2c2b9f6c7aef5b
10:32:56.845 CNTRLR « [Node 004] Received updated node info
10:32:57.012 SERIAL « 0x010d00040004055b03868001b6001e                                    (15 bytes)
10:32:57.016 SERIAL » [ACK]                                                                   (0x06)
10:32:57.018 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 134
                          key attribute:   KeyPressed
                          scene number:    1
10:32:57.025 CNTRLR   [Node 004] [!] [Central Scene] scene[001]: 0
10:32:57.034 CNTRLR « [Node 004] received CentralScene notification {
                          "nodeId": 4,
                          "ccId": "Central Scene",
                          "ccCommand": "0x03",
                          "payload": "0x868001"
                      }
10:32:59.772 SERIAL « 0x010d00040004052603636300b50067                                    (15 bytes)
10:32:59.776 CNTRLR   [Node 004] [~] [Multilevel Switch] targetValue: 0 => 99           [Endpoint 0]
10:32:59.782 CNTRLR   [Node 004] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} => {"value":0,"unit":"seconds"}
10:32:59.787 CNTRLR   [Node 004] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
10:32:59.789 SERIAL » [ACK]                                                                   (0x06)
10:32:59.792 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 99
                          target value:  99
                          duration:      [Duration: 0seconds]
10:33:03.961 SERIAL « 0x010d00040004055b03878002b5001f                                    (15 bytes)
10:33:03.963 SERIAL » [ACK]                                                                   (0x06)
10:33:03.965 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 135
                          key attribute:   KeyPressed
                          scene number:    2
10:33:03.969 CNTRLR   [Node 004] [!] [Central Scene] scene[002]: 0
10:33:03.975 CNTRLR « [Node 004] received CentralScene notification {
                          "nodeId": 4,
                          "ccId": "Central Scene",
                          "ccCommand": "0x03",
                          "payload": "0x878002"
                      }
10:33:06.669 SERIAL « 0x010d00040004052603000000b60064                                    (15 bytes)
10:33:06.671 CNTRLR   [Node 004] [~] [Multilevel Switch] targetValue: 99 => 0           [Endpoint 0]
10:33:06.677 CNTRLR   [Node 004] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} => {"value":0,"unit":"seconds"}
10:33:06.682 CNTRLR   [Node 004] [~] [Multilevel Switch] currentValue: 99 => 0          [Endpoint 0]
10:33:06.688 SERIAL » [ACK]                                                                   (0x06)
10:33:06.690 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 0
                          target value:  0
                          duration:      [Duration: 0seconds]
10:33:06.764 SERIAL « 0x010d00040004052603000000b8006a                                    (15 bytes)
10:33:06.767 CNTRLR   [Node 004] [~] [Multilevel Switch] targetValue: 0 => 0            [Endpoint 0]
10:33:06.773 CNTRLR   [Node 004] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} => {"value":0,"unit":"seconds"}
10:33:06.777 CNTRLR   [Node 004] [~] [Multilevel Switch] currentValue: 0 => 0           [Endpoint 0]
10:33:06.780 SERIAL » [ACK]                                                                   (0x06)
10:33:06.783 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 0
                          target value:  0
[... 240 lines cut ...]
                      └─[SupervisionCCGet]
                        │ session id:      9
                        │ request updates: true
                        └─[MultilevelSwitchCCSet]
                            target value: 255
10:33:55.214 SERIAL « [ACK]                                                                   (0x06)
10:33:55.220 SERIAL « 0x0104011301e8                                                       (6 bytes)
10:33:55.221 SERIAL » [ACK]                                                                   (0x06)
10:33:55.221 DRIVER « [RES] [SendData]
[... cut ...]
10:33:55.356 SERIAL « 0x010d00040004056c02090000ca005a                                    (15 bytes)
10:33:55.357 SERIAL » [ACK]                                                                   (0x06)
10:33:55.359 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[SupervisionCCReport]
                          session id:          9
                          more updates follow: false
                          status:              NoSupport
                          duration:            [Duration: 0seconds]

```

AlCalzone, 2021-02-23:
Hm no that doesn't make sense to me. Can you re-interview again, then control from Z-Wave first and make a log of the controlling only?

danieljy, 2021-02-23:
Here is a different WD200+ device that I did not yet manually interact with yet. Just turned it on via the interface:

```
11:12:11.092 SERIAL » 0x010a001308032601ff253626                                          (12 bytes)
11:12:11.096 DRIVER » [Node 008] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      54
                      └─[MultilevelSwitchCCSet]
                          target value: 255
11:12:11.098 SERIAL « [ACK]                                                                   (0x06)
11:12:11.104 SERIAL « 0x0104011301e8                                                       (6 bytes)
11:12:11.105 SERIAL » [ACK]                                                                   (0x06)
11:12:11.106 DRIVER « [RES] [SendData]
                        was sent: true
11:12:11.125 SERIAL « 0x011800133600000200d07f7f7f7f000003000000000301000011              (26 bytes)
11:12:11.126 SERIAL » [ACK]                                                                   (0x06)
11:12:11.128 DRIVER « [REQ] [SendData]
                        callback id:     54
                        transmit status: OK
11:12:16.149 SERIAL » 0x01090013080226022537d9                                            (11 bytes)
11:12:16.150 DRIVER » [Node 008] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      55
                      └─[MultilevelSwitchCCGet]
11:12:16.152 SERIAL « [ACK]                                                                   (0x06)
11:12:16.159 SERIAL « 0x0104011301e8                                                       (6 bytes)
11:12:16.159 SERIAL » [ACK]                                                                   (0x06)
11:12:16.160 DRIVER « [RES] [SendData]
                        was sent: true
11:12:16.179 SERIAL « 0x011800133700000200d07f7f7f7f000003000000000301000010              (26 bytes)
11:12:16.180 SERIAL » [ACK]                                                                   (0x06)
11:12:16.181 DRIVER « [REQ] [SendData]
                        callback id:     55
                        transmit status: OK
11:12:16.195 SERIAL « 0x010d00040008052603636300d0000e                                    (15 bytes)
11:12:16.197 CNTRLR   [Node 008] [~] [Multilevel Switch] targetValue: 255 => 99         [Endpoint 0]
11:12:16.202 CNTRLR   [Node 008] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                      econds"} => {"value":0,"unit":"seconds"}
11:12:16.207 CNTRLR   [Node 008] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
11:12:16.210 SERIAL » [ACK]                                                                   (0x06)
11:12:16.212 DRIVER « [Node 008] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 99
                          target value:  99
                          duration:      [Duration: 0seconds]
```

Now I manually turn on and off:

```
11:15:46.702 SERIAL « 0x011c00498408160411015e86855955725a732627705b2c2b9f6c7aef5b3e      (30 bytes)
11:15:46.703 SERIAL » [ACK]                                                                   (0x06)
11:15:46.705 DRIVER « [Node 008] [REQ] [ApplicationUpdateRequest]
                        payload: 0x8408160411015e86855955725a732627705b2c2b9f6c7aef5b
11:15:46.709 CNTRLR « [Node 008] Received updated node info
11:15:46.869 SERIAL « 0x010d00040008055b03f58001d00007                                    (15 bytes)
11:15:46.871 SERIAL » [ACK]                                                                   (0x06)
11:15:46.872 DRIVER « [Node 008] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 245
                          key attribute:   KeyPressed
                          scene number:    1
11:15:46.875 CNTRLR   [Node 008] [!] [Central Scene] scene[001]: 0
11:15:46.880 CNTRLR « [Node 008] received CentralScene notification {
                          "nodeId": 8,
                          "ccId": "Central Scene",
                          "ccCommand": "0x03",
                          "payload": "0xf58001"
                      }
11:15:49.636 SERIAL « 0x010d00040008052603636300d0000e                                    (15 bytes)
[... 48 lines cut ...]
                      │ callback id:      58
                      └─[SupervisionCCGet]
                        │ session id:      10
                        │ request updates: true
                        └─[MultilevelSwitchCCSet]
                            target value: 255
11:17:29.832 SERIAL « [ACK]                                                                   (0x06)
11:17:29.838 SERIAL « 0x0104011301e8                                                       (6 bytes)
11:17:29.839 SERIAL » [ACK]                                                                   (0x06)
11:17:29.839 DRIVER « [RES] [SendData]
                        was sent: true
11:17:29.859 SERIAL « 0x011800133a00000200d07f7f7f7f00000300000000030100001d              (26 bytes)
11:17:29.860 SERIAL » [ACK]                                                                   (0x06)
11:17:29.861 DRIVER « [REQ] [SendData]
                        callback id:     58
                        transmit status: OK
11:17:29.875 SERIAL « 0x010d00040008056c020a0000d0004f                                    (15 bytes)
11:17:29.876 SERIAL » [ACK]                                                                   (0x06)
11:17:29.878 DRIVER « [Node 008] [REQ] [ApplicationCommand]
                      └─[SupervisionCCReport]
                          session id:          10
                          more updates follow: false
                          status:              NoSupport
                          duration:            [Duration: 0seconds]
```

AlCalzone, 2021-02-23:
Ahh I have a suspicion...

AlCalzone, 2021-02-23:
The device sends a NIF when you interact with it. After a restart from cache, the driver didn't correctly remember that it has received that information already, so the CCs got overwritten.

Please try #1871 if you can.

danieljy, 2021-02-23:
@AlCalzone Happy to try, but what's the easiest way for me to do this given my home assistant configuration.

Would it be to start an interactive session in the running docker container (zwavejstomqtt) and do the npm update?
Or is there a better way?

AlCalzone, 2021-02-23:
Uh with HASS, you need to wait for their update. HA is incompatible with zwave-js >= 6.3 right now.
There should be an update soon and I'll release 6.5 tonight, so maybe just wait instead.

danieljy, 2021-02-23:
Aha okay thanks. I will just wait then. Thanks for tracking this down. I figured something weird was happening with the stickiness of the configuration overrides!

closed 2021-02-23 by AlCalzone