zwave-js/zwave-js #1875: Abus SHHA10000 missing

Technikte, 2021-02-23:
**Which device is missing?**

One of my personal devices are missing.
1. Abus Wall Plug SHHA10000 https://products.z-wavealliance.org/products/3144?selectedFrequencyId=1

**What are the IDs?**
1. Abus Wall Plug
Manufacturer ID: 0x0403
Product type: 0x0003  
Product ID: 0x0003

**Did you check if we really don't have it?**

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

See the link to alliance above and a extra german manual -> https://www.abus.com/var/ImagesPIM/d110001/medias/docus/22/ABUS_SHHA10000_BDA_DE_1.3.pdf

**Info**

When I find some time I will try to add the devices by myself if nobody else do it..will create the issue to track it.

AlCalzone, 2021-02-23:
If you find the time to do this, please go ahead.

Technikte, 2021-02-27:
Closed as changes already merged.

closed 2021-02-27 by Technikte