zwave-js/zwave-js #1877: BUG: either all messages or all secure messages not received from locks during initial boot

scyto, 2021-02-24:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

**Describe the bug**
1. it appears (from my limited understand that) either all messages from locks during startup interview cause time-outs or all that require secure comms.   I say this because the controller can happily query basic information like supported command classes).
2. locks complete basic interview only after all nodes have completed
3. all locks are reported in UI as being 'unknown' in the secure column and things like user codes are missing
4. a subsequent re-interview of each lock works fully - secure messages don't time out and interview results in secure status in UI and all data is populated.

As such i don;t think this a basic comms issue.
This is also seems different to what RobC was reporting as his locks didn't report as 'unkown' at the end of first pass.
I realize the new changes will help this.

**my real question is this is something to be worried about that secure devices can't reach secure status / pass secure messgaes untill all other nodes are complete?**

**Device information**
Nodes 2,3,4 and 5
[image]

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [X] **Don't know**, t- i am unclear if this was n issue on all older versions or older controller due to my lack of knowledge at that time,
- [] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [X] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: master
    - `zwavejs2mqtt` branch: master
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. reboot
2. watch logs for first interview cycle
3. after cycle completes re-interview nodes with secure status as unknown.

**Additional context**
App Version 1.3.0+fec72b7.fec72b7
Zwavejs Version 6.5.0-e0b52b1
Zwavejs-server Version 1.0.0-beta.9
[zwavejs_1.log.txt](https://github.com/zwave-js/node-zwave-js/files/6033018/zwavejs_1.log.txt)

AlCalzone, 2021-02-24:
Please share your cache file (`<homeid>.json`) in the state before the log above.

scyto, 2021-02-24:
will do, might take a day or two as i can bever predict when it will what you see above vs restart from cache correctly (directly before this it had invalidated cache and started from scratch again and so I stopped and restarted to capture logs). I will keep an eye on it each time i upgrade the container.

AlCalzone, 2021-02-24:
Ultimately I believe that most of the weird behavior you're seeing is because of inconsistencies in your cache.
I don't know why it breaks so often, but you're the only one I'm getting these weird reports from.

The issue above looks like it happens because the interview starts almost from the beginning, but the Security CC claims it was already interviewed (which happens later), so that isn't attempted again.

scyto, 2021-02-27:
since i filed this i haven't see a single cache break, i know there was a bunch of things done to avoid it.
closing unless it comes back,

closed 2021-02-27 by scyto

scyto, 2021-03-02:
When the issue occurs the cache file is zero bytes in size.
It was invalidated when I did docker-compose down - which made it zero bytes.

I seem to be able to reproduce that 100% on the following build

App Version 2.0.1+a45f763.a45f763
Zwavejs Version 6.5.1-37f4b05
Zwavejs-server Version 1.1.

@AlCalzone are you still looking for cache invalidation bugs if one has repro?

AlCalzone, 2021-03-02:
If you have something that repros, it's something worth looking at. However I am not very experienced with Docker, so maybe the issue will get more traction in the zwavejs2mqtt repo.