zwave-js/zwave-js #1879: Missing device config for Zooz ZEN27 VER 3.0

rgreuel, 2021-02-24:
**Which device is missing?**

Zooz ZEN27 VER 3.0 - S2 Dimmer Light Switch

**What are the IDs?**

Manufacturer ID: 0x027a
Product type: 0xa000  
Product ID: 0xa002  

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
https://www.support.getzooz.com/kb/article/314-zen27-s2-dimmer-ver-3-01-advanced-settings/

AlCalzone, 2021-02-24:
@blhoward2 ?

bepsoccer, 2021-02-24:
Just happened to see this, so I'll chime in.  The device config already exists, https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x027a/zen27_2.8.json.  v2.8+ is still relevant to v3.1

blhoward2, 2021-02-24:
There were new options added.

rgreuel, 2021-02-24:
It looks like additional values were added to parameter 7 and the default value was changed from 1 to 15.

From support:

> The default value for this parameter is actually 15 (not 1) and should be set to 15 so that all of the commands for the switch trigger associated devices, including Z-Wave control.
> 
> For your reference, here's the full description of the setting:
> Association Reports
> Parameter 7: Choose which physical and Z-Wave triggers should prompt the dimmer to send a status change report to associated devices.
> 
> Values: 0 – none; 1 – physical tap on ZEN27 only; 2 – physical tap on connected 3-way switch only; 3 – physical tap on ZEN27 or connected 3-way switch; 4 – Z-Wave command from hub; 5 – physical tap on ZEN27 or Z-Wave command from hub; 6 – physical tap on connected 3-way switch or Z-Wave command from hub; 7 – physical tap on ZEN27 / connected 3-way switch or Z-Wave command from hub; 8 – timer only; 9 – physical tap on ZEN27 or timer; 10 – physical tap on connected 3-way switch or timer; 11 – physical tap on ZEN27 / connected 3-way switch or timer; 12 – Z-Wave command from hub or timer; 13 – physical tap on ZEN27, Z-Wave command from hub, or timer; 14 – physical tap on ZEN27 / connected 3-way switch, Z-Wave command from hub, or timer; 15 – all of the above. Default: 15.
> 
> Size: 1 byte dec

bepsoccer, 2021-02-24:
The default for parameter 7 is 15 the way it is setup using partial parameters(bit mask).
>binary 1111 = decimal 15

You can read more here, https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=partial-parameters.  It confused me at first as well, but definitely simplifies things.

blhoward2, 2021-02-24:
I hadn't looked and you're correct that each needs to be enabled to send the combined value. The changelog does indicate a difference, though, it may not warrant a new file (or conditional parameter) as it is a defaultValue change. Someone needs to compare the new 3.01 list of parameters to what we have and see if anything changes.

closed 2021-02-28 by blhoward2