zwave-js/zwave-js #1881: Error in driver ZWaveError: Failed to initialize the driver: core_1.shouldLogNode is not a function

kpine, 2021-02-24:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

After installing zwave-js-server and zwave-js@6.4.0, I get an error trying to run the server.

**Device information**

N/A

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *(specify application with versions)* 6.4.0 when installed about 4 days ago.
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [X] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch:  Any beta7-beta9.
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [X] Other:  Manually installed with npm or yarn

**To Reproduce**

Steps to reproduce the behavior:
1. Use npm or yarn to install the packages. Install zwave-js@6.4.0 or actually any older version. I have not tried 6.5.0.
2. Start the zwave-server.
3. Watch an exception.
4. See error

**Additional context**

I am building my own zwave-js-container. I build a beta8 container about 4 days ago with the same (?) 6.4.0 and it works fine. I am building a new container, beta9 with 6.4.0 and it no longer works. It doesn't matter whether I install 6.3.0 either.

```
Error in driver ZWaveError: Failed to initialize the driver: core_1.shouldLogNode is not a function
    at Immediate.<anonymous> (/app/node_modules/zwave-js/src/lib/driver/Driver.ts:739:6) {
  code: 5,
  context: undefined,
  transactionSource: undefined
}
```

**Logfile:**
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/6033828/zwave_js.log)

I notice that `npm list` is showing the version 6.5.0 of zwave-js even though I installed 6.4.0, not sure if that means anything.
```
/app # npm list | grep zwave-js
+-- @zwave-js/server@1.0.0-beta.9
`-- zwave-js@6.4.0
  +-- @zwave-js/config@6.5.0
  | +-- @zwave-js/core@6.5.0 deduped
  | +-- @zwave-js/shared@6.5.0 deduped
  +-- @zwave-js/core@6.5.0
  | +-- @zwave-js/shared@6.5.0 deduped
  +-- @zwave-js/serial@6.5.0
  | +-- @zwave-js/core@6.5.0 deduped
  +-- @zwave-js/shared@6.5.0
```

If I search for `shouldLogNode ` in my good container vs. bad, I notice that the function used to be exported, and is not now. Again, I don't know if that actually means anything, but it's a difference. There was a change to this function in 6.5.0 and now it's no longer exported. Not sure why changes there should affect an install of 6.4.0.

Bad
```
node_modules/@zwave-js/core/build/log/shared.js
282:    shouldLogNode(nodeId) {
```

Good
```
@zwave-js/core/build/log/shared.js
25:exports.restoreSilence = exports.unsilence = exports.tagify = exports.messageRecordToLines = exports.messageToLines = exports.messageFitsIntoOneLine = exports.shouldLogNode = exports.ZWaveLogContainer = exports.ZWaveLoggerBase = exports.getNodeTag = exports.LOG_PREFIX_WIDTH = exports.LOG_WIDTH = exports.directionPrefixPadding = exports.CONTROL_CHAR_WIDTH = exports.getDirectionPrefix = exports.timestampFormat = void 0;
281:function shouldLogNode(nodeId) {
290:exports.shouldLogNode = shouldLogNode;
```

kpine, 2021-02-24:
If I install with `yarn add zwave-js@6.4.0`, the yarn.lock file shows:

```
"@zwave-js/config@^6.4.0":
  version "6.5.0"
  resolved "https://registry.yarnpkg.com/@zwave-js/config/-/config-6.5.0.tgz#c4e561c361200ea5b7d6103fad1d1870dfae4ae1"
  integrity sha512-yY3HGrt5c5vTvbIIw5jKIMsT34CMl5r5YZw3NQ2XmPlJISgZPpZf1G7fzXrwz9oOlch5MTUL+8MGwU97YxGIGw==
  dependencies:
    "@zwave-js/core" "^6.5.0"
    "@zwave-js/shared" "^6.5.0"
    alcalzone-shared "^3.0.2"
    ansi-colors "^4.1.1"
    fs-extra "^9.0.1"
    json-logic-js "^2.0.0"
    json5 "^2.1.3"
    semver "^7.3.2"
    winston "^3.3.3"

"@zwave-js/core@^6.4.0", "@zwave-js/core@^6.5.0":
  version "6.5.0"
  resolved "https://registry.yarnpkg.com/@zwave-js/core/-/core-6.5.0.tgz#ee487be7f580225c59451ab43823da5f2f9b9514"
  integrity sha512-2PmOz+q8aFZGZueSQVmJGCksczws5yuxNiDI5mrdsHYVhbuRYXoOQvDkm4ECSY/MLrC7KK6etJgqTO1n6lg43Q==
  dependencies:
    "@alcalzone/jsonl-db" "^1.2.3"
    "@zwave-js/shared" "^6.5.0"
    alcalzone-shared "^3.0.2"
    ansi-colors "^4.1.1"
    moment "^2.29.0"
    moment-timezone "^0.5.31"
    nrf-intel-hex "^1.3.0"
    winston "^3.3.3"

"@zwave-js/serial@^6.4.0":
  version "6.5.0"
  resolved "https://registry.yarnpkg.com/@zwave-js/serial/-/serial-6.5.0.tgz#6734fd0bb0d00073760aad555e844f3d3e109686"
  integrity sha512-/WU/ChpuvX1h7W2KyvgtxCbd3VgGFcvF2J7AKApnXFZ2/Rt9sH+BWP3u0OP6930fcZTQO4OQsg/F6oF9THKAvg==
  dependencies:
    "@zwave-js/core" "^6.5.0"
    alcalzone-shared "^3.0.2"
    serialport "^9.0.1"
    winston "^3.3.3"

"@zwave-js/shared@^6.2.0", "@zwave-js/shared@^6.5.0":
  version "6.5.0"
  resolved "https://registry.yarnpkg.com/@zwave-js/shared/-/shared-6.5.0.tgz#e238950f883ae98b67e337c613262afe4de98e10"
  integrity sha512-1Ht/QXKL/WoqISqbVKlPLnmpMEQQG4Tdzf8jLjzW152NglJg7+4IY9uSTkECgGvPyUxTU/YFyniNbZ6ndvklAg==
  dependencies:
    alcalzone-shared "^3.0.2"
```
It wants 6.4.0 but it's installing 6.5.0 dependencies?

kpine, 2021-02-24:
Ok, I didn't know the semver syntax until now, so `^6.4.0` is `>= 6.4.0` and `< 7.0.0-0`. Did the dependencies at v6.5.0 break the public API?

AlCalzone, 2021-02-24:
The public API didn't change, but the internal one often breaks between releases. I guess I should configure the internal dependencies to use ~ ranges instead.

scyto, 2021-02-24:
is that what caused the original issue in https://github.com/zwave-js/zwavejs2mqtt/issues/669 (which by the way the versioning issue seemed to disappear)

AlCalzone, 2021-02-24:
I don't think so - 6.5.0 didn't exist until a few hours ago.

PR #1886 (2021-02-24): chore(lerna): use exact versions for bumps
fixes: #1881
(at least in the future)
closed 2021-02-24 by AlCalzone