zwave-js/zwave-js #1882: Yale lock YRD226, but it has a new product ID that doesn't exist in the z-wave DB

0x0a11c0de, 2021-02-24:
I have a Yale lock `YRD226-ZW2-619`. The `YRD226` is in the [z-wave product database](https://products.z-wavealliance.org/products/1878?selectedFrequencyId=2) with a product type of `0x8002` and a product ID of `0x0600`. However, my `YRD226` is reporting a product ID of `0x4600`, which means it shows up in `zwavejs2mqtt` as unknown. The problem is I cannot find this product ID in the z-wave products database. Is it normal for manufacturers to update the product ID but it doesn't show up in the z-wave products DB? Should I add this product ID to the existing YRD226 config file?

Here is the zwavejs log showing the product ID:

```
10:19:12.270 CNTRLR   [Node 005] [+] [Manufacturer Specific] manufacturerId: 297        [Endpoint 0]
10:19:12.271 CNTRLR   [Node 005] [+] [Manufacturer Specific] productType: 32770         [Endpoint 0]
10:19:12.271 CNTRLR   [Node 005] [+] [Manufacturer Specific] productId: 17920           [Endpoint 0]
10:19:12.272 DRIVER « [Node 005] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[ManufacturerSpecificCCReport]
                            manufacturer id: 0x0129
                            product type:    0x8002
                            product id:      0x4600
10:19:12.273 CNTRLR « [Node 005] received response for manufacturer information:
                        manufacturer: ASSA ABLOY (0x0129)
                        product type: 0x8002
                        product id:   0x4600
```

robertsLando, 2021-02-24:
@blhoward2

AlCalzone, 2021-02-24:
> Should I add this product ID to the existing YRD226 config file?

If you are certain that the config parameters etc. match, then yes.

blhoward2, 2021-02-25:
I'm handling this. I'm templating all of the Yale locks and adding new models all at once.

PR #1918 (2021-02-28): refactor(config): use templates for Yale locks
Adds templates to Yale locks. This batch took forever as there were a lot of errors in the files. I did the best I could going back to the manual for each one, but we'll likely have to fix some things as users test. A few of these have no manuals that I could find.

Closes #1882 as part of this push.
closed 2021-02-28 by AlCalzone