zwave-js/zwave-js #1884: [question] Zwave network stall after pressing Zwave remote button

djaydev, 2021-02-24:
Using a Zwave remote to send a "Central Scene" to Home Assistant.  After pressing any button on the remote, HA sees the scene and sends a command back to zwavejsmqtt to turn on or off a light.  The problem is the command to toggle the light is stalled until these response attempts messages timeout.  Sometimes it's stalled/delayed, sometimes the command is just never executed.  Is there a way to avoid trying to update these remote devices after receiving the wake up notification?

Log sample when the button is pressed, but no other command during and/or after can be executed:
```
2021-02-20 20:08:35.307 INFO ZWAVE: Node 52: value notification: 91-0-scene-001 0
20:08:35.309 CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x8a0001"
}
20:08:35.334 CNTRLR « [Node 052] received wakeup notification
2021-02-20 20:08:35.337 INFO ZWAVE: Node 52 is now awake
20:08:35.343 CNTRLR [Node 052] The node is now awake.
2021-02-20 20:08:35.399 INFO ZWAVE: Node 52: value updated: 132-0-wakeUpInterval 0 => 0
2021-02-20 20:08:35.401 INFO ZWAVE: Node 52: value updated: 132-0-controllerNodeId 1 => 1
2021-02-20 20:08:35.452 INFO ZWAVE: Node 52: value updated: 132-0-wakeUpInterval 0 => 0
2021-02-20 20:08:35.453 INFO ZWAVE: Node 52: value updated: 132-0-controllerNodeId 1 => 1
20:08:39.603 CNTRLR [Node 052] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-02-20 20:08:42.634 INFO ZWAVE: Node 52: value notification: 91-0-scene-001 0
20:08:42.636 CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x8b0001"
}
20:08:42.661 CNTRLR [Node 052] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
20:08:42.666 CNTRLR « [Node 052] received wakeup notification
20:08:47.280 CNTRLR [Node 052] The node did not respond after 3 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-02-20 20:08:47.282 INFO ZWAVE: Node 52 is now asleep
20:08:47.292 CNTRLR [Node 052] The node is now asleep.
```
[node_52.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6030557/node_52.json.txt)
[node_53.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6030558/node_53.json.txt)

AlCalzone, 2021-02-24:
I'm guessing that the driver is trying to refresh values from the remote. Since the device is a sleeping one, these have priority over all other messages because it won't be awake long.

As long as the device has been interviewed once before, the issue should be gone with https://github.com/zwave-js/node-zwave-js/pull/1787 - you might want to test that branch.

AlCalzone, 2021-03-14:
fixed in https://github.com/zwave-js/node-zwave-js/pull/2030

closed 2021-03-14 by AlCalzone

djaydev, 2021-03-28:
Hello,

Unfortunately the issue still exist.  If I press the button on the zwave remote, I get the logs below.  When the "Failed to execute controller command after 1/3 attempts" comes up, no other zwave devices can be controlled the whole network stalls for several seconds every press.  On the dashboard all the remotes show as "sleep" and "complete" under status and interview stage.

```
2021-03-28 09:45:11.186 INFO ZWAVE: Node 52 is now awake
2021-03-28T13:45:11.189Z CNTRLR [Node 052] The node is now awake.
2021-03-28T13:45:11.195Z CNTRLR Failed to execute controller command after 1/3 attempts. Scheduling next try i
n 100 ms.
2021-03-28 09:45:11.348 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 12 => 32
2021-03-28 09:45:11.362 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 32 => 32
2021-03-28T13:45:12.191Z CNTRLR » [Node 052] Sending node back to sleep...
2021-03-28 09:45:15.935 INFO ZWAVE: Node 52: value notification: 91-0-scene-002 0
2021-03-28T13:45:15.937Z CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x3c0002"
}
2021-03-28T13:45:15.960Z CNTRLR [Node 052] The node did not respond after 1 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-03-28 09:45:15.963 INFO ZWAVE: Node 52 is now asleep
2021-03-28T13:45:15.971Z CNTRLR [Node 052] The node is now asleep.
2021-03-28T13:45:15.990Z CNTRLR Failed to execute controller command after 1/3 attempts. Scheduling next try i
n 100 ms.
2021-03-28T13:45:15.999Z CNTRLR « [Node 052] received wakeup notification
2021-03-28 09:45:16.001 INFO ZWAVE: Node 52 is now awake
2021-03-28T13:45:16.005Z CNTRLR [Node 052] The node is now awake.
2021-03-28 09:45:16.128 INFO ZWAVE: Node 50: value updated: 38-0-currentValue 0 => 0
2021-03-28 09:45:16.183 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 32 => 52
2021-03-28 09:45:16.197 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 52 => 52
2021-03-28T13:45:17.008Z CNTRLR » [Node 052] Sending node back to sleep...
2021-03-28T13:45:21.156Z CNTRLR [Node 052] The node did not respond after 1 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-03-28 09:45:21.158 INFO ZWAVE: Node 52 is now asleep
2021-03-28T13:45:21.162Z CNTRLR [Node 052] The node is now asleep.
```

AlCalzone, 2021-03-28:
Honestly that looks like something different. Can you please open a new issue including a full logfile?

djaydev, 2021-03-28:
Hi AlCalzone,
No problem I'll open a new issue, especially since this one is closed but it's not something different.  My same nodes (Zwave remotes) on the same zwave networks and the logs are actually the same except the time stamps.

AlCalzone, 2021-03-28:
Well, it's not the same logs:
old: `[Node 052] did not respond after 1/3 attempts. Scheduling next try in 500 ms.`
new: `Failed to execute controller command after 1/3 attempts. Scheduling next try in 100 ms.`

djaydev, 2021-03-28:
My apologies, I incorrectly implied that the text was word for word exactly the same.  Correction, the wording of the logs have changed, but the meaning remains the same.  

"Hey there's a node that went from asleep to awake/alive, let me update it's settings while we're here. Oh no it's not responding [at this point the remote is back asleep, it's only up to activate the scene] lets schedule another attempt to see if it responds."

While this is happening, I can't control any other node on the network.

edit: just got your message on the new issue, I'll move over there.

PR #1787 (2021-03-08): refactor: do not query all values after cache restart
It is time to part with old habits. With this PR (when done) we no longer query all node values on restart and instead leave it up to the user to query them on demand. Because this might require changes in applications, I'm gonna mark this as breaking (although it is a good break).

As a result, traffic and startup times should be reduced dramatically once all nodes have been interviewed. 

fixes: #1587
fixes: #1884
fixes: #1909
resolves the discussion in https://github.com/zwave-js/node-zwave-js/discussions/1432