zwave-js/zwave-js #1891: GE/Jasco ZW3008 and ZW4006 support double taps but configuration doesn't support

skyw33, 2021-02-24:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is **not** described there.
- [ ] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

Double taps are supported in the hardware for ZW3008 and ZW4006, but the zwavejs json configuration is not set up to handle it. 

**Device information**

Which device(s) is/are affected (make/model)?  GE/Jasco ZW3008 and ZW4006
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ X] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-02-24:
@blhoward2 your turf :)

skyw33, 2021-02-24:
Not sure if this helps, but someone developed a configuration for smartthings for a similar model ZW3010 here:
https://github.com/mwav3/smartthingscode/blob/ccfd245160e67d04897dfc33219a33696507702a/devicetypes/mwav3/ge-46203-dimmer-switch.src/ge-46203-dimmer-switch.groovy

AlCalzone, 2021-02-24:
We have a compat flag to enable the double tap support. It just needs to be put in there.

blhoward2, 2021-02-24:
Yeah, this is a 30 second change. We already support it on 25+ GE models. I'll handle it tonight.

blhoward2, 2021-02-24:
Actually, the problem is just that we don't have files for these yet that I see. What specific models are you referring to? The zw designation is usually the base hardware. I need the product type and product id.

skyw33, 2021-02-24:
26932 / 26933 / ZW3008
and
26931 / ZW4006

Is what I see in zwavejs dashboard

blhoward2, 2021-02-24:
Ok. They just haven't been updated yet to add the compat flag and metadata.

skyw33, 2021-02-24:
More info...
26932 / 26933 / ZW3008
product type: 18765
product ID: 12340

and

26931 / ZW4006
product type: 18765
product ID: 12338

closed 2021-02-25 by AlCalzone

AlCalzone, 2021-03-10:
@blhoward2 got any insight here?

blhoward2, 2021-03-10:
That's how these work. It doesn't add central scene support. Please post the zwave log itself so we can see what's going through.

AlCalzone, 2021-03-10:
That does look like the `event` event gets passed through:
> command_class_name": "Basic", "label": null, "property": "event", "property_name": "event"

blhoward2, 2021-03-10:
Confirmed with the integration developers. As of now at least you need to watch for the event in an automation.

skyw33, 2021-03-11:
I am indeed using an automation to watch for the event, but the problem is that a single press and double press both produce the same "Basic" event.  Maybe my firmware is out-of-date?

blhoward2, 2021-03-11:
That shouldn't happen. It should be sending a single press as a binary switch or a multi-level switch. Please open an issue and fill out the template. We'll need to see the zwave log of the interview, the switch event, and export of the node so we can see what CCs are setup here.

skyw33, 2021-03-11:
https://github.com/zwave-js/node-zwave-js/issues/2042

PR #1900 (2021-02-25): feat: add Double Tap to GE switches
Closes #1891