zwave-js/zwave-js #1903: Door Lock not reporting Alarm Level / Alarm Type

alterbridge86, 2021-02-25:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

I have a Yale YRD216 deadbolt which is not reporting user code used to unlock the door. Previously on OZW these were reported via Alarm Level / Alarm Type. From what I understand however the YRD216 does in fact support the v2 notification CC, so I'm wondering if these values are being ignored when sent.

I'd be fine with the v2 "Access Control" notifications if it included data about which user code unlocked the door, but it does not currently. The only report received is "Keypad unlock operation". 

**Device information**

Node 5 - Yale YRD216

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *OpenZwave 1.6*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Attempt to unlock deadbolt with keypad
2. Observe notification received in zwavejs logs.

**Additional context**

**Debug logging from keypad unlock**
15:23:31.933 SERIAL « 0x010700137f00007ce8                                                 (9 bytes)
15:23:31.935 SERIAL » [ACK]                                                                   (0x06)
15:23:31.937 DRIVER « [REQ] [SendData]
                        callback id:     127
                        transmit status: OK
15:23:31.957 SERIAL « 0x0128000400052298817bee2b6e388921409ff4c89a2687f3851083f24117ac292 (42 bytes)
                      782283ebe088dbd70f4
15:23:31.960 SERIAL » [ACK]                                                                   (0x06)
15:23:31.966 DRIVER « [Node 005] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[NotificationCCReport]
                            notification type:   Access Control
                            notification status: 255
                            notification event:  Keypad unlock operation
                            event parameters:    [object Object]
15:23:31.972 CNTRLR   [Node 005] [~] [Door Lock] currentMode: 0 => 0                    [Endpoint 0]
15:23:31.986 SERIAL « 0x0128000400052298811a17fe5b2f5e60b8374a2e60b33c8c12d601d1d7ae598a2 (42 bytes)
                      777441c9628eae5b76c
15:23:31.990 DRIVER   Dropping message with invalid data (Reason: Nonce 0x27 expired, cannot decode
                      security encapsulated command.):
                      0x0128000400052298811a17fe5b2f5e60b8374a2e60b33c8c12d601d1d7ae598a2777441c9628
                      eae5b76c
15:23:31.992 SERIAL » [ACK]                                                                   (0x06)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6043988/zwavejs_1.log)

**Logfile:**

AlCalzone, 2021-02-25:
Since this issue will involve decoding encrypted communication, I'm going to need your network key. You can send it to me privately via the email address in my profile. Please make sure to reference this issue.

alterbridge86, 2021-02-25:
@AlCalzone Network key sent to the email in your profile. Let me know if anything further is needed.

ehsman, 2021-02-25:
Maybe just for my lock though - I have a Yale YRD256 - I thought the same thing originally due to the log output. As I saw a similar output in my logs. So I built a quick test in nodered - listening for zwave_js_event I saw payload did contain userid at 'payload.event.parameters.userId'. 

But I think it would be awesome if this was stored on an entity for easier reference.

AlCalzone, 2021-02-26:
@alterbridge86 The thing you're not seeing in the log:
```
event parameters: [object Object]
```
contains `userId: 1`. That info is emitted to applications alongside the `"notification"` event, so it should be available in zwavejs2mqtt. It is currently not optimal and we're going to rework this for v7.x, but you should find it under the `notification` topic AFAIK.