zwave-js/zwave-js #1908: Driver.updateLogConfig doesn't work as expected

raman325, 2021-02-26:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

I have attempted to update the log config via a Home Assistant websocket command. When I get the log config, it shows that the change is applied, but I don't see debug statement in the Z-Wave JS server addon logs when the change is applied.

**Device information**

Which device(s) is/are affected (make/model)?  N/A
What are the node IDs? N/A

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [x] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version 2021.3.0b2 (using Z-Wave JS server 0.1.10 which has zwave-js-server 1.1.0 embedded
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

```console
$0.hass.callWS({type: "zwave_js/update_log_config", entry_id: "blah", config: {level: "debug"}}).then(response =>
  console.log("response: ", response));
Promise {<pending>}
VM2316:2 response:  null
$0.hass.callWS({type: "zwave_js/get_log_config", entry_id: "blah"}).then(response =>
  console.log("response: ", response));
Promise {<pending>}
VM2322:2 response:  {enabled: true, level: 5, log_to_file: false, filename: "/usr/lib/node_modules/@zwave-js/server/dist/bin/zwave-248.log", force_console: true}
```

raman325, 2021-02-26:
I tried enabling logToFile, and the action was successful in that getLogConfig returned `log_to_file` as `true`, but I can't find the file at the filepath I specified

AlCalzone, 2021-03-14:
merged in https://github.com/zwave-js/node-zwave-js/pull/2030

PR #1911 (2021-03-08): fix: recreate transports, reconfigure loggers when updating log config
fixes: #1908
closed 2021-03-14 by AlCalzone