zwave-js/zwave-js #1909: Actively requested interviews should be prioritized over restart-from-cache

brett19, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
During the initial setup of my network, I've had to restart zwave-js a number of times.  As part of doing this, zwave-js must reinterview all the nodes in my network as part of the restart-from-cache step.  I have 4 Yale Deadbolt locks which have 200+ door codes each.  This means I need to wait for 4 * 200+ zwave messages to ping-pong to my devices in order to continue with configuration of my devices since I cannot interview new nodes while those 800+ messages are still in the queue (as far as I can tell).

**Describe the solution you'd like**
It would be ideal if things that the user actively requests were prioritized over 'procedural' events such as restart-from-cache, considering in nearly every case a restart-from-cache is going to be receiving unchanged values.  One step further leads to having more details about the values for each command class and their likelihood of change through a restart, for instance a dimmers current value is likely change, but my door codes are very unlikely to change.  In large networks this would make a substantial difference with regards to the initial user experience when setting things up.

**Describe alternatives you've considered**
Waiting... but it takes nearly an hour to complete the full interview of all my nodes such that I am able to add new nodes.

brett19, 2021-02-26:
Just to add, a confounding factor here seems to be related to my battery powered devices.  From what I can tell in the logging, when zwavejs encounters a batter powered device which is asleep, it sleeps for 500ms in between retries to contact the device.  However during this 500ms, the network seems unable to dispatch additional commands (ie: It wont attempt to send messages to other devices that have queued requests during this wait period).

AlCalzone, 2021-02-26:
I agree, and the query after restart won't happen anymore in v7.0.0:
https://github.com/zwave-js/node-zwave-js/pull/1787

> A related interesting thing I've noticed is that after restarting, all my locks loose their meta-data (device manufacturer, etc...) in the UI until the full network has completed it's 'RestartFromCache'

Which application are you using? It is doing something wrong, because all nodes that have been interviewed previously are marked as **ready** immediately after restart and all the metadata and cached values are there.

>  I'm not sure if a failed deserialization somehow stops the interview process

It doesn't, except if it is a response that we need and requested.

brett19, 2021-02-27:
> I agree, and the query after restart won't happen anymore in v7.0.0:
Awesome!

> Which application are you using? It is doing something wrong, because all nodes that have been interviewed previously are marked as ready immediately after restart and all the metadata and cached values are there.
zwavejs2mqtt

> It doesn't, except if it is a response that we need and requested.
In hindsight, I might have accidentally factory reset the device...  So we are just going to promptly forget I brought this up.

AlCalzone, 2021-03-14:
obsolete with #2030

PR #1787 (2021-03-08): refactor: do not query all values after cache restart
It is time to part with old habits. With this PR (when done) we no longer query all node values on restart and instead leave it up to the user to query them on demand. Because this might require changes in applications, I'm gonna mark this as breaking (although it is a good break).

As a result, traffic and startup times should be reduced dramatically once all nodes have been interviewed. 

fixes: #1587
fixes: #1884
fixes: #1909
resolves the discussion in https://github.com/zwave-js/node-zwave-js/discussions/1432
closed 2021-03-14 by AlCalzone