zwave-js/zwave-js #1912: Server crashes with supportedCCs is not iterable error

amccardie, 2021-02-27:
I have a brand new install of the zwave-js-server application, manually built from github. It runs for 30-60 seconds and crashes with a "supportedCCs is not iterable" error. This error seems to be caused by node 3 which is an old Intermatic CA3750 dual switch module. If I disconnect power from the module, the application no longer crashes.

**Hardware involved:**
- Aeotec Z-Stick Gen5
- Intermatic CA3750, node id 3

**Error message:**
```
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
TypeError: supportedCCs is not iterable
    at new Endpoint (/srv/zwave-js-server/node_modules/zwave-js/src/lib/node/Endpoint.ts:49:21)
    at ZWaveNode.getEndpoint (/srv/zwave-js-server/node_modules/zwave-js/src/lib/node/Node.ts:841:5)
    at ZWaveNode.interviewCCs (/srv/zwave-js-server/node_modules/zwave-js/src/lib/node/Node.ts:1433:26)
    at ZWaveNode.interview (/srv/zwave-js-server/node_modules/zwave-js/src/lib/node/Node.ts:1016:8)
    at Driver.interviewNode (/srv/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:895:10)
```
[zwave-6345.log](https://github.com/zwave-js/node-zwave-js/files/6053584/zwave-6345.log)
[logs.zip](https://github.com/zwave-js/node-zwave-js/files/6053586/logs.zip)

LunkwillAndFook, 2021-03-02:
I am also running into this error with a Nortek HUSBZB-1 Zigbee & Z-Wave Plus interface and an Intermatic CA-3750.

AlCalzone, 2021-03-02:
This has to do with how `Multichannel V1 CC` saves its information. Something's wrong here.

LunkwillAndFook, 2021-03-30:
@AlCalzone this issue may be fix-fail. From Zwave2MQTT logs...

2021-03-30 19:17:44.620 ERROR ZWAVE-SERVER: Unexpected error supportedCCs is not iterable
TypeError: supportedCCs is not iterable
at new Endpoint (/opt/node_modules/zwave-js/src/lib/node/Endpoint.ts:56:21)
at ZWaveNode.getEndpoint (/opt/node_modules/zwave-js/src/lib/node/Node.ts:879:5)
at ZWaveNode.getAllEndpoints (/opt/node_modules/zwave-js/src/lib/node/Node.ts:899:19)
at Object.dumpNode (/opt/node_modules/@zwave-js/server/dist/lib/state.js:104:36)
at /opt/node_modules/@zwave-js/server/dist/lib/state.js:214:72
at Function.from ()
at Object.dumpState (/opt/node_modules/@zwave-js/server/dist/lib/state.js:214:22)
at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:67:36)
at WebSocket. (/opt/node_modules/@zwave-js/server/dist/lib/server.js:38:45)
at WebSocket.emit (events.js:315:20)

I did open an issue with HomeAssistant core in case the root cause is on their side... https://github.com/home-assistant/core/issues/48529

amccardie, 2021-03-31:
FWIW, my particular crash issue appears to be fixed in the latest couple of versions of zwave-js. Also, in my case the crash occurred when I ran the zwave-js-server as a standalone service without Home Assistant in the loop at all.

I still can't control my CA-3750, but that's a different problem (issue #2073)

LunkwillAndFook, 2021-03-31:
@amccardie  Thank you for adding info. I'm not sure where this belongs at this point. I had initially avoided opening an issue on the HA side because I saw this was already open here and I didn't want to have folks duplicating effort. I am currently in a position though where none of my Z-Wave devices work due to the issue so this is quickly becoming a show-stopper for me and I want to get some visibility around it.

AlCalzone, 2021-03-31:
@LunkwillAndFook can you please share your cache files `<homeid>.json` and `<homeid>.values.jsonl`?

LunkwillAndFook, 2021-03-31:
> @LunkwillAndFook can you please share your cache files `<homeid>.json` and `<homeid>.values.jsonl`?

@AlCalzone I have emailed the requested files to info@zwave-js.io. Please let me know if I can be of further assistance.

AlCalzone, 2021-04-05:
Unfortunately, the fix in #1977 causes problems when the old value is already in the cache.
https://github.com/zwave-js/node-zwave-js/pull/2242 adds a workaround - until then, clearing the cache or re-interviewing the affected devices (if they can be identified) is the best option.

closed 2021-04-05 by AlCalzone

LunkwillAndFook, 2021-04-08:
I was able to confirm the workaround. I ended up clearing both files and re-interviewing the entire network and that corrected the problem. The Intermatic CA3750s still do not work but I believe that is being tracked by another issue. Thank you again for getting this working.

PR #2242 (2021-04-05): fix: add check for outdated supportedCCs cache value
fixes: #1912