zwave-js/zwave-js #1916: GoControl remote wireless switch WA00z-1 produces "KeyReleased" event prematurely

kklicker777, 2021-02-27:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here. (A developer referred me here from HomeAssistant, marcelveldt)
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

When holding down and releasing the button on this switch, a "KeyHeldDown" event is produced and then an immediate "KeyReleased" event is produced, followed by a second "KeyReleased" event when I actually release the button. There seems to be an extra "KeyReleased" event produced which makes holding down the button pointless. This is different behaviour then the previous zwave 1.4 implementation from Home Assistant.

**Device information**

Which device(s) is/are affected (make/model)? - GoControl remote wireless switch WA00z-1
What are the node IDs?  6 and 10

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *(specify application with versions)* HomeAssistant using zwave 1.4 implementation
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version 2021.2.3
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Using the developer tool in Home Assistant, I first listen for zwave_js_event.
2. Next, I hold down the button (top or bottom).
3. I observe 2 events: initial "KeyHeldDown" event, then immediately a "KeyReleased" event before I release the button.
4. When I release my finger, I observe another "KeyReleased" event.

**Additional context**

Add any other context about the problem here.

**Logfile:**
zwave_js_dump.jsonl
[zwave_js_dump.jsonl.zip](https://github.com/zwave-js/node-zwave-js/files/6054950/zwave_js_dump.jsonl.zip)

AlCalzone, 2021-02-28:
I really need the zwave-js log to figure out what's happening. If you cannot do that in HA, please use zwavejs2mqtt.

kklicker777, 2021-03-01:
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6063715/zwavejs2mqtt-store.zip)
Hi, so I finally was able to get the zwavejs2mqtt working, and I got the log.  The node I am looking at is node 10.  You will see it toward the bottom of the log when I hold down the button and release.  Please let me know if you need any more info.

Thanks.

AlCalzone, 2021-03-01:
Alright, I think I found what's happening. From the specs:

> A controller SHOULD apply an adaptive approach based on the reception of the Key Released
Notification. Initially, the controller SHOULD time out if not receiving any Key Held Down Notification
refresh after 400ms and consider this to be a Key Up Notification.

Your device takes ~850 ms between the key held down and the key released notification, so the driver forces a keyup. Apparently, we're not doing this part correctly:

> If, however, the controller subsequently receives a Key Released Notification, the controller SHOULD consider the sending node to be operating with the Slow Refresh capability enabled.

kklicker777, 2021-03-01:
Fantastic!  I'll look forward to a fix.  Thank you.

kvs, 2021-03-04:
FWIW, this is happening with [Logic Group ZHC5010](https://devices.zwave-js.io/?jumpTo=0x0234:0x0003:0x010a:2.3) as well - not sure if you needed the datapoint

AlCalzone, 2021-03-10:
Are you able to test the linked PR?

kklicker777, 2021-03-11:
It's unclear to me how to test this as I am using zwave-js through Home Assistant.

AlCalzone, 2021-03-11:
That makes it harder - if you are using zwavejs2mqtt, you can just [build a custom Docker container](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds)

kklicker777, 2021-03-17:
I have attempted the instructions you have linked several times on 3 different computers.  The only one I was successful in building was on my Mac, only to find that I cannot link to my USB device through the Docker container.  Most recently, I have attempted it on my Pi 4.  The build takes over 10 minutes, and invariably crashes somewhere toward the end and I am forced to reboot the Pi as it becomes inaccessible.  I am still at a loss for how to test this, I'm afraid.

AlCalzone, 2021-03-17:
Guess I'll have to release it as part of 7.0.0 and you can report back once HA has caught up.

I gotta say this whole slew of docker containers makes it much harder that just cloning the repo and running from source 🤷🏻‍♂️

kklicker777, 2021-03-17:
Yes, if you can, please release it and I'll test it when it is released in HA.  I'll continue to investigate why this is crashing on my end, since I'll probably need a test environment for future issues.  Thanks.

closed 2021-03-18 by AlCalzone

kklicker777, 2021-03-27:
I just updated the zwave-js version in HA to 7.0.0, and I can report that this works great now.  Thanks again for your effort on this!

AlCalzone, 2021-03-27:
Thanks for reporting back!

PR #2033 (2021-03-18): fix(cc): add fallback to slowRefresh behavior for Central Scene V2
fixes: #1916