zwave-js/zwave-js #1923: Including a battery device requires at least one manual wakeup

kpine, 2021-03-01:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

Including a battery device seems to stall at stage ProtocolInfo. The only way to complete the interview is a manual wake up. For most devices I expect to be able to include battery devices without a second wake up (minus some exceptional cases).

**Device information**

Which device(s) is/are affected (make/model)?  Many, but for this issue, a Vision ZD2102-5
What are the node IDs? Node 20

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *(specify application with versions)* OpenZWave
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Start inclusion of a battery device, non-securely
2. Press the button on the device to start inclusion
3. Wait for interview stage to get to ProtocolInfo, watch logs and see it stall
4. Manually wake up the device
5. Watch the interview complete

**Additional context**

I have noticed, and there's been some mentions on HA Discord, that it's been difficult to include battery devices. I have been moving nodes from an old OZW network to a Z-Wave JS based network. I don't believe I've seen a single battery device (no exaggeration) successfully include without requiring a manual wake up from the device. The interview seems to always end at ProtocolInfo, and after waiting I wake up the device and the interview completes very quickly.

In OZW by comparison, adding a battery node has never required me to perform any such wake ups, even the notorious Z-COMBO alarms (for _me_ at least). The battery devices always add without any manual intervention.

I tested the inclusion of a battery door sensor with zwave-js 6.5.0 and OZW. There's nothing particularly unique about this device (Vision ZD2102-5) and any of the others I've added. The manual states that when including it will stay awake for 15 seconds. Zwave-js seems to give up immediately and only completes after a manual wake up. I've attached two logs, one is OZW inclusion and the other is zwave-js inclusion.

**Logfile:**
[inclusion_ozw.log](https://github.com/zwave-js/node-zwave-js/files/6058652/inclusion_ozw.log)
[inclusion_zjs.log](https://github.com/zwave-js/node-zwave-js/files/6058653/inclusion_zjs.log)

In OZW the notable timeline is:

1. 22:32:06.219398883 - ADD_NODE_STATUS_NODE_FOUND message is received, inclusion starts
2. 22:32:28.441782096 - After receiving the last Configuration report, the interview is considered to be complete (query stage Complete)
3. 22:32:38.052427583 - OZW tries to send a Wake Up No More Information command but its already asleep

The device is completely usable and all information known after this process. The actual inclusion process seems to be about 22 seconds (step 1 to 2).

In Z-Wave JS the notable timeline is:

1. 14:36:39.602 - Add Node To Network message handled
2. 14:36:40.583 - Protocol Info is received
3. 14:36:40.584 - The node is marked as asleep. It then says it's querying node info, but nothing happens.
4. 14:36:52.313 - A wake up notification message is received, after I manually woke up the device
5. 14:37:33.499 - The interview is completed and zwave-js tells the device to sleep

The time between steps 1 and 3 is barely even one second. Why does it think the node went to sleep after only one second? Did it actually sleep (seems suspect)? I used the exact same physical process to include the device between both OZW and zwave-js.

Once the device wakes up at step 4, the interview process takes about 41 seconds. This still seems like quite a long time (nearly double OZW). I didn't compare the logs in detail, so I can accept the interview takes longer because maybe zwave-js is asking the device for more information (to be spec compliant?) The point of this issue is regarding the seemingly necessary manual wake up.

michaelbaldy, 2021-03-01:
Is this not the expected behavior?  My battery device acted the same way.

AlCalzone, 2021-03-01:
>     3\. 14:36:40.584 - The node is marked as asleep

That step sounds like it is unintended.

PR #1986 (2021-03-03): fix: do not immediately mark newly included nodes as asleep
With this PR, we now mark newly included nodes as alive until we know if they can sleep or not.

Fixes: #1923
closed 2021-03-03 by AlCalzone