zwave-js/zwave-js #1925: [bug] Unable to set config parameters on Dome DMS01 Siren

PhAzE-Variance, 2021-03-01:
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 1.0.0
zwavejs version: 6.5.0

### Describe the bug

When trying to set any of the config parameters for the Dome DMS01 Siren, the settings instantly revert in the GUI and the changes were not made on the device. Due to this, most of the functionality of the device is locked out.

### To Reproduce

Steps to reproduce the behavior:

1. Go to 'zwavejs2mqtt interface'
2. Navigate to the device
3. Expand the Configuration section
4. Adjust any of the configuration settings
5. Click the save arrow
6. Message shows that API call was successful
7. Message shows that Value Updated
8. Configuration setting has been reverted to original state and device was not updated

### Expected behavior

When the configuration parameter is changed, the device should be updated accordingly.

### Additional context

The siren can be triggered and stopped from the binary switch as expected, the issue seems to be only with setting config parameters. Node 19 is the node in question.

Updated log set to Silly:
[zwavejs_1.log](https://github.com/zwave-js/zwavejs2mqtt/files/6052889/zwavejs_1.log)

PhAzE-Variance, 2021-03-04:
I looked up the manual for this device and all valuesizes should be 1, so I'm thinking it's the config that is incorrect.

AlCalzone, 2021-03-04:
Thanks for testing, this should be fixed by #1992

PR #1992 (2021-03-05): fix(config): correct DMS01 configuration file
fixes: #1925
closed 2021-03-05 by AlCalzone