zwave-js/zwave-js #1928: [HASS] Wink D/W Sensors not recognized

jack-apple-sr, 2021-03-01:
### Version

Build/Run method

Docker **[X]**  on Raspberry Pi 3+ /  Buster 5.10.11-v7+ #1399   w/ Aeotec Gen5+
Git Clone + npm run install + npm run build + npm start
Pkg

zwavejs2mqtt version: 1.2.2
zwavejs version: 6.4.0

### Describe the bug

Migrated from Hass OZW to zwavejs2mqtt on Docker/Pi Bluster two days ago and zwavejs2mqtt has successfully recognized only 2 (nodes 21 & 37) of 26 Wink Door/Window Sensors (https://products.z-wavealliance.org/products/2494).

Tried healing, waking, inclusion/exclusion,  taking my aeotec Gen5+ stick to Node 61 and including,  to no avail...

Files included in zips:
    Node_21.json   (fully recognized)
    Node_61.json   (same zwave device, unrecognized)
    zwavejs2mqtt.log 
    zwavejs_1.log

[node_21.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6057853/node_21.zip)
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6057797/zwavejs2mqtt-store.zip)

Do I just need to wait longer?
Thanks...

AlCalzone, 2021-03-01:
I checked what happens with node 61. The first message should be queued until the device wakes up. For some reason, it doesn't and this causes the interview to fail.
@jack-apple-sr can you share the `<homeid>.json` from your store aswell?

jack-apple-sr, 2021-03-01:
Thanks for looking into this.

Server config uploaded...:
[zwavejs_1.zip](https://github.com/zwave-js/node-zwave-js/files/6063202/zwavejs_1.zip)

btw-   I failed node_61,  manually factory reset it and it successfully re-included as node_83

jack-apple-sr, 2021-03-01:
crud,  wrong file... sorry-   here'tis:
[ca75ee1a.zip](https://github.com/zwave-js/node-zwave-js/files/6063236/ca75ee1a.zip)

AlCalzone, 2021-03-05:
> it successfully re-included as node_83

So it works now? Can't tell what was the problem with 61 if I don't have the cache where 61 is included 🤷🏻‍♂️

jack-apple-sr, 2021-03-05:
My apologies - didn't think ahead...   physically resetting 61 allowed it to be included as new node 83.  
I still have the same issue with my 24 other Wink D/W sensors.   Those failures should also be in the logs.
Uploading another failing node json.     Let me know if you need anything else...
[node_39.zip](https://github.com/zwave-js/node-zwave-js/files/6093418/node_39.zip)
Thanks.

AlCalzone, 2021-03-06:
That looks like what I fixed in https://github.com/zwave-js/node-zwave-js/pull/1990.
Workaround: re-interview each of the devices

AlCalzone, 2021-03-06:
Duplicate of #1797

closed 2021-03-10 by AlCalzone