zwave-js/zwave-js #1967: Missing association group for YRD216

bburan, 2021-03-01:
https://github.com/zwave-js/node-zwave-js/blob/959689b213b692ce67d03b541b8a18b5059ba833/packages/config/config/devices/0x0129/yrd216.json#L19

This lock is missing the Lifeline association group (see https://products.z-wavealliance.org/products/1971?selectedFrequencyId=-1), I think. It should have:

```
	"associations": {
		"1": {
			"label": "Alarm Reports",
			"description": "Alarm reports are sent out to all devices in the association group",
			"maxNodes": 5,
			"isLifeline": true
		}
	},
```

AlCalzone, 2021-03-01:
The device supports Z-Wave+, so the only thing you would be missing out on is the metadata. Group 1 must exist for Z-Wave+ devices and must be the lifeline group.
And zwave-js defaults to group 1 for Z-Wave+ devices.

Is there a problem (like a terrible label etc) which adding that to the config file would solve?

bburan, 2021-03-01:
Closing since it looks like this is a more relevant place for discussion: https://github.com/zwave-js/node-zwave-js/issues/1713

AlCalzone, 2021-03-01:
@bburan #1713 is about devices that use V1 alarms. Your's uses V2+ Notifications. Please raise a new issue and post zwave-js logs with loglevel debug, so I can see if something is missing.