zwave-js/zwave-js #1973: Make debug log easier to structurally ingest into a logging system like Logstash/Elasticsearch

robinsmidsrod, 2021-03-01:
I'd really like to ingest the debug log into my ELK (Elasticsearch, Logstash & Kibana) stack so that I can more easily analyze it for anomalies and graph how often issues occur in my Z-Wave mesh. For this to be possible, the information in the logs must be separated into more atomic units so that it can be parsed in a simple way.

For the logs to be easy to ingest with a log parser they must be **structured** in a way that makes it possible to _know where one value starts and ends and the next value begins_. In the current log this is mostly implemented by using indenting with space. It is also _vital to know what the value represents_. This is currently implemented with fixed line character offsets, and the message part always start at a certain offset of the line. Unfortunately this only means the message key begins there, the message **value** is still just opaque text with no inherent structure. Some long sentences are also split over multiple lines, although they technically don't need to be. This seems to have been done to keep the log max line length to a fixed number. I'm assuming this is to accommodate easier human consumption on a screen with limited width.

Introducing a structured logging format would still make it possible to read for humans and also ingestable by automatic tools. It would also make it easier to search using `grep(1)` and filter with tools like `sed(1)`. My suggestion is to use the [logfmt](https://www.brandur.org/logfmt) format championed by Heroku. There is even an [npm package available](https://www.npmjs.com/package/logfmt) for it.

Here are some examples of how log data from my own log could be represented using logfmt fully structured:

```
20:23:02.101 SERIAL » 0x01030015e9                                                         (5 bytes)
ts=2021-03-01T20:23:02.101Z c=SERIAL dir=tx msg="Transmit payload" payload="0x01030015e9" length=5

20:23:02.101 DRIVER » [REQ] [GetControllerVersion]
ts=2021-03-01T20:23:02.101Z c=DRIVER dir=tx msg="GetControllerVersion" cmd="REQ"

20:23:02.102 SERIAL « [ACK]                                                                   (0x06)
ts=2021-03-01T20:23:02.102Z c=SERIAL dir=rx msg="Payload received" payload="0x06" length=1 cmd="ACK"

20:23:02.103 SERIAL « 0x011001155a2d5761766520332e3935000199                              (18 bytes)
ts=2021-03-01T20:23:02.103Z c=SERIAL dir=rx msg="Payload received" payload="0x011001155a2d5761766520332e3935000199" length=18

20:23:02.104 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520332e39350001
ts=2021-03-01T20:23:02.104Z c=DRIVER dir=rx msg="GetControllerVersion" payload="0x5a2d5761766520332e39350001" cmd="RES"

20:23:02.105 CNTRLR   received version info:
                        controller type: Static Controller
                        library version: Z-Wave 3.95
ts=2021-03-01T20:23:02.105Z c=CNTRLR msg="received version info" controller_type="Static Controller" library_version="Z-Wave 3.95"

20:23:02.122 CNTRLR   received API capabilities:
                        serial API version:  1.0
                        manufacturer ID:     0x86
                        product type:        0x01
                        product ID:          0x5a
                        supported functions:
                        · GetSerialApiInitData (0x02)
                        · FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION (0x03)
                        · ApplicationCommand (0x04)
                        · GetControllerCapabilities (0x05)
ts=2021-03-01T20:23:02.122Z c=CNTRLR msg="received API capabilities" serial_api_version=1.0 manufacturer_id=0x86 product_type=0x01 product_id=0x5a
ts=2021-03-01T20:23:02.122Z c=CNTRLR msg="received API capabilities" supported_function_id=0x02 supported_function_name=GetSerialApiInitData
ts=2021-03-01T20:23:02.122Z c=CNTRLR msg="received API capabilities" supported_function_id=0x03 supported_function_name=FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION"
ts=2021-03-01T20:23:02.122Z c=CNTRLR msg="received API capabilities" supported_function_id=0x04 supported_function_name=ApplicationCommand
ts=2021-03-01T20:23:02.122Z c=CNTRLR msg="received API capabilities" supported_function_id=0x05 supported_function_name=GetControllerCapabilities

20:23:02.286 DRIVER   Cache file for homeId 0xdc364e03 found, attempting to restore the network from
                       cache...
ts=2021-03-01T20:23:02.286Z c=DRIVER msg="Cache file for home identifier found, attempting to restore the network from cache" home_id=0xdc364e03

20:23:02.275 CNTRLR   received node information:
                        controller role:            primary
                        controller is a SUC:        true
                        controller is a slave:      false
                        controller supports timers: false
                        nodes in the network:       1, 3, 15, 17, 18, 22, 23, 24, 26, 30, 31
ts=2021-03-01T20:23:02.275Z c=CNTRLR msg="received node information" controller_role=primary controller_is_a_suc=true controller_is_a_slave=false controller_supports_timers=false nodes_in_network="1,3,15,17,18,22,23,24,26,30,31"

20:23:02.287 CNTRLR   [Node 001] trying to load device config
ts=2021-03-01T20:23:02.287Z c=CNTRLR node_id=1 msg="trying to load device config"

20:23:02.330 CNTRLR   [Node 001] [Manufacturer Specific] manufacturerId: metadata updated
20:23:02.330 CNTRLR   [Node 001] [~] [Manufacturer Specific] manufacturerId: 134 => 134
ts=2021-03-01T20:23:02.330Z c=CNTRLR node_id=1 msg="metadata updated" cc="Manufacturer Specific" key=manufacturerId value=134 old_value=134
```

If going full-blown with structured logging is too much to bite over in one go, you could start simple by encoding the existing message with the use of indent, but add some structure to make it easier to parse. Under you can see an example of this half-way approach shown on the same messages:

```
20:23:02.101 SERIAL » 0x01030015e9                                                         (5 bytes)
[... 37 lines cut ...]
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=5/10 msg="  product ID:          0x5a"
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=6/10 msg="  supported functions:"
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=7/10 msg="  · GetSerialApiInitData (0x02)"
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=8/10 msg="  · FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION (0x03)"
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=9/10 msg="  · ApplicationCommand (0x04)"
ts=2021-03-01T20:23:02.122Z c=CNTRLR part=10/10 msg="  · GetControllerCapabilities (0x05)"

20:23:02.286 DRIVER   Cache file for homeId 0xdc364e03 found, attempting to restore the network from
                       cache...
ts=2021-03-01T20:23:02.286Z c=DRIVER msg="Cache file for homeId 0xdc364e03 found found, attempting to restore the network from cache"

20:23:02.287 CNTRLR   [Node 001] trying to load device config
ts=2021-03-01T20:23:02.287Z c=CNTRLR node_id=1 msg="trying to load device config"

20:23:02.330 CNTRLR   [Node 001] [Manufacturer Specific] manufacturerId: metadata updated
20:23:02.330 CNTRLR   [Node 001] [~] [Manufacturer Specific] manufacturerId: 134 => 134
ts=2021-03-01T20:23:02.330Z c=CNTRLR node_id=1 part=1/2 msg="[Manufacturer Specific] manufacturerId: metadata updated"
ts=2021-03-01T20:23:02.330Z c=CNTRLR node_id=1 part=2/2 msg="[~] [Manufacturer Specific] manufacturerId: 134 => 134"
```

The lines that start with `ts=` are logfmt format. With this halfway approach you could most likely convert existing logging to use this new format with not too much effort, and then you could slowly move over to the fully structured format as time permits on individual pieces of code as you work on them.

I don't know what all the log information represents, so I've taken liberties with naming the keys in the logfmt syntax. The only one that is commonly referred to in the logfmt documentation is `msg`, and it should be kept named as is. I've also seen `ts` being used in lots of places to represents the timestamp, other then that it is up to you if you'd prefer to go with short 1, 2 or 3-character keys or fully named for what they represent. I think the logfmt format generally use underscores, but if everything is already in camelCase I don't see a reason why you can't use that. I tried to come up with ways to represent both key/value maps and arrays in the log syntax, as that can be a point of challenge when you're trying to dump complex structures to simple lines.

I hope this hasn't been too much of a rambling mess and that it makes sense to you. I hope you will consider it.

AlCalzone, 2021-03-02:
My thoughts:
Full-blown structured logging is too much at the moment. I've certain locations where the base exists for the messages (e.g. `toLogEntry` in CCs), but that needs to be expanded on, for example in lists.

The latter part should just be a matter of a custom winston transport. Most of that structure already exists as JSON.

AlCalzone, 2021-03-02:
So, I was interested in testing out #1981 and built a little `LogfmtTransport` on that basis:

```ts
import type { ZWaveLogInfo } from "@zwave-js/core";
import { padStart } from "alcalzone-shared/strings";
import Transport from "winston-transport";

export class LogfmtTransport extends Transport {
	public constructor() {
		super({
			level: "silly",
		});
	}

	private logOnce(info: ZWaveLogInfo & { message: string; part?: string }) {
		const dir =
			info.direction === "  "
				? ""
				: info.direction === "« "
				? "dir=RX"
				: "dir=TX";
		const tags = info.primaryTags ? `tags="${info.primaryTags}"` : "";
		const tags2 = info.secondaryTags ? `info="${info.secondaryTags}"` : "";
		const part = info.part ? `part=${info.part}` : "";
		const msg = [
			`ts=${info.timestamp}`,
			`lbl=${info.label}`,
			dir,
			tags,
			part,
			!!info.message ? `msg="${info.message.replace(/"/g, "'")}"` : "",
			tags2,
		]
			.filter((blob) => !!blob)
			.join(" ");
		console.log(msg);
	}

	public log(info: ZWaveLogInfo, next: () => void): any {
		if (typeof info.message === "string" && info.message.includes("\n")) {
			info.message = info.message.split("\n");
		}

		if (typeof info.message === "string") {
			this.logOnce(info as any);
		} else {
			let part = 0;
			const partLength = 1 + Math.floor(Math.log10(info.message.length));
			for (const msg of info.message) {
				part++;
				this.logOnce({
					...info,
					message: msg,
					part: `${padStart(
						part.toString(),
						partLength,
						"0",
					)}/${padStart(
						info.message.length.toString(),
						partLength,
						"0",
					)}`,
				});
			}
		}
		next();
	}
}
```

Put that into the driver options:
```ts
const driver = new Driver("COM4", {
	// ...
	logConfig: {
		transports: [new LogfmtTransport()],
	},
})
//...
```

And out comes this log:
```
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=1/8
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=2/8 msg="███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=3/8 msg="╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=4/8 msg="  ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=5/8 msg=" ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=6/8 msg="███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=7/8 msg="╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER part=8/8
ts=2021-03-02T17:06:33.589Z lbl=DRIVER msg="version 6.6.0"
ts=2021-03-02T17:06:33.589Z lbl=DRIVER
ts=2021-03-02T17:06:33.589Z lbl=DRIVER msg="starting driver..."
ts=2021-03-02T17:06:33.589Z lbl=DRIVER msg="opening serial port COM4"
ts=2021-03-02T17:06:33.605Z lbl=DRIVER msg="serial port opened"
ts=2021-03-02T17:06:33.605Z lbl=SERIAL dir=TX tags="[NAK]" info="(0x15)"
ts=2021-03-02T17:06:35.126Z lbl=DRIVER msg="loading configuration..."
ts=2021-03-02T17:06:35.447Z lbl=DRIVER msg="beginning interview..."
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=1/2 msg="added request handler for AddNodeToNetwork (0x4a)..."
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=2/2 msg="1 registered"
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=1/2 msg="added request handler for RemoveNodeFromNetwork (0x4b)..."
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=2/2 msg="1 registered"
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=1/2 msg="added request handler for ReplaceFailedNode (0x63)..."
ts=2021-03-02T17:06:35.448Z lbl=DRIVER part=2/2 msg="1 registered"
ts=2021-03-02T17:06:35.449Z lbl=CNTRLR msg="beginning interview..."
ts=2021-03-02T17:06:35.449Z lbl=CNTRLR msg="querying version info..."
ts=2021-03-02T17:06:35.497Z lbl=SERIAL dir=TX msg="0x01030015e9" info="(5 bytes)"
ts=2021-03-02T17:06:35.498Z lbl=DRIVER dir=TX part=1/1 msg="[REQ] [GetControllerVersion]"
ts=2021-03-02T17:06:35.500Z lbl=SERIAL dir=RX tags="[ACK]" info="(0x06)"
ts=2021-03-02T17:06:35.501Z lbl=SERIAL dir=RX msg="0x011001155a2d5761766520362e3032000192" info="(18 bytes)"
ts=2021-03-02T17:06:35.502Z lbl=SERIAL dir=TX tags="[ACK]" info="(0x06)"
ts=2021-03-02T17:06:35.502Z lbl=DRIVER dir=RX part=1/2 msg="[RES] [GetControllerVersion]"
ts=2021-03-02T17:06:35.502Z lbl=DRIVER dir=RX part=2/2 msg="  payload: 0x5a2d5761766520362e30320001"
ts=2021-03-02T17:06:35.505Z lbl=CNTRLR part=1/3 msg="received version info:"
ts=2021-03-02T17:06:35.505Z lbl=CNTRLR part=2/3 msg="  controller type: Static Controller"
ts=2021-03-02T17:06:35.505Z lbl=CNTRLR part=3/3 msg="  library version: Z-Wave 6.02"
```

robinsmidsrod, 2021-03-03:
@AlCalzone That looks like a really neat result! Very close or exactly like my halfway suggestion. If I might just suggest a few more additions:

1. The severity level should really be a part of the log line (either `severity=info` or `lvl=4` depending on how verbose you'd prefer to go). This makes it so much easier to always log at max verbosity and let the ingest software deal with filtering during analysis. The only reason to not log at maximum is that it generates a lot of data, but that is a segway into log rotation, which we don't need to talk about now.
2. If you want to make it more readable to humans, include the fields that are always present first and always in the same order (ts,lbl,lvl,msg) and then the optional ones (part,dir,tags,info)

AlCalzone, 2021-03-03:
1. 👍🏻
2. `dir` is pretty important to understand the following message and `tags` contain which node a message is from/to, so that needs to go first IMO
3. I didn't want to add more dependencies to the repo for the test. My plan is to shell this out into a separate repo that just contains custom log transports. There we can use `node-logfmt` for correct encoding.

robinsmidsrod, 2021-03-03:
@AlCalzone I'm good with all that and can understand the reasoning for both 2 and 3. This already looks so much better from a log ingestion perspective, while also being much friendlier to grep and still human-readable. Go for it!

AlCalzone, 2021-03-03:
This will be handled in a separate repo once zwave-js@7 has landed:
https://github.com/zwave-js/log-transports

PR #2030 (2021-03-14): chore: prepare version 7
fixes: #1973
closed 2021-03-14 by AlCalzone