zwave-js/zwave-js #1978: [feat]Prevent data loss when zwavejs2mqtt restarted during RestartFromCache

NYZack, 2021-03-02:
**Is your feature request related to a problem? Please describe.**
If zwavejs2mqtt is restarted while a device is in "RestartFromCache" status, all data for the device are lost after the restart (the device is listed as "Unknown," all values and metadata are lost, and the status is changed to "ProtocolInfo").  The device data only repopulate once the device interrogation is completed, which can be quite a while for a battery-powered device.  Creates some grief in Home Assistant - all of the related device entities are listed as "Unknown".

**Describe the solution you'd like**
If a restart occurs while a device is in "RestartFromCache" status, all data and metadata should carry through the restart (and it should remain in "RestartFromCache" status).  I suspect this is a zwavejs issue, but not 100% sure about this.

AlCalzone, 2021-03-02:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

This log should include a restart where at least one device was in RestartFromCache stage.

AlCalzone, 2021-03-02:
Also please share your cache file `<homeid>.json` after you've shutdown the container (and before starting it back up) while a device was in RestartFromCache stage.

NYZack, 2021-03-02:
So this is strange.  It seems to be quite variable.  When I tried this the first time just now to create the log, I had exactly the behavior I described above (actually, interview status shows up as "None" rather than "ProtocolInfo"; I was wrong about that),  and the <homeid>.json file was empty, which I guess explains the behavior.  But on trying a few more times, I couldn't reproduce it.  I'll have to mess around with it some more.

AlCalzone, 2021-03-02:
Same as in https://github.com/zwave-js/node-zwave-js/issues/1877 it seems.

kpine, 2021-03-04:
I just encountered this for the first time. I shutoff zwavejs2mqtt (docker) sometime yesterday, and restarted today and it reported the syntax error. Some other things I noticed besides the battery devices being unknown:

- All the mains powered devices came back, but the CC versions were all 0. It required a re-interview to get the proper versions
- My garage door opener suddenly reported having a battery CC, which it has never shown before. It also did not report having a binary switch any longer. After restoring from a backup the binary switch was back and battery no longer detected.

Unfortunately I didn't keep any copies of the cache files before or after I noticed this, as I wasn't expecting this to happen. I'll keep an eye on it from now. I'm attaching my log file, not sure if it'll be any use. It does show the Node 12 (garage door opener) trying to query for the Battery CC.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6080459/zwavejs_1.log)