zwave-js/zwave-js #1982: GE 14287 Fan Switch Multilevel switch command not working

fletchan, 2021-03-02:
**Describe the bug**
GE 14287 Fan switch is not responding to multilevel switch commands sent from zwavejs2mqtt control panel. Sending a Basic command does work.

**Device information**
GE 14287 Fan Switch
Firmware: 5.24
NodeId: 8

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, Dont know if this have ever working in zwave-js however this device worked in OpenZwave
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
App version: 2.1.0 
Zwavejs Version 6.6.0
Zwavejs-server version 1.1.0
- [x] `zwavejs2mqtt` (latest) docker image 
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to zwavejs2mqtt control panel
2. Click on fan switch node and then multilevel switch under values section
3. Set target value to 255 and send cmd
4. Device never turns on and a timeout message appears in the logs

**Additional context**

For comparison I have a working OpenZwave serial command and was able to compare the two.

OpenZwave command: SwitchMultilevelCmd_Set (Node=8): 0x01, 0x0a, 0x00, 0x13, 0x08, 0x03, 0x26, 0x01, 0xff, 0x25, 0x2b, 0x3b

Zwave-js command: 0x01 0e 00 13 08 07 6c 01 9d 03 26 01 ff 25 28 cb

Zwave-js seems to be sending the command as a Supervision get compared to OpenZwave just sending the MultiLevel set command. This specific switch does not support the supervision command as can be seen in interview log file attached and from https://products.z-wavealliance.org/products/2506/classes.

**Logfile:**
[node8_interview.log](https://github.com/zwave-js/node-zwave-js/files/6071025/node8_interview.log)
[multiLevelSet_timeout.log](https://github.com/zwave-js/node-zwave-js/files/6071027/multiLevelSet_timeout.log)

Joeviocoe, 2021-03-02:
GE 14287 Fan switch also known as **GE/Jasco - In-Wall Smart Fan Control (14287 / ZW4002)**

[https://github.com/home-assistant/core/issues/47179](url)

AlCalzone, 2021-03-02:
@Joeviocoe edit `/usr/src/app/node_modules/@zwave-js/config/config/devices/0x0063/zw4002.json`, scroll to the end, edit the `compat` block to this:
```json
    "compat": {
        "treatBasicSetAsEvent": true,
        "commandClasses": {
            "remove": {
                "0x6c": {
                    // The device does not support supervision although it is mandatory
                    "endpoints": "*"
                }
            }
        }
    }
```
Then re-interview the node or restart.

fletchan, 2021-03-02:
This change worked for me

Joeviocoe, 2021-03-02:
Works for me too.  Thanks.

_I forgot the comma, so took a while troubleshooting why interviews failed._

closed 2021-03-03 by AlCalzone

SAABoholic, 2021-03-03:
Sweet ! 
I can also confirm that the fix works.

PR #1983 (2021-03-03): fix(config): remove Supervision support for GE 14287 / ZW4002
fixes: #1982