zwave-js/zwave-js #1988: Performance downgrade in node interviewing after 6.6.0

maivorbim, 2021-03-04:
Hello, I am seeing a performance downgrade in node interviewing after updating to zwavejs 6.6.0. Prior to 6.6.0, the nodes completed the interview in about 30-60 minutes (60 nodes). However, after updating to 6.6.0 I am at over 2 hours with about 25% of the nodes being complete and many of the powered nodes showing as Dead, which has never happened before.

Log appears to be full of NoAck messages:

```
02:10:38.185 DRIVER » [Node 092] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      171
                      └─[NoOperationCC]
02:10:38.187 SERIAL « [ACK]                                                                   (0x06)
02:10:38.193 SERIAL « 0x0104011301e8                                                       (6 bytes)
02:10:38.194 SERIAL » [ACK]                                                                   (0x06)
02:10:38.194 DRIVER « [RES] [SendData]
                        was sent: true
02:10:38.709 SERIAL « 0x01080004005c0284072e                                              (10 bytes)
02:10:38.710 SERIAL » [ACK]                                                                   (0x06)
02:10:38.713 DRIVER « [Node 092] [REQ] [ApplicationCommand]
                      └─[WakeUpCCWakeUpNotification]
02:10:38.714 CNTRLR « [Node 092] received wakeup notification
02:10:51.338 SERIAL « 0x01070013ab01052367                                                 (9 bytes)
02:10:51.338 SERIAL » [ACK]                                                                   (0x06)
02:10:51.340 DRIVER « [REQ] [SendData]
                        callback id:     171
                        transmit status: NoAck
02:10:51.343 CNTRLR   [Node 092] Node 92 did not respond after 1 attempts, it is presumed dead
02:10:51.348 CNTRLR   [Node 092] The node is now dead.
02:10:51.349 SERIAL » 0x010800135c010025ac30                                              (10 bytes)
02:10:51.349 DRIVER » [Node 092] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      172
                      └─[NoOperationCC]
02:10:51.350 SERIAL » 0x01030016ea                                                         (5 bytes)
02:10:51.350 DRIVER » [REQ] [SendDataAbort]
02:10:51.350 CNTRLR   [Node 092] ping failed: Failed to send the command after 1 attempts (Status No
                      Ack)
02:10:51.350 CNTRLR   [Node 092] ping failed: Node 92 did not respond after 1 attempts, it is presum
                      ed dead
02:10:51.350 CNTRLR » [Node 092] querying node info...
02:10:51.351 CNTRLR » [Node 092] pinging the node...
02:10:51.352 CNTRLR   [Node 092] Interview attempt (4/5) failed, node is dead.
02:10:51.359 SERIAL « [ACK]                                                                   (0x06)
02:10:52.362 CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                      n 100 ms.
02:10:52.463 SERIAL » 0x01030016ea                                                         (5 bytes)
02:10:52.464 DRIVER » [REQ] [SendDataAbort]
02:10:52.545 SERIAL « 0x0104011301e8                                                       (6 bytes)
02:10:52.546 SERIAL » [ACK]                                                                   (0x06)
02:10:52.547 DRIVER « [RES] [SendData]
                        was sent: true
02:10:52.549 DRIVER « [RES] [SendData]                                                  [unexpected]
                        was sent: true
02:10:52.549 DRIVER   unexpected response, discarding...
02:10:52.550 DRIVER « [RES] [SendData]                                                  [unexpected]
                        was sent: true
02:10:52.550 DRIVER   unexpected response, discarding...
02:10:52.566 SERIAL « [CAN]                                                                   (0x18)
02:10:52.568 CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                      n 1100 ms.
02:10:52.569 SERIAL « [CAN]                                                                   (0x18)
02:10:53.670 SERIAL » 0x01030016ea                                                         (5 bytes)
02:10:53.671 DRIVER » [REQ] [SendDataAbort]
02:10:53.809 SERIAL « [ACK]                                                                   (0x06)
02:10:53.817 SERIAL » 0x010800135c010025ad31                                              (10 bytes)
02:10:53.817 DRIVER » [Node 092] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      173
                      └─[NoOperationCC]
02:10:53.819 SERIAL « [ACK]                                                                   (0x06)
```

I have attached Zwavejs silly logs. 
[zwavejs_359.log](https://github.com/zwave-js/node-zwave-js/files/6079956/zwavejs_359.log)

maivorbim, 2021-03-04:
As a general observation, it appears that nodes become alive then quickly turn dead. This happens regularly. Except for an unexpected zombie infection of my zwave network, could it be a timeout issue?

AlCalzone, 2021-03-04:
> This happens regularly

This sounds like the responses get lost. Did you change anything physically - like moving mains-powered devices or the controller?

Healing the network could help.

maivorbim, 2021-03-04:
No, nothing was moved. Currently I am waiting 1-2 hours until node interview becomes stuck, trigger a heal network, wait about 30 minutes, stop healing network and then restart the network. Nodes seem to slowly come back online, but it is inconsistent as previously alive nodes sometimes remain dead after a network restart.

AlCalzone, 2021-03-04:
There are some strange things happening, for example Node 2 spamming setpoint changes (look at the timestamps):
```
21:39:31.253 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.255 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 72 => 73        [Endpoint 0]
21:39:31.256 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.256 SERIAL » [ACK]                                                                   (0x06)
21:39:31.258 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:31.388 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.389 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:31.389 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.390 SERIAL » [ACK]                                                                   (0x06)
21:39:31.391 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:31.444 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.446 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:31.446 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.446 SERIAL » [ACK]                                                                   (0x06)
21:39:31.447 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:31.489 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.492 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:31.492 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.493 SERIAL » [ACK]                                                                   (0x06)
21:39:31.493 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:31.699 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.700 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:31.700 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.701 SERIAL » [ACK]                                                                   (0x06)
21:39:31.702 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:31.856 SERIAL « 0x010f0004000209600d0000430301094993                                (17 bytes)
21:39:31.857 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:31.858 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                       1 => 1
21:39:31.858 SERIAL » [ACK]                                                                   (0x06)
21:39:31.859 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
[... 31 lines cut ...]
                       1 => 1
21:39:31.996 SERIAL » [ACK]                                                                   (0x06)
21:39:31.997 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
21:39:33.562 SERIAL « 0x010f0004100209600d0000430301094983                                (17 bytes)
21:39:33.563 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpoint[1]: 73 => 73        [Endpoint 0]
21:39:33.564 CNTRLR   [Node 002] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
[... cut ...]
21:39:33.565 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      0
                        │ destination: 0
                        └─[ThermostatSetpointCCReport]
                            setpoint type: Heating
                            value:         73 °F
```
Then there's node 5 that's not responding half the time - maybe because node 2 is flooding the network...

Did you include the devices with zwave-js or was that a migration?

maivorbim, 2021-03-04:
Mmm, I think you are looking at wrong logs? I do not have node 002 in my network.

AlCalzone, 2021-03-04:
Uh weird. I'll take another look later.

maivorbim, 2021-03-04:
So after two heals and two restarts, I have managed to bring all powered nodes online and complete the interview process. Currently only left with around 30 TRVs (flirs) and battery powered devices to revive.

maivorbim, 2021-03-04:
You are however right that some values are being repeated multiple times, do not know if this is normal or not:

```
17:23:06.289 CNTRLR « [Node 028] received response for battery information:
                      level:                           70
17:23:06.289 CNTRLR   [Node 028] [~] [Battery] interviewComplete: true => tr [Endpoint 0] [internal]
                      ue
17:23:06.301 CNTRLR   [Node 028] MultilevelSwitchCC: doing a partial interview...
17:23:06.301 CNTRLR » [Node 028] requesting current switch state...
17:23:06.303 SERIAL » 0x010900131c0226022557ad                                            (11 bytes)
17:23:06.303 DRIVER » [Node 028] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      87
                      └─[MultilevelSwitchCCGet]
17:23:06.304 SERIAL « [ACK]                                                                   (0x06)
17:23:06.311 SERIAL « 0x0104011301e8                                                       (6 bytes)
17:23:06.311 SERIAL » [ACK]                                                                   (0x06)
17:23:06.311 DRIVER « [RES] [SendData]
                        was sent: true
17:23:06.600 SERIAL « 0x01090004101c0380034638                                            (11 bytes)
17:23:06.601 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:06.605 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:06.606 SERIAL » [ACK]                                                                   (0x06)
17:23:06.607 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:07.051 SERIAL « 0x01090004101c0380034638                                            (11 bytes)
17:23:07.052 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:07.057 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:07.060 SERIAL » [ACK]                                                                   (0x06)
17:23:07.063 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:07.538 SERIAL « 0x01090004001c0380034628                                            (11 bytes)
17:23:07.539 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:07.543 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:07.544 SERIAL » [ACK]                                                                   (0x06)
17:23:07.545 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:07.717 SERIAL « 0x01090004101c0380034638                                            (11 bytes)
17:23:07.719 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:07.725 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:07.729 SERIAL » [ACK]                                                                   (0x06)
17:23:07.731 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:08.574 SERIAL « 0x01090004001c0380034628                                            (11 bytes)
17:23:08.575 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:08.579 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:08.580 SERIAL » [ACK]                                                                   (0x06)
17:23:08.581 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:09.519 SERIAL « 0x01090004001c0380034628                                            (11 bytes)
17:23:09.520 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:09.524 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
17:23:09.525 SERIAL » [ACK]                                                                   (0x06)
17:23:09.526 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  70
                          is low: false
17:23:10.838 SERIAL « 0x01090004001c0380034628                                            (11 bytes)
17:23:10.839 CNTRLR   [Node 028] [~] [Battery] level: 70 => 70                          [Endpoint 0]
17:23:10.843 CNTRLR   [Node 028] [~] [Battery] isLow: false => false                    [Endpoint 0]
[... 184 lines cut ...]
                      : 0 => 0
17:23:33.142 SERIAL » [ACK]                                                                   (0x06)
17:23:33.143 DRIVER « [Node 028] [REQ] [ApplicationCommand]
                      └─[ThermostatSetpointCCReport]
                          setpoint type: Energy Save Heating
                          value:         18 °C
17:23:33.144 CNTRLR « [Node 028] received current value of setpoint Energy Save Heating: 18 °C
[... cut ...]
17:23:33.160 CNTRLR   [Node 028] [Configuration] 4: metadata updated
17:23:33.161 CNTRLR   [Node 028] [Configuration] 5: metadata updated
17:23:33.162 CNTRLR   [Node 028] [Configuration] 6: metadata updated
17:23:33.163 CNTRLR   [Node 028] [Configuration] 7: metadata updated
17:23:33.163 CNTRLR   [Node 028] [Configuration] 8: metadata updated
17:23:33.164 CNTRLR   [Node 028] [~] [Configuration] isParamInformationFromConfig: false  [internal]
                      => true
17:23:33.164 CNTRLR » [Node 028] querying parameter #1 value...
```

AlCalzone, 2021-03-04:
No that's absolutely not normal. I've seen this quite a bit in logs for setups that were migrated from other systems.

I'm not sure what causes it yet, but in most cases it went away after re-inclusion with zwave-js.
So it is probably caused by a different boostrapping process.

You could try this with one of your spamming nodes (if not battery-powered):
* Shut down zwave-js
* Hard-reset device
* Start zwave-js, wait for device to be detected as dead
* Replace failed node -> re-include previously reset device under the same node ID

If that causes the problem to go away for the node, we could look into bootstrapping nodes as part of the initial interview.

maivorbim, 2021-03-04:
I can confirm that most of the trouble nodes were included via the offline Aeotec Z-stick Gen5 method, i.e. pressing the button on the stick and including the device, then connecting the zstick to HA and letting zwave-js discover the nodes and do its magic.

I will try reincluding the problem nodes via your proposed method and let you know if that helps.

AlCalzone, 2021-03-04:
Did you move around with the stick to include the devices? Then the routes might be messed up.

maivorbim, 2021-03-04:
Yes, the nodes were installed on their final position and I moved around with the zwave stick. First few interviews were also slow and with multiple failures, having to also heal the network multiple times. After a few tries I got them back online. But this happened about a month ago and I did not make any changes since then. The current open issue happened after updating to 6.6.0.

AlCalzone, 2021-03-04:
tbh, I don't know which kind of controller update would cause the nodes to stop responding or start spamming.
Could just be a coincidence - like RF noise or whatever.

AlCalzone, 2021-03-05:
@maivorbim I have bad news for you. I'm not sure why it used to work but maybe you just didn't notice.I've verified with my contact at Silabs that the "walk around" inclusion method is a recipe for disaster.

After the inclusion, the controller thinks it's in direct range of everything.
Effectively, its network map is as wrong as you can possibly get it - and the return routes it is assigning to all the devices won't work.

Best case, you need to do multiple network heals with the stick in intermediate locations. Worst case: you redo everything. Re-including the problematic nodes might work, but you might also be faster off if you just re-include everything from scratch.
Leave the controller/stick in the physical location its meant to be, and start with mains powered devices (to give the controller routing options) then the FLiRS devices.

maivorbim, 2021-03-05:
Thank you for your time and effort to reach out to external parties! I was afraid of this answer, but cannot lie that I have also thought of it myself. Especially since using the stick method is something that is not recommened in the HA docs: https://www.home-assistant.io/docs/z-wave/adding

I will try a couple of more heals and if that does not solve it I will have to reinclude all devices again. Thank you again for your help. Closing this issue for now as it appears to be specific to my installation.

closed 2021-03-05 by maivorbim

AlCalzone, 2021-03-05:
Good luck 😐