zwave-js/zwave-js #1989: [bug] CT200X Thermostat Unreliable communication / not getting updates

srirams, 2021-03-04:
Before submitting a bug please read: <https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report>

### Version

Build/Run method

-  [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 2.1.0
zwavejs version: 6.6.0

### Describe the bug

CT200X Thermostat status switches between Alive / Dead constantly. Setting setpoint from zwavejs seems to work, but changing values on thermostat does not update on zwavejs. Temperature values from thermostat are not updated.

### To Reproduce

Steps to reproduce the behavior:

1. Change temperature on thermostat
2. Value is not updated on zwavejs2mqtt
3. Change setpoint on zwavejs2mqtt
4. Value is updated on thermostat

### Expected behavior

1. Values updated on both thermostat and zwavejs2mqtt

### Additional context

Attached logs below. Node 2 is the thermostat I'm adjusting temperatures on. Node 5 is also a CT200X. Using OpenZWave, thermostat operation was more reliable. https://github.com/zwave-js/node-zwave-js/issues/1721 is related issue, but I don't believe it fixed the issue.

[zwavejs2mqtt.log](https://github.com/zwave-js/zwavejs2mqtt/files/6080800/zwavejs2mqtt.log)
[zwavejs_1.log](https://github.com/zwave-js/zwavejs2mqtt/files/6080801/zwavejs_1.log)
[fd661d38.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6080802/fd661d38.json.txt)

srirams, 2021-04-08:
I think with version 7, this has stopped  working again. Attached log below
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6281498/zwavejs_1.log)

AlCalzone, 2021-04-08:
Can you tell me what I'm looking for? Which node? What did you control and when?

srirams, 2021-04-08:
Sorry, basically the same symptoms as the original report. There are no temperature updates from the thermostat. If I hit "Refresh Values" I get the updated values. It is nodes 10 and 11, which I've filtered by.

I've tried "Reinterview node" as well as deleting the cache files (only keeping the settings.json), and it doesn't make a difference. Unsure if maybe its the same issue as https://github.com/zwave-js/node-zwave-js/issues/2165?

AlCalzone, 2021-04-08:
By temperature updates, do you mean the setpoint or the current measured temperature or both?

srirams, 2021-04-08:
Both. The previous log only has measured temperature updates, I've attached another log where I set the temperature (on the thermostat) to 71, but the value is not updated in zwavejs2mqtt (which still shows 73), even though I see this in the log:

`2021-04-08T22:21:05.413Z SERIAL « 0x010f0004000b09600d0000430301094794                                (17 bytes)
2021-04-08T22:21:05.422Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpoint[1]: 73 => 71        [Endpoint 0]
2021-04-08T22:21:05.423Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                                   1 => 1
2021-04-08T22:21:05.424Z SERIAL » [ACK]                                                                   (0x06)
2021-04-08T22:21:05.426Z DRIVER « [Node 011] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 0
                                    └─[ThermostatSetpointCCReport]
                                        setpoint type: Heating
                                        value:         71 °F
[zwavejs_1(1).log](https://github.com/zwave-js/node-zwave-js/files/6281913/zwavejs_1.1.log)
`

AlCalzone, 2021-04-08:
First off - this happens because of https://github.com/zwave-js/node-zwave-js/pull/2160. We no longer map from the root endpoint to a higher one if the mapping is ambiguous. It worked for this device and some others, but that was a pure coincidence. In other cases the mapping was plain wrong.

It looks like this is just one more case of the confusion around Multi Channel Associations, where the specs are too vague. The Lifeline association is correctly set up to point to Node 1, Endpoint 0. However, the device reports **only** from the root endpoint, so we cannot be sure which endpoint the reports come from. Maybe the device expects us to set up normal associations on the endpoints directly, but zwave-js can't do that yet.

---

We could try and bend that device to our will with compat flags. Take the attached file, map it as a Docker volume into `/usr/app/node_modules/@zwave-js/config/config/devices/0x0098/ct200x.json` and restart. (I'm not sure about the first part of the path, please check if it exists in the container).

You'll probably still see some useless values, but this should cause the Thermostat values to be mapped to endpoint 1 and the Multilevel Sensor stuff to endpoint 2.

[ct200x.json](https://github.com/zwave-js/node-zwave-js/files/6281956/ct200x.json.txt)

srirams, 2021-04-09:
That fixed updating the setpoint when I change the thermostat, but measured temperatures from the thermostat are still not getting mapped correctly:

```2021-04-08T23:35:14.391Z SERIAL « 0x010f0004000b09600d000043030109499a                                (17 bytes)
2021-04-08T23:35:14.394Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpoint[1]: 72 => 73        [Endpoint 0]
2021-04-08T23:35:14.394Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 0] [internal]
                                   1 => 1
2021-04-08T23:35:14.395Z SERIAL » [ACK]                                                                   (0x06)
2021-04-08T23:35:14.396Z DRIVER « [Node 011] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 0
                                    └─[ThermostatSetpointCCReport]
                                        setpoint type: Heating
                                        value:         73 °F
2021-04-08T23:35:14.399Z CNTRLR   [Node 011] Mapping unsolicited report from root device to endpoint #1
2021-04-08T23:35:14.399Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpoint[1]: 72 => 73        [Endpoint 1]
2021-04-08T23:35:14.401Z CNTRLR   [Node 011] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
                                   1 => 1
2021-04-08T23:35:14.407Z SERIAL « 0x01100004000b0a600d00003105012a0348d3                              (18 bytes)
2021-04-08T23:35:14.408Z CNTRLR   [Node 011] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2021-04-08T23:35:14.409Z CNTRLR   [Node 011] [~] [Multilevel Sensor] Air temperature: 83.5 => 84    [Endpoint 0]
2021-04-08T23:35:14.409Z SERIAL » [ACK]                                                                   (0x06)
2021-04-08T23:35:14.410Z DRIVER « [Node 011] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 0
                                    └─[MultilevelSensorCCReport]
                                        type:  Air temperature
                                        scale: Fahrenheit
                                        value: 84
```

I'm not too familiar with zwave so what I say may be nonsense, but I believe this device has three endpoints. Endpoint 0 the "normal" thermostat, Endpoint 1, the "economy" or "away" mode thermostat, Endpoint 2- I'm not sure, perhaps it has something to do with how the thermostat is supposed to display the outside temperature. Perhaps if a report comes from the root device, it should be mapped to all compatible endpoints?

The thermostat is actually quite nice hardware wise, but seems to have many software/firmware bugs unfortunately.

The vivint hub which this is supposed to be paired with doesn't seem to be using multichannel at all. I've attached zniffer logs from setting the setpoint from a thermostat connected to the vivint hub (home id c5 62 89 8b) and from a thermostat connected to zwavejs (home id fd 66 1d 38). Also a log of me including a thermostat to the vivint hub...

This may help to see how their hub sets things up...

[zniffer.zip](https://github.com/zwave-js/node-zwave-js/files/6282380/zniffer.zip)

AlCalzone, 2021-04-09:
> what I say may be nonsense

Endpoints aren't used to address a device in different modes - heat/away are set through the Thermostat Mode CC. They are used to address separate functionality, like different switch channels.
Therefore we cannot blindly map to all endpoints and we cannot blindly map to the first one (we used to do the latter). Consider a 2 channel switch that sends a report for endpoint 2 via 0. Now if we map to 1, endpoint 1 has the wrong state. Same in the other direction.

Your Zniffer log however shows that the hub is using the Association CC on individual endpoints instead of using the Multi Channel Association capability. Afterwards, I don't see any multi channel messages, so maybe the device isn't using the endpoints after all 🤷🏻‍♂️

AlCalzone, 2021-07-11:
https://github.com/zwave-js/node-zwave-js/pull/2997 is going to fix this. We may need to add an additional compat flag afterwards though.

PR #1996 (2021-03-05): fix(node): don't look at MCA support to check if report should be mapped
With this PR, we no longer assume that devices which support Multi Channel Association CC V3+ behave correctly when they send a report from endpoint 0 that should be coming from a higher endpoint.

fixes: #1989
commit 10e53aff (2021-07-13): feat: rework how endpoints and lifeline associations are handled (#2997)

fixes: #2286
fixes: #2257
fixes: #1989
fixes: #2714
fixes: #1860
fixes: #2910
closed 2021-07-17 by AlCalzone