zwave-js/zwave-js #1991: BE469ZP nonce expired on keypad operations

hanerd, 2021-03-04:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

**Describe the bug**

When performing a keypad unlock on a BE469ZP the communication (and thus notifications) for actions appear to be dropped due to an expiring nonces.  

Manual thumb turns reflect appropriately, it's only lock / unlock operations from the keypad.

**Device information**

Which device(s) is/are affected (make/model)?   Allegion Shclage BE469ZP
What are the node IDs? 26

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [X] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image -- App 2.1.1 | ZwaveJS 6.6.0 | Zwave-js-server 1.1.0
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Add a BE469ZP to your mesh
2. Unlock and lock manually with thumbturn and confirm status
3. Unlock with code from keypad
4. See error in logs | Status does not reflect

**Additional context**

**Logfile:**

AlCalzone, 2021-03-04:
This seems to be a persistent problem with those locks. They request a nonce (number used **once**) twice in quick succession, the second request invalidates the first nonce and then they use the first nonce **twice**, also in quick succession.

Unfortunately, this is nothing we can fix or work around without breaking compliance with the Z-Wave specs.

Unless you included the lock with zwave-js, you could try and reset, then re-include the device to see if that changes its behavior.

hanerd, 2021-03-05:
I did include the lock with zwave-js.  That's where I'm a bit stuck--this model works fine with Smartthings and even open-zwave.  

I have two locks -- one a bit older that was included in ozw 1.6 and continues to work fine and this one (node 26) that works mostly fine, but for the issue here.  It was a replacement of a BE468 which gave me a ton of trouble in zwve-js.  I assumed it was because it was a non-plus device with the v1 notifications so I upgraded to since my other BE469ZP works fine.

This one is a brand new device added to zwave-js via zwave-js-mqtt and it doesn't work properly.  My other BE469ZP works totally fine, but was added via HomeAssistant / OZW 1.6.

hanerd, 2021-03-05:
[image]

Just to clarify as well, the BE469ZPs are nodes 25 and 26.  25 is about 9 months old, 26 is about 2.   It replaced a BE468.   The manufacturer specific values are the same for the BE269ZP.  This makes me feel like it's something in the way they're included rather than an issue with the device itself.

AlCalzone, 2021-03-05:
> I did include the lock with zwave-js.

Any chance you have a portable stick and moved that around for the inclusion? Or maybe you moved the device after inclusion? I'm starting to suspect that a lot of the weird issues that get reported all boil down to bad routes.

Please try healing the network (this can take quite a while) to fix the routes and rule that out.

hanerd, 2021-03-05:
I’ll run a heal today. 

Definitely did not do the walk around with the stick option. 

I haven’t moved anything, but I did remove a plug-in unit (on the other side of the house from the lock.)

 The working door lock is a lot closer to the stick so It’s possible that non-working lock was somehow routing through that plug-in unit?  Odd the routes wouldn’t update when I removed it. 
 
Anyway, I’ll heal and update you!  Thanks for looking into this one!

AlCalzone, 2021-03-05:
> Odd the routes wouldn’t update when I removed it.

AFAIK that only happens on demand because updating the routes is a very traffic intensive operation.

hanerd, 2021-03-05:
Ah, good to know.  I assumed removing it from the controller updated the routes.   The OZW folks used to advise against running a heal because of the network traffic load .   I'm betting you're right about some of the weird issues.  I think a lot of us Home Assistant zwave users see the "heal network" button as a "make things worse" button.

That said, the heal worked!   The lock is reporting everything properly and I have no nonce errors in my log.  Also, heal is **significantly** faster with zwave-js than it was with OZW.

AlCalzone, 2021-03-05:
> Also, heal is significantly faster with zwave-js than it was with OZW.

That's weird - we're doing more work (at least last I checked)

closed 2021-03-05 by AlCalzone

hanerd, 2021-03-05:
Looking at my timestamps, 44 zwave plus devices took 8 minutes to heal.

On OZW, it was about 25.

Anyway, thanks again for looking into this one and seriously thank you for creating such an awesome project.