zwave-js/zwave-js #1994: Schlage BE469 support for alarm_level, access_control and alarm_type

gb53smith, 2021-03-05:
**Which device is missing?**

Schlage BE469NXCEN Touchscreen DeadBolt

**What are the IDs?**

Manufacturer ID: 0x003b 
Product type: 0x6341  
Product ID: 0x5044  

from a zwcfg_...xml:

**Did you check if we really don't have it?**

There is a  BE469.json file but the manufacturer is Allegion?  There is a similiar file under Schlage but the manufacturers' id is 0x0003?

- [ ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

Manual [here](https://www.schlage.com/content/dam/sch-us/documents/pdf/installation-manuals/24352403.pdf).

If so, it helps in creating the config file.

I tried integration with Home Assistant.   I can control lock and see battery level.  I am missing sensors seen in Open Zwave 1.4 for alarm_level, access_control and alarm_type.  Not sure where these sensors came from?

I got this in the Zwave-JS log which does not look good:

```16:31:34.698 CNTRLR [Node 004] ConfigurationCC: skipping interview because CC version is < 3 and there is no config file``

AlCalzone, 2021-03-05:
> There is a BE469.json file but the manufacturer is Allegion? There is a similiar file under Schlage but the manufacturers' id is 0x0003?

@blhoward2 any idea?

>  I am missing sensors seen in Open Zwave 1.4 for alarm_level, access_control and alarm_type. Not sure where these sensors came from?

`alarmLevel` and `alarmType` are to be found under `Notification CC`. I don't know where they appear in HomeAssistant.

For the future, please use the correct issue template. The device is identified, so it is not a **missing** file:
[image]
This ensures that you are asked for the relevant information (especially a log) so we can narrow down the problem.

AlCalzone, 2021-03-05:
Heh, I just saw @blhoward2 asked you to open this. Anyways, this

gb53smith, 2021-03-05:
From Google search it seems that Allegion now owns the Schlage electronic lock product line.  Hopefully, only the Manufacturers ID is important and not the assigned name.  I thought that the configuration was missing because of this log file message:
```16:31:34.698 CNTRLR [Node 004] ConfigurationCC: skipping interview because CC version is < 3 and there is no config file```

blhoward2, 2021-03-05:
We only match on the manufacturer ID.

gb53smith, 2021-03-06:
Tried again with latest versions and got this in the log file:
```
16:05:19.153 CNTRLR   [Node 004] ConfigurationCC: skipping interview because CC version is < 3 and t
                      here is no config file
16:05:19.161 CNTRLR   [Node 004] NotificationCC: doing a complete interview...
16:05:19.163 CNTRLR   [Node 004] Interview stage completed: CommandClasses
16:05:19.165 CNTRLR   [Node 004] Interview stage completed: OverwriteConfig
16:05:19.166 CNTRLR » [Node 004] requesting node neighbors...
16:05:19.198 CNTRLR « [Node 004]   node neighbors received: 1, 7, 8, 13
16:05:19.199 CNTRLR   [Node 004] Interview stage completed: Neighbors
16:05:19.200 CNTRLR   [Node 004] Interview completed
16:05:19.202 CNTRLR   [Node 004] The node is ready to be used
```
Now seems to complete the interview despite saying it would skip it?
I now get sensor.none_alarmlevel and sensor.none_alarmtype in Home Assistant although they are both unknown.
Probably, there is no problem with the config file so this issue can be closed.

AlCalzone, 2021-03-06:
> ConfigurationCC: skipping interview because CC version is < 3 and there is no config file

Means that the ConfigurationCC interview is skipped, not the entire interview. Prior to V3 we'd have to query 255 possible parameters just to check if they exist and risk waiting for a timeout for 10 seconds each. Hence the necessity for configuration files.

gb53smith, 2021-03-06:
I got the BE469 working with zwave_js  in Home Assistant.  I used zwave_js_event to trigger an automation that set an input_text helper to display the reason for the locks' last action.

closed 2021-03-06 by gb53smith