zwave-js/zwave-js #1997: [bug] NAS-WR01ZE - Configuration values arent show on control panel

paszczaq, 2021-03-05:
### Version

HA Supervised on RPI4/4GB
core-2021.3.1
supervisor-2021.03.4

- [x] Docker - Z-Wave JS to MQTT: 0.6.0
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

App Version: 2.1.0
zwavejs2mqtt version: 1.1.0
zwavejs version: 6.6.0

### Describe the bug

When setting parameters in Config section for Neo Coolcam Power Plug (NAS-WR01ZE - Device ID
600-4135-512 (0x0258-0x0200-0x1027) ) these settings arent transferred correctly to the device (ie. setting option to disable LED doesnt disable LED on the device)

In debug log it shows some data send to the device:
`2021-03-05 11:58:04.943 INFO ZWAVE: Calling api writeValue with args: [ { nodeId: 29, commandClass: 112, property: 5 }, 0, [length]: 2 ]`
`2021-03-05 11:58:04.949 INFO ZWAVE: Writing 0 to 29-112-0-5`
`2021-03-05 11:58:05.131 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }`
`2021-03-05 11:58:06.430 INFO ZWAVE: Node 29: value updated: 112-0-5 0 => 120`

but settings page doesnt show any change

[image]

### To Reproduce

Steps to reproduce the behavior:

1. Go to 'Node - Power Plug'
2. Click on 'Configuration'
3. Configure some parameters, click ">" to save/submit value
4. After while value shows as before for example blank or '1'

### Expected behavior

Show proper configuration values ie. Disable LED / Enabled LED

PS. I can see the problem is only with NAS-WR01ZE PowerPlug. I have 3x neo coolcam and 1x fibaro which is working as expected

paszczaq, 2021-03-05:
@robertsLando should I set up log level to silly or normal will be enough?

robertsLando, 2021-03-05:
silly

porzino, 2021-03-06:
Hi.
I have the same problem.

I have two versions of the same device: the old version of Neo Coolcam Power Plug (NAS-WR01ZE) is Device ID 600-4231-3 (0x0258-0x0003-0x1087) and the configuration template works great, but the problem is with the new version, Device ID
600-4135-512 (0x0258-0x0200-0x1027).
I have the official user manual, and the problem is that they have changed the order of the parameters.
I attach an image of the manual where the order of the current parameters appears. 

[image]

paszczaq, 2021-03-08:
Hello, here goes full logs:
I did change some values - 2021-03-08 10:38:10.118 - node 29 is neo coolcam power plug
[zwavejs2mqtt.zip](https://github.com/zwave-js/node-zwave-js/files/6100374/zwavejs2mqtt.zip)

AlCalzone, 2021-03-09:
@paszczaq wrong logs. I need the ones from zwave-js, see https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs point 1.