zwave-js/zwave-js #1999: removeNodeFromAllAssocations misspelled

MartinHjelmare, 2021-03-05:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is **not** described there.
- [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

---

The controller command `removeNodeFromAllAssocations` seems to be misspelled. Looks like it should be `removeNodeFromAllAssociations`.

- Docs: https://zwave-js.github.io/node-zwave-js/#/api/controller?id=managing-associations
- Code: https://github.com/zwave-js/node-zwave-js/blob/fc5492fbe517901ea0fd30a81b6be6eb2020be4c/packages/zwave-js/src/lib/controller/Controller.ts#L2027

Would it be possible to fix this typo, even though it could be considered a breaking change?

---

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-03-05:
I'm collecting a bunch of breaking changes for 7.x, so this is just another one for that heap :)

AlCalzone, 2021-03-14:
fixed in https://github.com/zwave-js/node-zwave-js/pull/2030

PR #2001 (2021-03-08): refactor: fix typo in removeNodeFromAllAssociations
fixes: #1999
closed 2021-03-14 by AlCalzone