zwave-js/zwave-js #2000: Haseman R4D4 relay/dimmer

jvd-nl, 2021-03-05:
**Which device is missing?**

Haseman R4D4

**What are the IDs?**

Manufacturer ID: 0x0115
Product type: 0xf222
Product ID: 0x2111  

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
http://ekitzone.com/image/catalog/PDF/Haseman_R4D4_Manual.pdf

Device is working using OpenZwave 1.4 with Home Assistant (although as generic device, not by name), here my zwcfg:
```
	<Node id="15" name="" location="" basic="4" generic="17" specific="1" roletype="21" devicetype="1536" nodetype="0" type="Multilevel Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
		<Manufacturer id="115" name="Z-Wave.Me">
			<Product type="f222" id="2111" name="Unknown: type=f222, id=2111" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="5" innif="true" mapping="38">
				<Instance index="1" />
				<Instance index="2" endpoint="1" />
				<Instance index="3" endpoint="2" />
				<Instance index="4" endpoint="3" />
				<Instance index="5" endpoint="4" />
			</CommandClass>
			<CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Instance index="2" endpoint="5" />
				<Instance index="3" endpoint="6" />
				<Instance index="4" endpoint="7" />
				<Instance index="5" endpoint="8" />
				<Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="bool" genre="user" instance="2" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="bool" genre="user" instance="3" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="bool" genre="user" instance="4" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="bool" genre="user" instance="5" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
			</CommandClass>
			<CommandClass id="38" name="COMMAND_CLASS_SWITCH_MULTILEVEL" version="1" request_flags="1" innif="true">
				<Instance index="1" />
				<Instance index="2" endpoint="1" />
				<Instance index="3" endpoint="2" />
				<Instance index="4" endpoint="3" />
				<Instance index="5" endpoint="4" />
				<Value type="byte" genre="user" instance="1" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="99" />
				<Value type="button" genre="user" instance="1" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="button" genre="user" instance="1" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="bool" genre="system" instance="1" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="byte" genre="system" instance="1" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="byte" genre="user" instance="2" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="99" />
				<Value type="button" genre="user" instance="2" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="button" genre="user" instance="2" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="bool" genre="system" instance="2" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="byte" genre="system" instance="2" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="byte" genre="user" instance="3" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="button" genre="user" instance="3" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="button" genre="user" instance="3" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="bool" genre="system" instance="3" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="byte" genre="system" instance="3" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="byte" genre="user" instance="4" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="button" genre="user" instance="4" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="button" genre="user" instance="4" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="bool" genre="system" instance="4" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
[... 94 lines cut ...]
			<CommandClass id="134" name="COMMAND_CLASS_VERSION" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3" />
				<Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.38" />
				<Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="2.16" />
			</CommandClass>
			<CommandClass id="142" name="COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Associations num_groups="9">
					<Group index="1" max_associations="8" label="Group 1" auto="true" multiInstance="true">
						<Node id="1" />
					</Group>
					<Group index="2" max_associations="8" label="Group 2" auto="false" multiInstance="true" />
					<Group index="3" max_associations="8" label="Group 3" auto="false" multiInstance="true" />
					<Group index="4" max_associations="8" label="Group 4" auto="false" multiInstance="true" />
					<Group index="5" max_associations="8" label="Group 5" auto="false" multiInstance="true" />
					<Group index="6" max_associations="8" label="Group 6" auto="false" multiInstance="true" />
					<Group index="7" max_associations="8" label="Group 7" auto="false" multiInstance="true" />
					<Group index="8" max_associations="8" label="Group 8" auto="false" multiInstance="true" />
					<Group index="9" max_associations="8" label="Group 9" auto="false" multiInstance="true" />
				</Associations>
			</CommandClass>
		</CommandClasses>
	</Node>
```

AlCalzone, 2021-03-05:
If you are willing to provide a PR, we can include this pretty soon.
https://zwave-js.github.io/node-zwave-js/#/development/config-files for starters, we can help you along the way.

marcus-j-davies, 2021-03-11:
Have provided a PR for this - hope you don't mind @jvd-nl.

jvd-nl, 2021-03-11:
Thank you @marcus-j-davies. I didn't contribute here yet, so I wanted to test this e2e first, which would take me some time to setup.

closed 2021-03-12 by AlCalzone