zwave-js/zwave-js #2002: Inovelli LZW30-SN missing Parameter 13

jayscovill, 2021-03-05:
**Which device is missing?**

Inovelli LZW30-SN configuration parameter 13

**What are the IDs?**

Manufacturer ID: 0x031e 
Product type: Where do I find this? 
Product ID: 0x0002:0x0001 

- [x ] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ ] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ ] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

As of FW 1.17 Inovelli added a parameter 13 to specify load type but it is not in the configuration database for the device.

https://support.inovelli.com/portal/en/kb/articles/firmware-v1-17-beta-lzw30-sn-switch-red-series-gen-2

ehsman, 2021-03-06:
I'm guessing that the LZW30-SN might need multiple configs like LZW31-SN. As there is also a parameter 51 (same in the existing LZW31-SN config) added in firmware 1.19

https://support.inovelli.com/portal/en/kb/articles/firmware-v1-19-beta-lzw30-sn-switch-red-series-gen-2

AlCalzone, 2021-03-06:
FYI: We now have the capability to define config parameters conditionally for specific firmware ranges in one file.
https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=conditional-configuration-settings

Maverick1872, 2021-03-09:
@AlCalzone I assume a PR to update https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x031e/lzw30-sn.json with a conditional configuration on parameters 13 & 51 would be all that's needed?

AlCalzone, 2021-03-09:
I think so, yes.

AlCalzone, 2021-03-10:
fixed in https://github.com/zwave-js/node-zwave-js/pull/2025

closed 2021-03-10 by AlCalzone