zwave-js/zwave-js #2008: zw6302.json motion sensor config is saved under "zooz" please copy or move to "GE/Jasco"

djaydev, 2021-03-07:
The config for zw6302 Portable Smart Motion Sensor is saved under `node-zwave-js/packages/config/config/devices/0x027a/zw6302.json` in github, but I believe this config should instead be saved under `node-zwave-js/packages/config/config/devices/0x0063/46201.json` in github. From what I can find there isn't a device called Zooz zw6302 but if it is, please copy the file to GE/Jasco folder.

This issue causes the GE/Jasco zw6302 to work improperly with zwavejs and stay awake continuously, resulting in the 2 cr123a batteries to die in weeks instead of years.
Thank you

djaydev, 2021-03-07:
zwave-js/zwavejs2mqtt#416

AlCalzone, 2021-03-07:
@blhoward2 that's your turf :)

blhoward2, 2021-03-08:
Interesting. Not really sure how that would have happened.

blhoward2, 2021-03-08:
It turns out I already merged a file for this last night. It'll get cleaned up when I template the GE devices but it's there.

closed 2021-03-08 by blhoward2