zwave-js/zwave-js #2012: Some Qubino 2-relay ZMNHBD devices (depending on firmware version) do not send instant status reports

enredador, 2021-03-08:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Some Qubino 2-relay ZMNHBD devices that do not send "instant status" reports to zwavejs2mqtt 
 when they are  operated manually. They are all firmare version 1.1. However other ones, versioned 1.2 and 5.0 work perfectly. All these devices worked ok in the ancient zwave 1.4. 

**Device information**

Which device(s) is/are affected (make/model)?   QUBINO ZMNHBD Flush 2 relay (firmware 1.1)
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *(Zwave 1.4)*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image 2.1.1
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

NA

**Additional context**

Log file for node 37 (37.txt) when switched on, then off first for endpoint 1, then for endpoint 2. Firmware version 1.1. (issue reported)
Log file for node 39 (39.txt) when switched on, then off first for endpoint 1, then for endpoint 2. Firmware version 1.2. (no issue)
**Logfile:**

AlCalzone, 2021-03-08:
:( you didn't read the issue template - especially the part where it tells you which logfile to select.

enredador, 2021-03-09:
OMG. I may have skipped that part. Sorry! Please find log and updated info in my original report

AlCalzone, 2021-03-09:
This should be fixed in v6.6.1 - please try that.

AlCalzone, 2021-03-09:
Okay it might be that endpoint 0 is reporting something that cannot be mapped to endpoint 1. Can you share your cache files `<homeid>.json` and `<homeid>.values.jsonl`, so I can verify?

enredador, 2021-03-09:
Sure!

[e22379bc.values.jsonl.TXT](https://github.com/zwave-js/node-zwave-js/files/6108728/e22379bc.values.jsonl.TXT)
[e22379bc.json.TXT](https://github.com/zwave-js/node-zwave-js/files/6108729/e22379bc.json.TXT)

AlCalzone, 2021-03-09:
Oh you were faster :D Looks like each endpoint reports different things. There's two things that could be done.

1. Add Node 1 to the following association groups:
[image]
and check if you get the expected reports now

2. if that doesn't help, we need to add a compat flag for FW 1.1

enredador, 2021-03-09:
No luck. I have tried to add Node 1 to the association groups 4 & 7. Same behavior.
In any case, since the missing reports are for the switches status -not power meter reports, I also tried to add Node 1 to the association groups 3 & 6 and even groups 2 & 5 (in separate tests). Negative in all cases.

It is curious, zwavejs adds one device in HA for sensor_electric_w_value for devices in FW 1.1 and 1.2, that aggregates the power consumption for both endpoints, and -aggregated- reports are received without issues. For devices in FW 5.0 three devices are created, two of them get the separate consumptions for both endpoints.

AlCalzone, 2021-03-09:
Forgive me, but I might have been looking at the wrong values then 🙈

AlCalzone, 2021-03-09:
Hm no I don't get it. The difference between the devices is that FW 1.2+ uses multi channel encapsulation to report those changes and they are already on the correct endpoint.

FW 1.1 does not and sends a report for the root endpoint (0) instead. However we have a test case that tests this exact scenario and it passes for me, meaning the value does get mapped to endpoint 1.

AlCalzone, 2021-03-09:
Wait a minute...
```
{"k":"{\"nodeId\":37,\"commandClass\":142,\"endpoint\":0,\"property\":\"endpoints\",\"propertyKey\":1}","v":[{"nodeId":1,"endpoint":1}]}
```

Did you fiddle with the lifeline association? This should be `{"nodeId":1,"endpoint":0}`

enredador, 2021-03-09:
Yes, I was meaning that. I changed node 37 to 1:0 and re-interviewed it. The same behavior.

This log maybe have fiddles with associations, trying what you suggested in [https://github.com/zwave-js/node-zwave-js/issues/2012#issuecomment-793885764](url)

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6110476/zwavejs_1.log)

AlCalzone, 2021-03-09:
That doesn't help - I'm going to build you a test version that enhances logging.

AlCalzone, 2021-03-09:
Can you either build your container against the branch in https://github.com/zwave-js/node-zwave-js/pull/2026 or install the version of zwave-js that the bot is going to post there in a minute, then make another log with that one?

enredador, 2021-03-10:
Working in it.... I need to setup a Node.js environment first. May be a couple of days....

AlCalzone, 2021-03-10:
Do you know https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds ?

enredador, 2021-03-11:
No I didn't. Thanks! It has been better -and faster- option, once solved silly issues with a hanging rpi3 and the lack of credentials for the docker build.

Here you are the logs. I operated the two switches in node 037, via HA and pushing the buttons.
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6124395/zwavejs_1.log)

AlCalzone, 2021-03-11:
Hmm... I think I was blind earlier. The config file for your device is [this one](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0159/zmnhbd.json), which sets the compat option `preserveRootApplicationCCValueIDs` (I got this info from Qubino that it is necessary).
This prevents the value from being mapped to endpoints. 

You could ask the Qubino support if there is something you can do to make the device report values for each endpoint.

For me, the remaining question is now: do you see the value for endpoint 0 change in zwavejs2mqtt?

enredador, 2021-03-11:
Regarding the last question. Each capture is with one of the endpoints switched on.

[image]
[image]

AlCalzone, 2021-03-11:
I'm looking at `37-37-0-currentValue` - my guess this shows if any of the endpoints are on, so you won't see anything in a screenshot. Can you turn off both and see if that value changes?

enredador, 2021-03-11:
Yes. I omitted that picture because I thought it was obvious.
[image]

AlCalzone, 2021-03-11:
Ok so the reporting for the root endpoint works. We won't get further here without Qubino's answer what to do about the single endpoints.
Worst case you'll need an automation that polls both endpoints when the combined channel is updated (not necessarily changed).

enredador, 2021-03-11:
I will ask Qubino, although I guess they will offer a firmware upgrade [https://support.qubino.com/support/solutions/articles/44001936790-do-qubino-devices-support-ota-can-i-update-the-firmware-ota-](url), a process that is expected to be slow, long-term and (maybe) expensive.

Is it possible to poll just one device?

AlCalzone, 2021-03-11:
It is possible to poll just one value (and the associated ones):
https://zwave-js.github.io/zwavejs2mqtt/#/usage/setup?id=poll-values

enredador, 2021-03-12:
As I understand, one can poll just one value for ALL devices of the same type. I had tried this but the z-network becomes laggy. I have also tried automations, as you suggested:

[image]

But they do not work as expected, since it only triggers when root value changes, not when it updates, what would. be the desired behavior (HA switch devices created have no attributes). May be I misunderstood your suggestion for the automations.

AlCalzone, 2021-03-12:
I don't know how HA operates in this regard - best if you ask in the HA channels for support. You need to be able to react to a value update, even it if is not a change. If that is not possible with HA, that should definitely be added.

`zwave-js` offers a method to poll a single value ID once and `zwavejs2mqtt` also offers this API via MQTT AFAIK.

blhoward2, 2021-03-12:
> As I understand, one can poll just one value for ALL devices of the same type. I had tried this but the z-network becomes laggy. I have also tried automations, as you suggested:
>
>
>
> [image]
>
>
>
> But [...]

You should be able to refresh just one value from an automation. 

SERVICE ZWAVE_JS.REFRESH_VALUE

This service will refresh the value(s) for an entity. This service will generate extra traffic on your Z-Wave network and should be used sparingly. Updates from devices on battery may take some time to be received.

Service Data Attribute	Required	Description
entity_id	yes	Entity or list of entities to refresh values for.
refresh_all_values	no	Whether all values should be refreshed. If false, only the primary value will be refreshed. If true, all watched values will be refreshed.

https://www.home-assistant.io/integrations/zwave_js/

enredador, 2021-03-12:
> 
> You should be able to refresh just one value from an automation, or all of them.
> 
> SERVICE ZWAVE_JS.REFRESH_VALUE
> 
> refresh_all_values no Whether all values should be refreshed. If false, only the primary value will be refreshed. If true, all watched values will be refreshed.
> 
> https://www.home-assistant.io/integrations/zwave_js/
> 
> There is a last updated attribute I believe it just doesn't show in the list. If you go to developer tools it does list a last updated time down towards the bottom. Can you access that through a jinja template? It should be triggering on updates anyway, though, if you don't have from/to.
> 
> https://community.home-assistant.io/t/how-to-trigger-on-changes-to-last-updated/230354

Thank you for four comment.
May be my post was confusing. The first statement about polling any value for all devices of the same type referred to the zwavejs2mqtt control panel. However, the statement below pointed in the same direction than yours. The problem is that the HA zwave2js integration does not pass any atribute, such as the last_updated, what would make -I hope- the rules using the service  ZWAVE_JS.REFRESH_VALUE. Part of my "developer tools" shows as follows. All the switches shown have been set up by the zwavejs integration: No atribute, except for the state value.
[image]

enredador, 2021-03-12:
> You should be able to refresh just one value from an automation, or all of them.
>
> SERVICE ZWAVE_JS.REFRESH_VALUE
>
> refresh_all_values no Whether all values should be refreshed. If false, only the [...]

Thank you for four comment.
May be my post was confusing. The first statement about polling any value for all devices of the same type referred to the zwavejs2mqtt control panel. However, the statement below pointed in the same direction than yours. The problem is that the HA zwave2js integration does not pass any atribute, such as the last_updated, what would make work -I hope- the rules using the service  ZWAVE_JS.REFRESH_VALUE. Part of my "developer tools" shows as follows. All the switches shown have been set up by the zwavejs integration: No atribute, except for the state value.
[image]

@AlCalzone  is right.The integration should pass further information to HA from zwavejs than the raw state value. I will also follow his suggestion.

enredador, 2021-03-15:
I have been able to solve the issue enabling MQTT in zwavejs2mqtt and using automations to catch the value updates (wether changing state or not) to endpoint 0., forcing the polling to endpoints 1 and 2. This way:

[image]

It seems to work fine (I have to watch performance). Thank you for your help, @AlCalzone. I also forwarded the issue to Qubino support. Waiting for answer....

enredador, 2021-03-16:
What Qubino says about the issue follows:

> The Flush 2 Relay device is an older device, that doesn't have an endpoint1, which would have the same functionality as the root endpoint. Usually, the same values are reported from the root endpoint and endpoint1, but the Flush 2 Relay has cumulative values on the root endpoint, relay1/channel1 values on endpoint1 and relay2/channel2 values on endpoint2.

> The unsolicited reports (state reports, sent from the device automatically, without it being polled), that are sent from the device, depends on the configured Lifeline association;: 
i) If an ordinary association is configured, using ASSOCIATION_SET(groupId=1, nodeId=1), the Flush 2 Relay will only send the cumulative values for the root endpoint (W and kWh values) and will not send separate channel values. 
ii) if a multi channel lifeline is configured, using MULTI_CHANNEL_ASSOCIATION_SET(groupId=1, multichannelnodeid=1, endpoint=0), the device will send separate channel values, but will not send cumulative (root endpoint) values.

> It seems like zwavejs configured an ordinary lifeline for this older version of the device OR configuring the multi channel Lifeline failed, so you'll have to configure the lifeline association manually, by first issuing an ASSOCIATION_SET(groupId=1, nodeId=1) and then sending MULTI_CHANNEL_ASSOCIATION_SET(groupId=1, multichannelnodeid=1, endpoint=0). Since you are using the zwavejs2mqtt, you should have access to the control panel, which should have a section for configuring associations. Another option is to use Silicon Lab's PC Controller, which will enable you to send the same commands. This is shown in the attached gifs (you should follow the PCControllerSettingMCLifeline1). The tool can be downloaded from here. 

> If the device reported everything in ozw_1.4, we should be able to configure it to report everything also for zwavejs.

I am not sure on what they say about (MULTI_CHANNEL)_ASOCIATION_SET. My devices seemed to be configured to endpoint 1, but changing  --with difficulty-- to endpoint 0 seems not to work.

In any case, the management of associations is difficult with the interface. If I try to delete associations to lifeline, I get the old associations. In addition to that, if I get an association for a node, the same association appears in other nodes that are opened in the interface. I hardly could change endpoint 1 to endpoint 0 in one of my devices. 

Is there an issue with association management in the interface?

AlCalzone, 2021-03-16:
1. Can you share a screenshot of the association settings in the interface?

2. To start fresh and figure out where things go wrong, I suggest the following procedure:
- Enable zwave-js logging (if not already)
- Remove all associations of that device via the interface
- Do a re-interview and share:
   - a log of that
   - another screenshot of the associations after the interview
   - the cache files (like above)

enredador, 2021-03-16:
Screenshots for step 1 (zwavejs2mqtt recently restarted).

Lifeline for Node 37:
[image]

After that, screenshot for getting lifeline for Node 36, leaving open the data for Node 37. Please, note that its data has changed:
[image]

Step 2 along the afternoon....

GSzabados, 2021-03-16:
Hi,

I just want chime in to this discussion as I believe have seen this issue on an other platform (SmartThings) with Qubino, Fibaro and some other multi endpoint devices.

Here is a summary of what was going wrong with a firmware update there, it was written by one of the developers:

https://community.smartthings.com/t/z-wave-issues-with-firmware-update-30-3/192759/123

AlCalzone, 2021-03-16:
@enredador Best if you also make a log where you delete the lifeline. Not sure if that's the device messing up or the UI.

@GSzabados zwave-js does not care for the destination endpoint. The problem here is that the reports aren't getting sent at all.

GSzabados, 2021-03-16:
@AlCalzone, sorry for the misunderstanding.

I've looked at the Device Handler which was written for the Qubino 2 Relay for SmartThings. (Eric Maycock, who wrote it, is the CTO of Inovelli)

https://github.com/erocm123/SmartThingsPublic/blob/master/devicetypes/erocm123/qubino-flush-2-relays.src/qubino-flush-2-relays.groovy

If you look at it, it is full of get commands for endpoint 1 and 2. The behaviour must have been known before.

enredador, 2021-03-16:
> Step 2 along the afternoon....

Here it is finally.

> Remove all associations of that device via the interface

Done.  I tried both the red trash icon near the association info, and then the button "remove all"

> Do a re-interview and share:
> a log of that

[zwavejs_1.log - Log for all the process from restart](https://github.com/zwave-js/node-zwave-js/files/6149764/zwavejs_1.log)

> another screenshot of the associations after the interview
[image]

> the cache files (like above)

[e22379bc.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6149838/e22379bc.values.jsonl.txt)
[e22379bc.json.txt](https://github.com/zwave-js/node-zwave-js/files/6149836/e22379bc.json.txt)

AlCalzone, 2021-03-16:
That's during your re-interview, so the multi channel association seems to be set up correctly.
```
14:30:34.948 CNTRLR « [Node 037] received information for multi channel association group #1:
                      maximum # of nodes:           1
                      currently assigned nodes:     
                      currently assigned endpoints: 1:0
```
Which reports does your device send now, or rather on which endpoints do you receive updates now?

> i) If an ordinary association is configured, using ASSOCIATION_SET(groupId=1, nodeId=1), the Flush 2 Relay will only send the cumulative values for the root endpoint (W and kWh values) and will not [...]

This could be a problem if it is really required to have both. Currently, zwave-js removes the node association if it needs to set an endpoint association.

enredador, 2021-03-16:
Ok. I deleted all the associations of these type of nodes and reinterviewed all of them. They all are associated to 1:0 as desired.

FW 1.1 nodes do not report instant status, as usual, when the zwave2mqtt catching updates rules are disabled.
Power consumption (Watt) reports are got aggregated on (I guess) endpoint 0, BUT energy consumption (kWh) are received in different reports, and also accumulate  (There are at least values, but LED lightning do not update this frequently).

Screenshots for one light, the other and both:
[image]
[image]
[image]

The only FW1.2 node report instant status, and separate power consumption values (endpoints 1, 2) and no aggregated values on endpoint 0 (reports are always 0W). Energy reports are got in endpoints 0 and 1, but due to the big difference I am not sure of their meaning (may be one of the lights and aggregated? separated lights?)
[image]

AlCalzone, 2021-03-16:
FW 1.1:
With status do you mean switch status (`...-1-currentValue` and `...-2-currentValue`)? Does `...-0-currentValue` still get updated?

FW 1.2: Do you get cumulative updates if you add an association to Node 1 (no endpoint)?

enredador, 2021-03-16:
> FW 1.1: With status do you mean switch status (...-1-currentValue and ...-2-currentValue)? Does ...-0-currentValue still get updated?

Oh! I meant that only 0-currentValue is updated, as before. I think that is shown in the screen capture. If I force refresh values, then 1-currentValue or 2-currentValue is also updated.

> FW 1.2: Do you get cumulative updates if you add an association to Node 1 (no endpoint)?

The behavior changes: I get cumulative updates for power (endpoint 0) and separated values for energy for both endpoints !!
[image]

In any case, the "forced" association to Node 1 (no endpoint)  is lost as soon as a new node interview is done!
These devices are funny, full of surprises.

AlCalzone, 2021-03-16:
> In any case, the "forced" association to Node 1 (no endpoint) is lost as soon as a new node interview is done!

That's zwave-js removing an association it considers wrong. I can think about removing that - is the behavior with both associations what you'd consider correct?

> Oh! I meant that only 0-currentValue is updated, as before. I think that is shown in the screen capture.

Hard to see changes in a static image ;) I guess they don't update either if you also add the node association in Firmware 1.1?
In that case, it sounds like it is not behaving as the support said it should.

enredador, 2021-03-16:
> Hard to see changes in a static image ;) I guess they don't update either if you also add the node association in Firmware 1.1?
> In that case, it sounds like it is not behaving as the support said it should.

No, they don't update neither with the lifeline association to Node 1 (No endpoint) nor with association to Node 1 (endpoint 0) in FW 1.1. I think this association group do not accept more than one association.
I agree, the observed behavior is not as support said.

> That's zwave-js removing an association it considers wrong. I can think about removing that - is the behavior with both associations what you'd consider correct?

I am not sure which association -to node or to node+endpoint- is correct (regarding W and kWh reports). I think maybe neither is correct at all.  I am not very interested in kWh reports, but I am in report status, that do not work in any case as they should.

AlCalzone, 2021-03-16:
Hm :( Guess you'll have to do another round with the support.

AlCalzone, 2021-03-24:
Anything new?

enredador, 2021-03-24:
Not yet. Qubino support people want to access a computer running the Silabs PCcontroller with this device included to have a look (they say that do not have any module with so old FW). I am still preparing how to, since I do not have any Windows system and I want to test if that program runs under emulation on a Mac. I will be also out for 10 days....

I haven't given it up .....

enredador, 2021-04-07:
Hello. I have some news about this.

This morning, some people from Qubino support team have been make some tests directly on my zwave network using the SIlabs PCController. They have report this two texts:

[image]

[image]

I am not sure whether this is different from what they suggested before or not. I am a little bit dizzy with all these dances with endpoints. The point is that, after connecting the dongle newly to the machine that runs zwavejs2mqtt+Homeassistant, and reinterview the node (it was excluded, and included with the PC Controller) the device reports perfectly!

I have tried to mimic these associations for a different node with no success. But I think that the problem is that the associations are not made.For instance, trying to set the following association:

`Node: 36 - Group: Q1 switch binary - Target: NodeID_1 - Endpoint: 1`

The log says:

> Calling api addAssociations with args: [ 36, 3, [ { nodeId: 1, endpoint: 1 }, [length]: 1 ], [length]: 3 ]
> Error while adding associations to 36: The endpoint 1 was not found on node 1!

Instead, If I try:

`Node: 36 - Group: Q1 switch binary - Target: NodeID_1 - Endpoint: 0`

The log says:

> Calling api addAssociations with args: [ 36, 3, [ { nodeId: 1, endpoint: 0 }, [length]: 1 ], [length]: 3 ]
> Assocaitions: Adding Node 1 to Group 3 of  Node 36
>  Success zwave api call addAssociations { success: true, message: 'Success zwave api call', result: undefined }
>  Calling api getAssociations with args: [ 36, 3, [length]: 2 ]
> Success zwave api call getAssociations {
>    success: true,
>    message: 'Success zwave api call',
 >   result: [ { nodeId: 0, endpoint: 0 }, [length]: 1 ]

nodeid 0???. The fact is that in the UI "no associations" is shown when requested, but the log shows:

> result: [ { nodeId: 0, endpoint: 0 }, [length]: 1 ]

I have also updated my zwavejs2mqtt to docker :latest (7.1.1) and I observe the same behavior.

Am I missing something when working with the associations in zwavejs? May be the multichannel lifeline setup needs something else?

enredador, 2021-04-07:
Some pics from PC Controller screen show how the associations were configured:

[image]
[image]
[image]
[image]

AlCalzone, 2021-04-07:
> The endpoint 1 was not found on node 1!

Z-Wave JS checks if the association you're trying to make is valid and doesn't like that you're trying to set an association to endpoint 1. I think it would be okay to omit this check for the controller itself.

enredador, 2021-04-08:
No, they didn't. Just that this is a very old firmware. 

BUT, I have a FW1.2 that reports status with no issues. So may be you can act only in FW1.1 or older ones.

AlCalzone, 2021-04-14:
Associations to Node 1 with a different endpoint than 1 are now possible with #2347.
As for setting up the associations, I'll have to revisit this after #2286 is solved.

enredador, 2021-04-25:
Hello. I just updated to zwavejs 7.2.4 / zwavejs2mqtt 3.5.0. Associations can be configured now to get all unsolicited reports for old Qubino 2 relay devices, including independent power consumption reports for endpoints.

Very nice job, @AlCalzone. Thank you very much.

AlCalzone, 2021-04-27:
Huh, I didn't release the association changes yet 👀

enredador, 2021-04-27:
Then, how did I?
[image]

AlCalzone, 2021-04-27:
Ahh, right. That part was released

Mr-Hell, 2021-11-05:
Hi!
I’ve got similar issues with old Qubino Flush 1 Relays ZMNHAA (Z-Wave chip firmware versions 1.12, Z-Wave protocol version 3.28). I’m not receiving any unsolicited updates e.g. energy reports or status updates when manually operating a switch.

Currently running zwavejs2mqtt version 5.10.1 (Z-wave JS 8.6.1).

I migrated from Fibaro Home Center 2 (where all Qubino devices worked just fine) to running Home Assistant on a NUC with an AEON Labs Gen5 USB Z‐Stick Controller.

How should I proceed, should I open a separate issue? But I’m not sure which logs you want and how to get them...

AlCalzone, 2021-11-05:
> How should I proceed, should I open a separate issue? But I’m not sure which logs you want and how to get them...

Yes please. The issue template will tell you what's required and where to find it

adjego, 2022-11-30:
Maybe not the correct place to add a comment, but i'm having the same issue here where the manual light switch presses are not sent towards the controller.

I've been struggling quite a bit but not been able to add the correct associations, and i was wondering if someone can help.

I have two of these Relay's and it seems that some part of it did work on my side, when target endpoint was set to 1. **But for testing i removed these accociations, to see this was making it work**, but i'm not able to re-add them again, as i can only select "no endpoint" or "endpoint 0" in the drop down. Is there an easy way to also be able to add "1" as target endpoint?

I'm running:
zwave-js-ui: 8.5.1.ad42de0
zwave-js: 10.3.1
home id: 3985200507
home hex: 0xed89557b

firmware of both my relay's is FW1.1

GSzabados, 2022-11-30:
It might help you, but I struggled with a single relay module on v1.1.

And it requires to set the `Endpoints I2 Notification Type and Event` and `Endpoints I3 Notification Type and Event` to `Home Security (default)`. It is under Configuration. 
If those were set to `Disabled`  it just did not report the physical switch changes.

AlCalzone, 2022-11-30:
Note that you need to exclude and re-include Qubino devices after changing the endpoint configuration.

adjego, 2022-12-01:
> It might help you, but I struggled with a single relay module on v1.1.
>
> And it requires to set the `Endpoints I2 Notification Type and Event` and `Endpoints I3 Notification Type and Event` to `Home [...]

Thanks for the remark, but these properties are not available for the 2 replay versions. I did check my other 2 single qubino modules and they both have this set to disabled, but i have never had a problem with those. They always report manual switch changes correctly.

[1 more reply]