zwave-js/zwave-js #2020: [bug]replace failed node for locks results in non-secure lock inclusion

scyto, 2021-03-08:
### Version

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

App Version 2.1.1+92d502a.92d502a
Zwavejs Version 6.6.0-fc5492f
Zwavejs-server Version 1.1.0

### Describe the bug

I removed a zwave module from my already included kwikset 910 and with a zwave+ module and attempt to do a replace failed node.
Pressing the include button the lock doesn't appear to be recognized by  zwavejs.

### To Reproduce

Steps to reproduce the behavior:

1.  remove batteries from lock
2. remove zwave module from lock
3. press program button to drain system
4. insert zwave+ module and batteries
5. start a tail -f of zwavejs_1.log
6. choose replace failed node on the lock and say yes to secure inclusion 
8. wait for log to state inclusion process has started
9. press inclusion button on lock
10.  exclusion process stops and lock is never included

### Expected behavior

Lock to be included

### Additional context

```
18:12:27.636 SERIAL « 0x011d0013a801014e007f7f7f7f7f00010600000000420300007f7f7f7f7f51    (31 bytes)
18:12:27.637 SERIAL » [ACK]                                                                   (0x06)
18:12:27.638 DRIVER « [REQ] [SendData]
                        callback id:     168
                        transmit status: NoAck
18:12:27.644 CNTRLR   [Node 003] Node 3 did not respond after 1 attempts, it is presumed dead
18:12:27.647 CNTRLR   [Node 003] ping failed: Failed to send the command after 1 attempts (Status No
                      Ack)
18:12:27.660 SERIAL » 0x0105006303a933                                                     (7 bytes)
18:12:27.661 DRIVER » [REQ] [ReplaceFailedNode]
                        payload: 0x03a9
18:12:27.667 SERIAL « [ACK]                                                                   (0x06)
18:12:27.670 SERIAL « 0x010401630099                                                       (6 bytes)
18:12:27.670 SERIAL » [ACK]                                                                   (0x06)
18:12:27.672 DRIVER « [RES] [ReplaceFailedNode]
                        payload: 0x00
18:12:31.043 SERIAL « 0x01050063a90333                                                     (7 bytes)
18:12:31.043 SERIAL » [ACK]                                                                   (0x06)
18:12:31.045 DRIVER « [REQ] [ReplaceFailedNode]
                        payload: 0xa903
18:12:31.047 DRIVER   handling request ReplaceFailedNode (99)
18:12:31.048 DRIVER     1 handler registered!
18:12:31.048 DRIVER     invoking handler #0
18:12:31.048 CNTRLR   handling replace node request (status = FailedNodeReplace)
18:12:31.049 CNTRLR   The failed node is ready to be replaced, inclusion started...
18:12:31.059 DRIVER       the message was handled
18:14:24.211 CNTRLR   stopping inclusion process...
18:14:24.225 SERIAL » 0x0105004ac5aadf                                                     (7 bytes)
18:14:24.226 DRIVER » [REQ] [AddNodeToNetwork]
                        payload: 0xc5aa
18:14:24.233 SERIAL « [ACK]                                                                   (0x06)
18:14:24.239 SERIAL « 0x0107004aaa0615000b                                                 (9 bytes)
18:14:24.240 SERIAL » [ACK]                                                                   (0x06)
18:14:24.242 DRIVER « [REQ] [AddNodeToNetwork]
                        payload: 0xaa061500
18:14:24.244 DRIVER   handling request AddNodeToNetwork (74)
18:14:24.245 DRIVER     1 handler registered!
18:14:24.245 DRIVER     invoking handler #0
18:14:24.245 CNTRLR   handling add node request (status = Done)
18:14:24.245 CNTRLR   done called for 21
18:14:24.246 CNTRLR   the inclusion process was stopped
18:14:24.254 DRIVER       the message was handled
```

zwavejs2mqtt log from same time slice
```
2021-03-04 18:12:24.203 DEBUG SOCKET: Event ZWAVE_API emitted to 39aqHARkdf3DsT9qAAAL
2021-03-04 18:12:24.210 INFO ZWAVE: Calling api replaceFailedNode with args: [ 3, true, [length]: 2 ]
2021-03-04 18:12:31.050 INFO ZWAVE: Controller status: Secure inclusion started
2021-03-04 18:12:31.052 DEBUG MQTT: Publishing to zwavejs2mqtt/_EVENTS/ZWAVE_GATEWAY-zwavejs2mqtt/controller/inclusion_started: { data: [ true, [length]: 1 ] } with options { qos: 1, retain: false }
2021-03-04 18:12:31.061 INFO ZWAVE: Success zwave api call replaceFailedNode { success: true, message: 'Success zwave api call', result: true }
2021-03-04 18:14:18.527 DEBUG SOCKET: User disconnected 39aqHARkdf3DsT9qAAAL
2021-03-04 18:14:18.538 INFO APP: GET /control-panel 304 2.968 ms - -
2021-03-04 18:14:18.574 INFO APP: GET /static/css/app.addc55e5a7c06b2e220d.css 304 4.915 ms - -
2021-03-04 18:14:18.578 INFO APP: GET /static/js/app.7ac2d44817e089e0e76f.js 304 5.966 ms - -
2021-03-04 18:14:18.884 INFO APP: POST /api/authenticate 200 8.727 ms - 208
2021-03-04 18:14:18.891 INFO APP: GET /api/auth-enabled 304 7.997 ms - -
2021-03-04 18:14:18.905 INFO APP: GET /static/favicons/favicon-32x32.png 200 2.635 ms - 1831
2021-03-04 18:14:19.080 INFO APP: GET /static/logo.png 304 2.056 ms - -
2021-03-04 18:14:19.101 DEBUG SOCKET: New connection KQOLolOuuG-vZ2oVAAAN
2021-03-04 18:14:19.109 INFO APP: GET /static/fonts/MaterialIcons-Regular.fa3334f.woff2 304 3.535 ms - -
2021-03-04 18:14:19.121 DEBUG SOCKET: Event INITED emitted to KQOLolOuuG-vZ2oVAAAN
2021-03-04 18:14:24.247 INFO ZWAVE: Controller status: Inclusion stopped
```

scyto, 2021-03-08:
what should i do at this point, just exclude it and then include normally?

AlCalzone, 2021-03-08:
You could build your container with the branch in #2021 and just do another replace (secure again)

scyto, 2021-03-08:
will try that, i assume i have to power down the node so it 'fails' the is node alive check (i have been watching the logs careful :-) ) then when the log indicates inclusion is ready to be done power node up and press action.

AlCalzone, 2021-03-08:
You need to reset the device in order to be able to include it / replace it. Probably best if you shut down zwavejs first to avoid the case that the device sends a ResetLocally notification.

scyto, 2021-03-09:
yes that fixed it, i also had to do an exclude on the device after the factory reset (i used a random aeotec key i have lying around).

PR #2021 (2021-03-09): fix: do not remove node from other node's association during replace
With this PR we no longer do the cleanup work of removing a node from other node's associations of the removal is part of a **replace failed node** process. Cleaning up interferes with the security bootstrapping and can be unexpected since the node ID will still be part of the network after replacing.

fixes: #2020
(if only it were that easy)
closed 2021-03-09 by scyto