zwave-js/zwave-js #2023: Notification CC: automatic idle reset after 5 minutes not happening

AlCalzone, 2021-03-09:
https://github.com/home-assistant/core/issues/46982

closed 2021-03-09 by AlCalzone

macf0x, 2021-03-30:
@AlCalzone. I'm not 100% sure this addresses all the instances of auto idle.

See below **113-0-Home Security-unknown** does not auto-idle, where **113-0-Home Security-Sensor status** does.

2021-03-30 22:10:13.953 INFO ZWAVE: Node 19: value updated: 50-1-value-65537 706.233 => 706.234
2021-03-30 22:10:14.083 INFO ZWAVE: Node 19: value updated: 50-2-value-65537 5182.076 => 5182.089
2021-03-30 22:10:14.164 INFO ZWAVE: Node 19: value updated: 50-3-value-65537 2882.501 => 2882.503
2021-03-30 22:10:23.877 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:24.161 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:26.912 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:28.012 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:29.401 INFO ZWAVE: Node 24: value updated: 48-0-Any true => false
2021-03-30 22:10:29.420 INFO ZWAVE: Node 24: value added: 113-0-Home Security-Sensor status => 2
2021-03-30 22:10:29.421 INFO ZWAVE: Node 24: value added 24-113-0-Home Security-Sensor status => 2
2021-03-30 22:10:33.774 INFO ZWAVE: Node 24: value updated: 48-0-Any false => true
2021-03-30 22:10:33.798 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-03-30 22:10:37.869 INFO ZWAVE: Node 24: value updated: 48-0-Any true => false
2021-03-30 22:10:37.892 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-03-30 22:10:39.257 INFO ZWAVE: Node 24: value updated: 48-0-Any false => true
2021-03-30 22:10:39.284 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-03-30 22:10:43.094 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:44.205 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-unknown 254 => 254
2021-03-30 22:10:46.404 INFO ZWAVE: Node 24: value updated: 48-0-Any true => false
2021-03-30 22:10:46.411 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-03-30 22:10:47.784 INFO ZWAVE: Node 24: value updated: 48-0-Any false => true
2021-03-30 22:10:47.806 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-03-30 22:11:06.584 INFO ZWAVE: Node 4: value updated: 128-0-level 100 => 100
2021-03-30 22:11:06.586 INFO ZWAVE: Node 4: value updated: 128-0-isLow false => false
2021-03-30 22:11:06.792 INFO ZWAVE: Node 4: metadata updated: 49-0-Illuminance
2021-03-30 22:11:06.794 INFO ZWAVE: Node 4: value updated: 49-0-Illuminance 0 => 0
2021-03-30 22:11:08.047 INFO ZWAVE: Node 4: metadata updated: 49-0-Air temperature
2021-03-30 22:11:08.049 INFO ZWAVE: Node 4: value updated: 49-0-Air temperature 19.5 => 19.5
2021-03-30 22:11:08.289 INFO ZWAVE: Node 4: metadata updated: 49-0-Humidity
2021-03-30 22:11:08.291 INFO ZWAVE: Node 4: value updated: 49-0-Humidity 29 => 30
2021-03-30 22:11:13.497 INFO ZWAVE: Node 19: value updated: 50-0-value-66049 1060.955 => 1064.084
2021-03-30 22:11:13.573 INFO ZWAVE: Node 19: value updated: 50-0-value-65537 8770.826 => 8770.842
2021-03-30 22:11:13.679 INFO ZWAVE: Node 19: value updated: 50-1-value-66049 78.013 => 78.579
2021-03-30 22:11:13.775 INFO ZWAVE: Node 19: value updated: 50-2-value-66049 814.862 => 817.226
2021-03-30 22:11:13.862 INFO ZWAVE: Node 19: value updated: 50-3-value-66049 168.08 => 168.279
2021-03-30 22:11:13.971 INFO ZWAVE: Node 19: value updated: 50-1-value-65537 706.234 => 706.235
2021-03-30 22:11:14.106 INFO ZWAVE: Node 19: value updated: 50-2-value-65537 5182.089 => 5182.102
2021-03-30 22:11:14.158 INFO ZWAVE: Node 19: value updated: 50-3-value-65537 2882.503 => 2882.505
2021-03-30 22:12:13.511 INFO ZWAVE: Node 19: value updated: 50-0-value-66049 1064.084 => 1074.906
2021-03-30 22:12:13.574 INFO ZWAVE: Node 19: value updated: 50-0-value-65537 8770.842 => 8770.858
2021-03-30 22:12:13.672 INFO ZWAVE: Node 19: value updated: 50-1-value-66049 78.579 => 78.146
2021-03-30 22:12:13.780 INFO ZWAVE: Node 19: value updated: 50-2-value-66049 817.226 => 828.88
2021-03-30 22:12:13.881 INFO ZWAVE: Node 19: value updated: 50-3-value-66049 168.279 => 167.88
2021-03-30 22:12:13.966 INFO ZWAVE: Node 19: value updated: 50-1-value-65537 706.235 => 706.236
2021-03-30 22:12:14.076 INFO ZWAVE: Node 19: value updated: 50-2-value-65537 5182.102 => 5182.115
2021-03-30 22:12:14.192 INFO ZWAVE: Node 19: value updated: 50-3-value-65537 2882.505 => 2882.507
2021-03-30 22:12:21.079 INFO ZWAVE: Node 13: value updated: 48-0-Any true => false
2021-03-30 22:13:13.510 INFO ZWAVE: Node 19: value updated: 50-0-value-66049 1074.906 => 1085.327
2021-03-30 22:13:13.581 INFO ZWAVE: Node 19: value updated: 50-0-value-65537 8770.858 => 8770.874
2021-03-30 22:13:13.681 INFO ZWAVE: Node 19: value updated: 50-1-value-66049 78.146 => 78.246
2021-03-30 22:13:13.784 INFO ZWAVE: Node 19: value updated: 50-2-value-66049 828.88 => 839.135
2021-03-30 22:13:13.888 INFO ZWAVE: Node 19: value updated: 50-3-value-66049 167.88 => 167.946
2021-03-30 22:13:14.009 INFO ZWAVE: Node 19: value updated: 50-1-value-65537 706.236 => 706.237
2021-03-30 22:13:14.315 INFO ZWAVE: Node 19: value updated: 50-2-value-65537 5182.115 => 5182.128
2021-03-30 22:13:14.381 INFO ZWAVE: Node 19: value updated: 50-3-value-65537 2882.507 => 2882.509
2021-03-30 22:13:16.322 INFO ZWAVE: Node 13: value updated: 48-0-Any false => true
[... cut ...]
2021-03-30 22:15:13.991 INFO ZWAVE: Node 19: value updated: 50-1-value-65537 706.239 => 706.242
2021-03-30 22:15:14.084 INFO ZWAVE: Node 19: value updated: 50-2-value-65537 5182.141 => 5182.154
2021-03-30 22:15:14.192 INFO ZWAVE: Node 19: value updated: 50-3-value-65537 2882.511 => 2882.513
2021-03-30 22:15:47.811 INFO ZWAVE: Node 24: value updated: 113-0-Home Security-Sensor status 2 => 0
2021-03-30 22:16:13.536 INFO ZWAVE: Node 19: value updated: 50-0-value-66049 1174.961 => 1178.591
2021-03-30 22:16:13.612 INFO ZWAVE: Node 19: value updated: 50-0-value-65537 8770.909 => 8770.927

AlCalzone, 2021-03-30:
> **113-0-Home Security-unknown** does not auto-idle

Yes, because that is not a known notification. Probably shouldn't even exist. To understand why it keeps popping up, I'll need to see a zwave-js log with loglevel debug.

macf0x, 2021-04-11:
In OZW this was supported. When you say not known notification, is that a device definition parameter?

With the most recent version 7.1.1 neither value is idling after 5 minutes.

I apologize if these logs do not have the necessary info. I'm having some issues with zwavejs2mqqt exporting logs.

If that is the case, I'll post updated logs once that is addressed.

[zwavejs2mqtt-store (2).zip](https://github.com/zwave-js/node-zwave-js/files/6291890/zwavejs2mqtt-store.2.zip)
[zwavejs2mqtt-store (3).zip](https://github.com/zwave-js/node-zwave-js/files/6291891/zwavejs2mqtt-store.3.zip)

AlCalzone, 2021-04-11:
> When you say not known notification, is that a device definition parameter?

No, the Z-Wave Alliance defines which combination of notification type and notification event means what: https://github.com/zwave-js/specs/blob/master/SDS13713%20Notification%20Command%20Class%2C%20list%20of%20assigned%20Notifications.xlsx

Notification event `254` is always labeled as an `unknown` notification. That means it isn't clear whether this is an event or a variable, and whether it may be idled if it is actually a variable. So `zwave-js` just tells you that it was received but does not make an assumption about idling.

> In OZW this was supported

That's a bad argument :) OZW supported many things but also did many things wrong.

> With the most recent version 7.1.1 neither value is idling after 5 minutes.

Yes, see https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#710-2021-04-05
This behavior is now opt-in on a per-device basis, because it broke motion sensors that don't continuously refresh their status.

I took a look at your logs. Neither of them seem to be from the newest version, so I can't confirm. Anyways, we can add the flag to the device files if you tell me which devices are affected. Node IDs plus the `<homeid>.values.jsonl` file from the store directory should suffice.

PR #2024 (2021-03-09): fix(cc): auto-idle notification variables after 5 minutes, not 5 hours
fixes: #2023