zwave-js/zwave-js #2031: V7 Alpha 1 : Unbale to output a log.

marcus-j-davies, 2021-03-10:
I would attach a log - to detail a problem with the log 🤣 .

The console is all glitched.
setting the ENV `LOGTOFILE` doesn't anything.

Apologies, if I am missing something here? 

[image]

AlCalzone, 2021-03-10:
Seems that the env variable no longer works. The recommended approach via driver options does:
https://zwave-js.github.io/node-zwave-js/#/api/driver?id=zwaveoptions
https://zwave-js.github.io/node-zwave-js/#/api/driver?id=logconfig

marcus-j-davies, 2021-03-10:
Cool,

On the school run currently.

Will give that a go soon
This is to grab a log for another potential early bug for V7.

closed 2021-03-10 by AlCalzone