zwave-js/zwave-js #2042: GE Switch event doesn't distinguish between single and double tap

skyw33, 2021-03-11:
**Checklist:**

- [ x ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ x ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ x ] I have checked the troubleshooting section and my problem is **not** described there.
- [ x ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.
If I look at the event stream a single tap and double tap of my GE switch results in the same "Basic" output.

**Device information**

Which device(s) is/are affected (make/model)?  GE/Jasco 26932 / 26933 / ZW3008

What are the node IDs? Node ID 8

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [ x ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ x ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ x ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Look at event stream in Home Assistant
2. Tap the GE switch once and look at event stream.  Tap the GE twice and look at event stream.
3. Look at event stream and see that single and double taps produce the same "Basic" output.

**Additional context**

App Version
2.1.0
Zwavejs Version
6.6.0
Zwavejs-server Version
1.1.0

**Logfile:**
Includes switch event, interview log and node export:
[ge_switch.zip](https://github.com/zwave-js/node-zwave-js/files/6122907/ge_switch.zip)

AlCalzone, 2021-03-11:
Please make a **zwave-js** log, loglevel `debug` or `silly` and attach it here as a file.
The archive only contains some HA log format I can't use.

skyw33, 2021-03-11:
[log.txt](https://github.com/zwave-js/node-zwave-js/files/6123413/log.txt)
My log to file function doesn't seem to be working so I selected "silly" and then copy and pasted from the debug window after interviewing the node.

AlCalzone, 2021-03-11:
There seems to be something broken - that's not a `silly` log either.

AlCalzone, 2021-03-11:
@robertsLando any idea?

kpine, 2021-03-11:
Is it possible that these motion switches do not natively support double tap events? I don't think all GE switches necessarily do. The DB and manual are not specific to whether the association refers to a single or double tap ("Supports Basic Set and is controlled by pressing On/Off button with local load"). It's possible it's only a single button press, which would explain the behavior seen.

I was searching around for SmartThings device handlers, and the one I [found](https://github.com/MichaelStruck/SmartThingsPublic/blob/87bfb8e1cea93e6dfa1fcf249aa0eab9db45c3e6/devicetypes/michaelstruck/ge-motion-dimmer-switch-26933.src/ge-motion-dimmer-switch-26933.groovy#L265) (via [here](http://thingsthataresmart.wiki/index.php?title=GE_Z-Wave_Motion_Sensor_Switch)) handles the basic set command and just counts the number of basic sets within a time interval (latest version is triple tap, prior versions were double tap). It's not the same thing as the "double tap" group.

>  If I compare that to a switch that works properly, a single tap produces a "Central Scene" event, while a double tap produces a "Basic" event.

Are you comparing to the same exact models of switches? This 26933 does not support Central Scene, according to the info zwave-js discovered, and the product DB [entry](https://products.z-wavealliance.org/products/2108/classes).

skyw33, 2021-03-11:
I'm comparing another GE model that I own which works with double tap.  

And you may be right that it does not support double tap natively, but I'm having trouble confirming that.

blhoward2, 2021-03-11:
That may be true and if so we can undo the compat flag, but that means this switch would only be working via basic set and not binary_switch or multilevel.

blhoward2, 2021-03-11:
Oh, also make sure your controller is not in group 2, as that will cause it to receive a basic set on switching. The controller should be in group 1 (lifeline) and group 3 (double tap).

skyw33, 2021-03-11:
I confirmed that the controller is associated for groups 1 and 3 but not 2.
[image]

skyw33, 2021-03-11:
Finally figured out how to generate a proper log...
[store.zip](https://github.com/zwave-js/node-zwave-js/files/6125774/store.zip)

kpine, 2021-03-11:
Seems the logs do not include any actual button presses, or did I miss it?

skyw33, 2021-03-11:
Hmm, I turned on silly logs, tried single and double tapping a bunch of times and then saved the logs.  I just tried again but I'm not sure what I'm looking for so attaching new ones...
[store (1).zip](https://github.com/zwave-js/node-zwave-js/files/6126066/store.1.zip)

blhoward2, 2021-03-12:
It doesn't appear to be doing as you describe to me:

```
21:11:39.464 CNTRLR   [Node 008] [~] [Multilevel Switch] currentValue: 99 => 5          [Endpoint 1]
21:11:39.469 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:45.009 CNTRLR   [Node 008] [~] [Multilevel Switch] currentValue: 5 => 99          [Endpoint 1]
21:11:45.016 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:47.469 DRIVER « [Node 008] [REQ] [ApplicationUpdateRequest]
21:11:47.471 CNTRLR « [Node 008] Received updated node info
21:11:48.110 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:48.111 CNTRLR   [Node 008] treating BasicCC Set as a value event
21:11:48.112 CNTRLR   [Node 008] [!] [Basic] event: 0                                   [Endpoint 1]
21:11:48.132 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:48.134 CNTRLR   [Node 008] [~] [Notification] Home Security[Motion sensor status] [Endpoint 2]
21:11:48.153 CNTRLR   [Node 008] [~] [Multilevel Switch] currentValue: 99 => 0          [Endpoint 1]
21:11:48.161 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:52.686 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:11:52.688 CNTRLR   [Node 008] [~] [Notification] Home Security[Motion sensor status] [Endpoint 2]
21:12:15.967 DRIVER « [Node 008] [REQ] [ApplicationCommand]
21:12:15.971 CNTRLR   [Node 008] [~] [Notification] Home Security[Motion sensor status] [Endpoint 2]
```

It only sends a basic set event once. The remainder of the time it is sending a Multilevel switch set.

skyw33, 2021-03-12:
Ok, apologies -- I was basing my information on the zwave event stream.  From the event stream, I was only seeing Basic commands.  .... But .... home assistant automations act on the event stream so I can't distinguish between tap vs double tap from Home Assistant.  There seems to be a disconnect between the event stream and the zwavejs log.

blhoward2, 2021-03-12:
I'm not sure why that would be. You'd have to open an issue on the integration. They'll need to see the log showing what the node is sending and then the logs from HA so they can figure it out. On our end it appears fine.

kpine, 2021-03-12:
It doesn't appear as if "double tap" is a thing for this switch. I think you'll have to implement your own automation that counts the number of basic sets received in some time interval, like the device handler I posted does.

blhoward2, 2021-03-12:
But he's getting a double tap 0 there via the basic set. @skyw33 can you log yourself doing a double up and a double down so we can see if we need to remove it from the device?

blhoward2, 2021-03-12:
I suspect this is the issue. You have two types of notifications coming through with 0/255. Double taps and motion:

```
21:11:27.165 DRIVER « [Node 008] [REQ] [ApplicationCommand]
                      └─[MultiChannelCCCommandEncapsulation]
                        │ source:      2
                        │ destination: 0
                        └─[NotificationCCReport]
                            notification type:   Home Security
                            notification status: 255
                            notification event:  Unknown (0x00)
```

You need to try to differentiate in the trigger.

kpine, 2021-03-12:
The Home Security notification is not a Basic event.

blhoward2, 2021-03-12:
I don't think that he's getting two types of basic set events. I think he's getting multiple 0/255 events because of the motion.

kpine, 2021-03-12:
@skyw33 It might help to have a more methodical approach, instead of just tapping the buttons a bunch of times. It's really difficult to correlate anything without knowing what is what. My suggestion would be to get a new log file, and do something like this:

1. Make sure the motion detection is disarmed (leave the room or whatever).
2. Trigger the motion detection w/o touching the buttons. Wait a few seconds.
2. Single tap a button. Wait a few seconds.
3. Double tap a button. Done.

If you do something like that and note the times you performed each step, it should be a lot easier to sort out the Basic Sets from the Notifications, and see which Basic Sets go with which tap actions.

AlCalzone, 2021-03-12:
I think @blhoward2 has it correct, except the Notification which should not affect the Basic event. Nonetheless it makes sense to generate a log where we know exactly what should happen and when like @kpine proposed.

skyw33, 2021-03-12:
I was able to sneak up behind it without triggering motion.  First, I tapped once, then waited 10 seconds, then tapped twice.  That's it.

[store (2).zip](https://github.com/zwave-js/node-zwave-js/files/6129531/store.2.zip)

AlCalzone, 2021-03-12:
Looks both identical to me

blhoward2, 2021-03-12:
Ok, so we'll remove the double tap. Works for me.

skyw33, 2021-03-12:
You guys are pros.  Thanks for investigating this.

PR #2055 (2021-03-13): fix(config): remove unsupported double tap on GE 26932; add double tap to 12730; fix parameters
Closes #2042. Also removes errnonous parameters reported by @freshcoast, and adds double tap support to the 12730 fan controller.
closed 2021-03-13 by AlCalzone