zwave-js/zwave-js #2047: 6.6.1 fails to initialize driver: "this.container.shouldLogNode is not a function"

cbowns, 2021-03-12:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

`zwave-js-server` fails to restore the network from cache or start the driver, due to an incorrect declaration of `shouldLogNode`:

```
Error in driver ZWaveError: Restoring the network from cache failed: TypeError: this.container.shouldLogNode is not a function
    at Driver.restoreNetworkStructureFromCache (/Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:2518:5)
    at /Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:817:6
    at ZWaveController.interview (/Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/controller/Controller.ts:485:3)
    at Driver.initializeControllerAndNodes (/Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:814:4)
    at Immediate.<anonymous> (/Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:731:5) {
  code: 12,
  context: undefined,
  transactionSource: undefined
}

```

**Device information**

Affected device: GoControl Z-Wave stick

**Did you change anything?**

- [x] **Yes**: *(please describe)*

Updated from `339d0bf (tag: 1.0.0-beta.6)` to `c1ddbd1 (HEAD, tag: 1.1.1) version 1.1.1`

- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *(specify application with versions)*
  
Yes, with 1.0.0-beta.6

**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [x] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

I'm running zwave-js server as a websocket for HA via https://github.com/zwave-js/zwave-js-server:

1. Launch `server.ts` via node with a GoControl stick: `node_modules/.bin/ts-node src/bin/server.ts /dev/tty.GoControl_zwave`
2. Launch succeeds, but driver init fails

**Additional context**

I think? [it's this line](https://github.com/zwave-js/node-zwave-js/blob/6101b8351343db76b4cb0f349147f2c6bf5b941c/packages/zwave-js/src/lib/log/Driver.ts#L126), but I'm sorta new to TS and don't know how to map TS's backtrace here: `Driver.restoreNetworkStructureFromCache (/Users/cbowns/Development/zwave-js-server/node_modules/zwave-js/src/lib/driver/Driver.ts:2518:5)` to a line number in the underlying file.

**Logfile:**

[zwave-startup.log](https://github.com/zwave-js/node-zwave-js/files/6126695/zwave-startup.log)

cbowns, 2021-03-12:
This works for me in 6.2.0 (which is 1.0.0-beta-6 for zwave-js-server).

AlCalzone, 2021-03-12:
This sounds like an incompatible dependency. Please run `npm ls` and post the output.

cbowns, 2021-03-12:
tl;dr toasted `node_modules` after seeing zwave-js 6.4.0 launched despite `npm` showing 6.6.1 as the dep. now the driver works great. something something node???

---

I checked out `e0d5dec` on `zwave-js-server`, which declares `zwave-js@6.6.1` as a dep:

```
@zwave-js/server@1.1.1 /Users/cbowns/Development/zwave-js-server
├── @types/minimist@1.2.1
├── @types/node@14.14.32
├── @types/ws@7.4.0
├── @types/yargs@16.0.0
├── @typescript-eslint/eslint-plugin@4.16.1
├── @typescript-eslint/parser@4.16.1
├── eslint-config-prettier@8.1.0
├── eslint@7.21.0
├── husky@4.3.8
├── lint-staged@10.5.4
├── minimist@1.2.5
├── prettier@2.2.1
├── ts-node@9.1.1
├── typescript@4.2.3
├── ws@7.4.4
└── zwave-js@6.6.1
```

and it's using 6.6.1 just fine now.

```
╰─ node_modules/.bin/ts-node src/bin/server.ts /dev/tty.GoControl_zwave                                                                    ─╯
                                                                                                                                [15:39:35]
15:39:39.085 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
15:39:39.089 DRIVER   version 6.6.1
15:39:39.090 DRIVER   
15:39:39.090 DRIVER   starting driver...
15:39:39.102 DRIVER   opening serial port /dev/tty.GoControl_zwave
15:39:39.115 DRIVER   serial port opened
…
```

closed 2021-03-12 by cbowns