zwave-js/zwave-js #2048: FGBS222 - not setting configuration parameters

Masterz69, 2021-03-12:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X ] I have checked the troubleshooting section and my problem is **not** described there.
- [ X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Can't change device configuration parameters, f.e. periodic report interval.
I set to 20, but it still remains 0. Parameter 64/66/68.

**Device information**

Which device(s) is/are affected (make/model)?  Fibaro Smart Implant FGBS222
What are the node IDs? Node 4

**Did you change anything?**
- [X ] **No** Just added device no network.

**Did this use to work before?**
- [ X] **Don't know**, this is a new device
    
**How are you using `node-zwave-js`**
- [X ] `HomeAssistant` version OS 5.12 core-2021.3.3

Zwavejs Version 6.6.1
Zwavejs-server Version 1.1.1

**To Reproduce**

Steps to reproduce the behavior:
ZWave JS -> Node 4 -> Configuration -> [4-112-0-66] Temperatur sensor period
Set to 20, press Arrow. And... still 0 .
Same for 64 & 68 parameters.

Some other parameter, like 20 & 21 - looks changing successefully.

**Logfile:**

2021-03-12 10:20:41.853 DEBUG SOCKET: Event ZWAVE_API emitted to ZO-tJFOQBNyYbr5QAAAY
2021-03-12 10:20:41.870 INFO ZWAVE: Calling api writeValue with args: [
  { nodeId: 4, commandClass: 112, endpoint: 0, property: 66 },
  20,
  [length]: 2
]
2021-03-12 10:20:41.873 INFO ZWAVE: Writing 20 to 4-112-0-66
2021-03-12 10:20:41.968 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-03-12 10:20:43.144 INFO ZWAVE: Node 4: value updated: 112-0-66 20 => 0
2021-03-12 10:20:46.926 DEBUG SOCKET: Event ZWAVE_API emitted to ZO-tJFOQBNyYbr5QAAAY
2021-03-12 10:20:46.930 INFO ZWAVE: Calling api writeValue with args: [
  { nodeId: 4, commandClass: 112, endpoint: 0, property: 64 },
  20,
  [length]: 2
]
2021-03-12 10:20:46.932 INFO ZWAVE: Writing 20 to 4-112-0-64
2021-03-12 10:20:47.033 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-03-12 10:20:48.214 INFO ZWAVE: Node 4: value updated: 112-0-64 20 => 0
2021-03-12 10:20:54.712 DEBUG SOCKET: Event ZWAVE_API emitted to ZO-tJFOQBNyYbr5QAAAY
2021-03-12 10:20:54.717 INFO ZWAVE: Calling api writeValue with args: [
  { nodeId: 4, commandClass: 112, endpoint: 0, property: 68 },
  20,
  [length]: 2
]
2021-03-12 10:20:54.719 INFO ZWAVE: Writing 20 to 4-112-0-68
2021-03-12 10:20:54.808 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-03-12 10:20:55.981 INFO ZWAVE: Node 4: value updated: 112-0-68 20 => 0
2021-03-12 10:21:03.795 DEBUG SOCKET: Event ZWAVE_API emitted to ZO-tJFOQBNyYbr5QAAAY
2021-03-12 10:21:03.798 INFO ZWAVE: Calling api writeValue with args: [
  { nodeId: 4, commandClass: 112, endpoint: 0, property: 20 },
  1,
  [length]: 2
]

[settings_20210312.zip](https://github.com/zwave-js/node-zwave-js/files/6128714/settings_20210312.zip)
[fgbs222_log_01.zip](https://github.com/zwave-js/node-zwave-js/files/6128718/fgbs222_log_01.zip)

marcus-j-davies, 2021-03-12:
No experience with your device, or environment - so this may be irrelevant 
but the manual states (for 64,66,68)

[image]
 
 
 So 20 wont be accepted. Just a thought.

AlCalzone, 2021-03-12:
This 👆🏻

Masterz69, 2021-03-12:
Yes, You right.
Unfortuantely this notice not included in parameter description in GUI.
[image]

It acccepted 60 with any issues.
Thank You very much.

Masterz69, 2021-03-12:
As well as desctiption and comments does not match :)
External sensors (par.68) <> analog inputs (par.64). 

Copy/paste issues, looks like :)

closed 2021-03-12 by Masterz69

AlCalzone, 2021-03-12:
Unfortunately there's not an easy way to model gaps in the allowed value range. As for the label, we can look at that.

blhoward2, 2021-03-13:
Interestingly, that copy/paste error actually exists in the manual: "This parameter defines reporting period of analog inputs value. "