zwave-js/zwave-js #2054: GE 14287 (others?) do not associate the controller with Group 3 automatically despite lifeline flag

blhoward2, 2021-03-13:
**Describe the bug**

The GE 14287 fan controller has a device config file specifying that Group 3 isLifeline: true. While troubleshooting another issued I noted my controller was nevertheless not associated with Group 3. Adding it manually works fine. Is this a problem on other devices?

```
	"associations": {
		"1": {
			"label": "Group 1",
			"maxNodes": 5,
			"isLifeline": true
		},
		"2": {
			"label": "Group 2",
			"maxNodes": 5
		},
		"3": {
			"label": "Double Tap",
			"maxNodes": 5,
			"isLifeline": true
		}
	},
```

**Device information**

Which device(s) is/are affected (make/model)?  GE 14287 - others?
What are the node IDs? 15

**How are you using `node-zwave-js`**
App Version
2.1.0+6858a08.6858a08
Zwavejs Version
6.6.0-ccce8b0
Zwavejs-server Version
1.1.0

**Logfile:**

```
22:49:54.940 CNTRLR » [Node 015] Association group #3: Querying name...
22:49:54.947 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      65
                      └─[AssociationGroupInfoCCNameGet]
                          group id: 3
22:49:54.955 DRIVER « [RES] [SendData]
                        was sent: true
22:49:55.058 DRIVER « [REQ] [SendData]
                        callback id:     65
                        transmit status: OK
22:49:55.141 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCNameReport]
                          group id: 3
                          name:     Fan control(Key2)
22:49:55.144 CNTRLR « [Node 015] Association group #3 has name "Fan control(Key2)"
22:49:55.144 CNTRLR » [Node 015] Association group #3: Querying info...
22:49:55.151 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      66
                      └─[AssociationGroupInfoCCInfoGet]
                          group id:      3
                          refresh cache: false
22:49:55.159 DRIVER « [RES] [SendData]
                        was sent: true
22:49:55.264 DRIVER « [REQ] [SendData]
                        callback id:     66
                        transmit status: OK
22:49:55.337 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCInfoReport]
                          is list mode:     false
                          has dynamic info: false
                          groups:
                          · Group #3
                            mode:       0
                            profile:    8194
                            event code: 0
22:49:55.339 CNTRLR « [Node 015] Received info for association group #3:
                      info is dynamic: false
                      profile:         Control: Key 02
22:49:55.339 CNTRLR » [Node 015] Association group #3: Querying command list...
22:49:55.351 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      67
                      └─[AssociationGroupInfoCCCommandListGet]
                          group id:    3
                          allow cache: true
22:49:55.359 DRIVER « [RES] [SendData]
                        was sent: true
22:49:55.463 DRIVER « [REQ] [SendData]
                        callback id:     67
                        transmit status: OK
22:49:55.538 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCCommandListReport]
                          group id: 3
                          commands:
                          · Basic: 0x01
                          · Multilevel Switch: 0x04, 0x05
22:49:55.620 CNTRLR   [Node 015] Interview stage completed: CommandClasses
22:49:55.620 CNTRLR   [Node 015] Interview stage completed: OverwriteConfig
22:49:55.620 CNTRLR » [Node 015] requesting node neighbors...
```

Compare this with the lifeline group:

```
22:49:53.787 CNTRLR » [Node 015] Association group #1: Querying name...
22:49:53.793 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      59
                      └─[AssociationGroupInfoCCNameGet]
                          group id: 1
22:49:53.800 DRIVER « [RES] [SendData]
                        was sent: true
22:49:53.856 DRIVER « [REQ] [SendData]
                        callback id:     59
                        transmit status: OK
22:49:53.937 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCNameReport]
                          group id: 1
                          name:     Lifeline
22:49:53.939 CNTRLR « [Node 015] Association group #1 has name "Lifeline"
22:49:53.939 CNTRLR » [Node 015] Association group #1: Querying info...
22:49:53.945 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      60
                      └─[AssociationGroupInfoCCInfoGet]
                          group id:      1
                          refresh cache: false
22:49:53.952 DRIVER « [RES] [SendData]
                        was sent: true
22:49:54.064 DRIVER « [REQ] [SendData]
                        callback id:     60
                        transmit status: OK
22:49:54.138 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCInfoReport]
                          is list mode:     false
                          has dynamic info: false
                          groups:
                          · Group #1
                            mode:       0
                            profile:    1
                            event code: 0
22:49:54.141 CNTRLR « [Node 015] Received info for association group #1:
                      info is dynamic: false
                      profile:         General: Lifeline
22:49:54.142 CNTRLR » [Node 015] Association group #1: Querying command list...
22:49:54.150 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      61
                      └─[AssociationGroupInfoCCCommandListGet]
                          group id:    1
                          allow cache: true
22:49:54.157 DRIVER « [RES] [SendData]
                        was sent: true
22:49:54.263 DRIVER « [REQ] [SendData]
                        callback id:     61
                        transmit status: OK
22:49:54.336 DRIVER « [Node 015] [REQ] [ApplicationCommand]
                      └─[AssociationGroupInfoCCCommandListReport]
                          group id: 1
                          commands:
                          · Multilevel Switch: 0x03
                          · Device Reset Locally: 0x01

```

Why does it pickup the name from the config file for Group 1 (Lifeline) in the log, but not Group 3 (Double Tap)? Note that Group 3 is actually named correctly in zwavejs2mqtt, though.

blhoward2, 2021-03-13:
[node 15 interview.txt](https://github.com/zwave-js/node-zwave-js/files/6134138/node.15.interview.txt)

blhoward2, 2021-03-13:
Confirmed that this DOES work correctly on a GE 14294. It DOES work correctly if I manually add the controller to group 3 and the command succeeds (and double tap then works). Also verified this still exists on the latest docker (6.6.0-ccce8b0)

blhoward2, 2021-03-14:
Did you change something other than the logging? This worked correctly in the debug logging branch:

```
22:53:56.894 CNTRLR « [Node 015] received information for association group #3:
                      maximum # of nodes: 5
                      currently assigned nodes: 
22:53:56.895 CNTRLR   [Node 015] getLifelineGroupIds(node 15) => supports ZW+
22:53:56.895 CNTRLR   [Node 015] associations found in config
22:53:56.896 CNTRLR   [Node 015] lifelineGroups = [1,3]
22:53:56.896 CNTRLR   [Node 015] lifelineNodeIds(group 1) = [1]
22:53:56.897 CNTRLR   [Node 015] lifelineNodeIds(group 3) = []
22:53:56.897 CNTRLR » [Node 015] Controller missing from lifeline group #3, assigning ourselves...
22:53:56.907 DRIVER » [Node 015] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      104
                      └─[AssociationCCSet]
                          group id: 3
                          node ids: 1
22:53:56.916 DRIVER « [RES] [SendData]
                        was sent: true
22:53:57.015 DRIVER « [REQ] [SendData]
                        callback id:     104
                        transmit status: OK

```

AlCalzone, 2021-03-14:
Not that I know of. Probably had an old device config version?

AlCalzone, 2021-03-15:
So... no bug?

closed 2021-03-16 by AlCalzone