zwave-js/zwave-js #2066: Incorrect identification of device (SM103 instead of HSP02)

drlobo, 2021-03-14:
**Describe the bug**

My HSP02 device (motion sensor) is incorrectly identified as SM103 (door/window contact)

**Device information**

Which device(s) is/are affected (make/model)?  
Product Manufacturer: 0x0116
Product Identifiers: 0x0001:0x0001

**Did this use to work before?**
- [x] **Yes**,  with the OpenZWave database
    
**How are you using `node-zwave-js`**
- [x] `zwavejs2mqtt` (latest) docker image

**Additional context**

this entry is the device database is wrong
https://devices.zwave-js.io/?jumpTo=0x0116:0x0001:0x0001:0.0

here are the info about the HSP02
https://products.z-wavealliance.org/products/606

it's a rebranding of the the everspring HSP02
https://devices.zwave-js.io/?jumpTo=0x0060:0x0001:0x0001:0.0

the correct configuration can also be found in OpenZWave
https://github.com/OpenZWave/open-zwave/tree/master/config/manufacturer_specific.xml#L374

marcus-j-davies, 2021-03-14:
@blhoward2,

Was going to submit a PR (to change the label), but didn't want to step on your toes, if its already in sight?

marcus-j-davies, 2021-03-14:
@drlobo,

I guess the params are also incorrect?

drlobo, 2021-03-14:
yes, the parameters should the same as https://devices.zwave-js.io/?jumpTo=0x0060:0x0001:0x0001:0.0

blhoward2, 2021-03-14:
> @blhoward2,
> 
> 
> 
> Was going to submit a PR, but didn't want to step on your toes, if its already in sight?

I'm working on the Aeotec ones at the moment so feel free.

closed 2021-03-14 by AlCalzone