zwave-js/zwave-js #2073: [bug] Intermatic CA3750 Appliance Switch not configuring correctly

shillingcl, 2021-03-15:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 2.2.0
zwave-js version: Driver Version: 6.6.1 Server Version: 1.1.1

### Describe the bug

I have an Intermatic CA3750 Appliance switch that did pair OK, but it is missing the 2 switches that would allow me to control it.  Normally on the Vera it show it 1 "Parent" switch that didn't do anything, then two "Child" switches.  It does show status as complete and I did try reinterviewing.  I also notice in the control panel that the Manufacturer, ProductID, and ProductType for endpoint 1 and 2 all show as undefined.  I'm wondering if it doesn't have the config information it needs to build out those child devices.

### To Reproduce

Steps to reproduce the behavior:

1. Go to inclusion process to add Intermatic CA3750 Applicance Switch
2. Review added device which shows 1 binary switch, but missing the 2 switches that allow control

### Expected behavior

On my Vera it shows up with 3 switches which is what I was expecting here.  The top level "Parent" switch doesn't do anything, but the two "Child" switches would allow me to control the 2 separate loads.

### Additional context

Debug log files and node export attached
[zwavejs2mqtt_intermatic_ca3750.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6138527/zwavejs2mqtt_intermatic_ca3750.zip)

AlCalzone, 2021-03-15:
@shillingcl try re-interviewing the device (and make a log of that).
In your export I see values for endpoint 2 at least, but endpoint 1 seems to be missing completely.
Might be bad connectivity, might be something else.

shillingcl, 2021-03-15:
Thanks for taking a look.  I attached the log files and node export after reinterviewing.  As for connectivity, the switch is only about 6 feet away from the hub.  Until yesterday when I moved it, I had been been using it with my Vera for several years. 

[ca3750_reinterview.zip](https://github.com/zwave-js/node-zwave-js/files/6140964/ca3750_reinterview.zip)

AlCalzone, 2021-03-15:
Your device supports Multi Channel CC V1, which is a bit different from the newer versions. Looks like Binary Switch CC is not being detected as supported on endpoint 1, although the device reports it.

For the time being, you should see the value for endpoint 2 though:
`10-37-2-targetValue` to control the switch status and `10-37-2-currentValue` to read it.

AlCalzone, 2021-03-15:
Well, it turns out `>=` is entirely different from `<=`...

Can you test with branch in the linked PR? You'll have to re-interview the device again with the new build.
==> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

shillingcl, 2021-03-16:
Been there... Really appreciate all the quick replies...  My first time building a custom docker image, so I had a little learning to do.  It does look like it is built successfully with the fix branch added, but I won't likely be able to get it tested until Wednesday.  I'll let you know.

shillingcl, 2021-03-16:
I'm pretty sure I did the branch build of the docker image correctly, but I'm guessing you'll be able to confirm that in the logs.  From my perspective, the behavior seems the same as before although I'm not seeing 10-37-0 show up in the binary switch section anymore. Here's what I think I know from what I've seen here and as well as the way it worked on my Vera.

10-37-0: Switch for Load 1 - When it was showing up under the binary switches, I could select it and hear the contactor click
10-37-1: Switch for Load 2 - Never showed as a binary switch, so can't confirm, but of course this is the load I need to control
10-37-2: Dummy Switch - Seems to show every time, but does not serve any purpose (This worked the same in Vera)

Thanks again and let me know if there is anything else I can do to help.

[ca3750_TestFix.zip](https://github.com/zwave-js/node-zwave-js/files/6152783/ca3750_TestFix.zip)

AlCalzone, 2021-03-17:
Hm doesn't look like it is working yet. I'll try to recreate this in a test case.

LunkwillAndFook, 2021-05-18:
@AlCalzone Have you had an opportunity to look into this? I'm curious if it may have been resolved in another bug and whether there is something I need to do as a workaround.

AlCalzone, 2021-05-21:
@LunkwillAndFook I'll merge the referenced PR and release it as part of 7.5.1. Please test again with that version to make sure this is not just a problem with building the branch.

I'd also be interested in a new log with that version to verify.

LunkwillAndFook, 2021-05-23:
@AlCalzone thank you for looking into this. I will test and send logs as soon as I am able.

AlCalzone, 2021-06-11:
@LunkwillAndFook did you have any chance to test this yet? HA's version should include this fix now.

LunkwillAndFook, 2021-07-30:
@AlCalzone  I apologize that it took me so long to test this. I ended up setting up an entirely new instance (with separate hardware) of HomeAssistant so that I could get some clean logs around this and to ensure proximity wasn't an issue. Unfortunately the fix does not work. I did manage to capture ZwaveJS logs as well as logs from OpenZwave (which does work) so hopefully some compare and contrast will point to the error there.
[zwavejs_2021-07-30.log](https://github.com/zwave-js/node-zwave-js/files/6910156/zwavejs_2021-07-30.log)
[OpenZwaveLog.txt](https://github.com/zwave-js/node-zwave-js/files/6910157/OpenZwaveLog.txt)

geoffbon, 2021-12-12:
Has there been any progress on this one?  I am also struggling with this same device, via Home Assistant -> zwavejs2mqtt.
I'm using versions:
zwavejs2mqtt: 6.1.0
zwave-js: 8.9.0-beta.0

The node appears to have two switches, but the device does nothing when either one is triggered.   I've attached the log of the interview if it's useful.  Anything else I can try to help debug this?

[intermatic-ca3750-interview.txt](https://github.com/zwave-js/node-zwave-js/files/7699924/intermatic-ca3750-interview.txt)

geoffbon, 2022-01-02:
I just re-upgraded to be able to generate the appropriate log file, per [this](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs).  
Attaching the zwave log (no mqtt file generated).
Behavior is still the same: triggering the device from either the control panel or Home Assistant does nothing.
Any help would be great so that I can complete my migration
[zwavejs_2022-01-02.log](https://github.com/zwave-js/node-zwave-js/files/7799923/zwavejs_2022-01-02.log)
!

kpine, 2022-01-03:
ZJS still seems to be sending V2 encapsulated messages.

OZW encapsulated Binary Switch Get (from the logs in the earlier comment):
```
[20210730 17:29:51.629 EDT] [ozw.library] [info]: Info - Node: 3 Sending (Send) message (Callback ID=0x23, Expected Reply=0x04) - MultiInstance Encapsulated (instance=2): SwitchBinaryCmd_Get (Node=3): 0x01, 0x0c, 0x00, 0x13, 0x03, 0x05, 0x60, 0x06, 0x02, 0x25, 0x02, 0x25, 0x23, 0xa3 
```

ZJS encapsulated Binary Switch Get (from Geoff's interview log):
```
2022-01-02T23:29:17.112Z SERIAL » 0x010d00131a06600d00022502251e8e                                    (15 bytes)
2022-01-02T23:29:17.112Z DRIVER » [Node 026] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      30
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 2
                                    └─[BinarySwitchCCGet]
```

OZW is sending a V1 Encapsulation command (command 0x06 with no source endpoint) while ZJS is sending a V2+ command (command 0x0d with a source endpoint).

blhoward2, 2022-01-03:
Can you post a node export so we can see how the device is setup?

geoffbon, 2022-01-03:
Is this what you're looking  for?
[node_26.json.txt](https://github.com/zwave-js/node-zwave-js/files/7802365/node_26.json.txt)

AlCalzone, 2022-01-24:
You'll need a bit more patience - I don't have much time currently.

geoffbon, 2022-01-27:
Thanks Al, will check back!

closed 2022-03-21 by AlCalzone

LunkwillAndFook, 2022-04-10:
@AlCalzone Seems like progress but this still isn't working. Unfortunately there is no more OpenZwave as of the current version of HomeAssistant which means no more workaround. 

The behavior I see since your last attempted fix is:
- Device appears to configure itself correctly (two switch entities are created)
- Turning entities on or off does not have any impact on physical device (in either 2xSPST or 1xDPST modes)
- In 2xSPST mode, when the device is turned on physically a new disabled switch entity appear

Troubleshooting Information:
- Screenshot of 2xSPST device:  [image]
- Diagnostics from 1xDPST device: [zwave_js-173ca2119119b1095342056bca1b8f2d-Contactor_Switch-3117ac66ab6144d9922ac11124bf3c46.json.txt](https://github.com/zwave-js/node-zwave-js/files/8459763/zwave_js-173ca2119119b1095342056bca1b8f2d-Contactor_Switch-3117ac66ab6144d9922ac11124bf3c46.json.txt)
- ZWaveJS Log (silly): [zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/8459809/zwave_js.log)

Note that 2xSPST device fails to generate diagnostics (likely due to extra "ghost" entity created when device is manually turned on/off). Error in logs for failed diagnostic generation is:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 234, in get
    data = await info[sub_type.value](hass, config_entry, device)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/diagnostics.py", line 142, in async_get_device_diagnostics
    entities = get_device_entities(hass, node, device)
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/diagnostics.py", line 86, in get_device_entities
    zwave_value = node.values[value_id]
KeyError: '6-37-0-currentValue'
```

Log entries from Supervisor Add-On when turning switch entities on/off:
```
2022-04-10T19:46:30.080Z CNTRLR   [Node 006] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:46:32.998Z CNTRLR   [Node 006] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:46:50.625Z CNTRLR   [Node 006] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:46:51.725Z CNTRLR   [Node 006] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:47:03.166Z CNTRLR   [Node 004] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:47:04.424Z CNTRLR   [Node 004] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:47:17.303Z CNTRLR   [Node 004] Timed out while waiting for a response from the node (ZW0201)
2022-04-10T19:47:18.408Z CNTRLR   [Node 004] Timed out while waiting for a response from the node (ZW0201)
```

AlCalzone, 2022-04-11:
@LunkwillAndFook The fix for this is included in the v9 release, which is expected to land in HomeAssistant soon, but hasn't been released there yet AFAIK.

LunkwillAndFook, 2022-04-12:
@AlCalzone Apologies. Mistake on my part. I should have verified the release.

PR #2087 (2021-05-21): fix: correct comparison for Multi Channel CC V1
fixes: #2073

Well - this is embarassing 😅
PR #4387 (2022-03-21): fix: encapsulate with `Multi Channel CC` V1 when necessary
Looks like we didn't actually use MultiChannel CC V1 for encapsulation.

fixes: #2073