zwave-js/zwave-js #2074: [bug] inovelli scenes not working properly

legomind, 2021-03-15:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 6.6
zwave-js version: 1.1

### Describe the bug

Double tap events are sometimes reported as 2 single tap events. 
After a double tap, (reported as two single taps), then a 5 second delay, another tap generates a double tap event. 

### To Reproduce

Steps to reproduce the behavior:

1. Double tap the switch
2. Wait a few seconds
3. Tap switch once. 
4. Observe wrong value_notification. 

### Expected behavior

Appropriate scene events should be raised. 

### Additional context

AlCalzone, 2021-03-15:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

Although this terribly sounds like a device bug, since we don't track single vs. double taps and just work with whatever event the device gives us.

legomind, 2021-03-15:
@AlCalzone I actually agree with you about it being a device bug, if it were not for the fact these devices were working flawlessly for 2 years using open zwave.

Ill provide the log tonight when I get a chance to capture it.

closed 2021-03-26 by zwave-js-assistant[bot]

legomind, 2021-03-31:
Ok. Finally figured it out. There is a predictable pattern.

Scene triggered (1) -> No Event
Scene triggered (2) -> Event for Scene (1)
Scene triggered (3) -> Event for Scene (2)

Basically scenes events are only emitted after another scene is triggered.

This behavior does not exist using ozw, therefore it is not the devices.

This issue needs to be reopened.

tech165, 2021-04-01:
Hi, I have the exact same problem. Both with my Inovelli switches and HomeSeer ... was ok with OZW too. I’m on 2021.3.4 and zwavejs2mqtt ...

legomind, 2021-04-01:
See #2208