zwave-js/zwave-js #2076: [bug] UNKNOWN units in device METER CC

Masterz69, 2021-03-15:
### Version

HA OS 5.12 on RPi4 4G

[image]

### Describe the bug

UNKNOWN entries in Plug device METER section.
Initially no one, but now 9.

### To Reproduce

When device paired - no UNKNOWN entries.
After several Core & Host restart - appears.

### Expected behavior

Only entries actually exposed by device.

### Additional context

[image]

[node_2.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6085653/node_2.zip)

AlCalzone, 2021-03-15:
Ok, closing for now. If it reappears and you happen to catch it in a log, feel free to reopen.

closed 2021-03-15 by AlCalzone

Masterz69, 2021-03-15:
Sure. Not appeared anymore.