zwave-js/zwave-js #2078: Firmware OTA does not seem to work for Qubino device?

ualex73, 2021-03-15:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I got a newer firmware for my Qubino Smart Plug 16A and I am trying to upgrade the firmware via OTA (2.0 to 4.0). I see in the log it tries to start, but it seems to retry every time packet 1 (out of 2266).

The logfile shows as follows:
13:04:52.043 CNTRLR » [Node 007] Sending firmware fragment 1 / 2266
13:04:52.093 SERIAL » 0x01350013072e7a060001800001739c3a640095ff0275e40475e40175e4022275f (55 bytes)
                      f0275f8941f02184e1a341a39007fb8ef901813251c3c
13:04:52.094 DRIVER » [Node 007] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      28
                      └─[FirmwareUpdateMetaDataCCReport]
                          report #: 1
                          is last:  false
13:04:52.100 SERIAL « [ACK]                                                                   (0x06)
13:04:52.113 SERIAL « 0x0104011301e8                                                       (6 bytes)
13:04:52.116 SERIAL » [ACK]                                                                   (0x06)
13:04:52.117 DRIVER « [RES] [SendData]
                        was sent: true
13:04:52.141 SERIAL « 0x010700131c000004f3                                                 (9 bytes)
13:04:52.141 SERIAL » [ACK]                                                                   (0x06)
13:04:52.141 DRIVER « [REQ] [SendData]
                        callback id:     28
                        transmit status: OK
13:05:03.900 SERIAL « 0x010b00040007057a050100018d                                        (13 bytes)
13:05:03.901 SERIAL » [ACK]                                                                   (0x06)
13:05:03.902 DRIVER « [Node 007] [REQ] [ApplicationCommand]
                      └─[FirmwareUpdateMetaDataCCGet]
                          total # of reports: 1
                          report number:      1

**Device information**

Which device(s) is/are affected (make/model)? Qubino Smart Plug 16A
What are the node IDs? 007

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [X] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Click on firmware update
2. Select target 0
3. Select firmware

**Additional context**

Add any other context about the problem here.

**Logfile:**
[zwavejs_X.log](https://github.com/zwave-js/node-zwave-js/files/6141251/zwavejs_X.log)

AlCalzone, 2021-03-15:
zwave-js doesn't decide which fragment to send, because the device requests it from the controller. It looks like the command with the firmware fragment is received by the device but then it doesn't understand it or something.

I'll ask my contact if they can make sense of this - for me it looks like zwave-js does the right thing.

ualex73, 2021-03-15:
@AlCalzone thanks for the quick response, and hopefully you get something from you Qubino contact.

AlCalzone, 2021-03-15:
Okay I was told that Qubino tested the OTA functionality with the PC controller and Z-Wave.me in unsecure, S0 and S2 mode and that it should work.

Can you try to do the update with PC controller (included in [Simplicity Studio](https://www.silabs.com/developers/simplicity-studio#releasenotes)) to check if it works there?

If it doesn't, there's likely a problem with your device and you should contact the Qubino support.
If it does work, please let me know so Qubino can try to replicate the issue.

ualex73, 2021-03-15:
Ok, i will give the simplicity studio a try.

ualex73, 2021-03-15:
I tried it, but not-secure and secure (S0) both end successfully with uploading the firmware. I hear a click after the firmware upgrade, but when I re-check firmware (checksum and version), nothing has changed.

AlCalzone, 2021-03-16:
Alright, can you contact the Qubino support about this please?

ualex73, 2021-03-16:
yes, I did ... so I will close this one.

closed 2021-03-16 by ualex73

AlCalzone, 2021-03-16:
Please keep me updated when you know more.