zwave-js/zwave-js #2080: Ecolink DW-ZWAVE2 showing unavailable indefinitely

mattbruman, 2021-03-15:
I have about 20 of these and they always show up as unavailable even after pairing a new one it never finish. Log shows it finishes and lists its abilities e.g burglar, battery...etc but in HA it stays unavailable/unknown without its sensors being created. Like I said I tried adding a new one which should negate the waking up issue. I also tried zwavejs and zwavejs2mqtt and both controllers did the same. I have since restored my snapshot back to using 1.4 since I cant have my door alarms offline. But I am willing to try any tests on the weekends.

AlCalzone, 2021-03-15:
You deleted the issue template and it seems like you didn't read it :(

If the log indicates that the interview is finished but it doesn't work in HA, I suspect there is a HA issue. Please report the issue in the HomeAssistant repo.
If they rule out HA as the source of the problem, you may come back here - but please fill out the entire template.

closed 2021-03-15 by AlCalzone

mattbruman, 2021-03-15:
> You deleted the issue template and it seems like you didn't read it :(
Didn't see a template it didn't show up after I clicked create and then logged in. Must be a github bug

AlCalzone, 2021-03-15:
Alright, then please open a new issue with all the requested info so I can investigate.